libusb/libusb #797: libusb_get_parent doc

a1ien, 2020-10-13:
In documentation about `libusb_get_parent` say

>You should issue a libusb_get_device_list() before calling this function and make sure that you only access the parent before issuing libusb_free_device_list(). The reason is that libusb currently does not maintain a permanent list of device instances, and therefore can only guarantee that parents are fully instantiated within a libusb_get_device_list() - libusb_free_device_list() block.

In implementation we just do this https://github.com/libusb/libusb/blob/eee6998395184d87bd8e9c07ce2637caed1207f4/libusb/core.c#L997-L1000

And in `libusb_get_port_numbers` https://github.com/libusb/libusb/blob/eee6998395184d87bd8e9c07ce2637caed1207f4/libusb/core.c#L961-L968
we also just access to `dev->parent_dev`.

how much the current documentation corresponds to reality? We can call `libusb_get_parent` in between calls  `libusb_get_device_list()` - `libusb_free_device_list()` or this part irrelevant?

dickens, 2020-10-16:
The statement

> libusb currently does not maintain a permanent list of device instances

is still true on all platforms that do not support hotplug.

A reference count to the parent device is kept in all cases, so if one were to do

```
libusb_get_device_list(NULL, &dev_list);
dev = libusb_ref_device(dev_list[n]);
libusb_free_device_list(dev_list, 1);
parent = libusb_get_parent(dev);
```

this would be valid.

What exactly is the issue or change request?

a1ien, 2020-10-16:
> is still true on all platforms that do not support hotplug.

> What exactly is the issue or change request?

Then the function  `libusb_get_port_numbers` does not work correctly. Because they not call  libusb_get_device_list befor access to `dev->parent_dev`

closed 2020-10-16 by a1ien