libusb/libusb #801: Track send progress using libusb_bulk_transfer or libusb_fill_bulk_transfer.

ganeshrvel, 2020-10-24:
How can I track the progress of data buffer transfer to the device while using libusb_fill_bulk_transfer?

diabolo38, 2020-10-24:
you can't even in kernel driver it's not possible 
if for any valid  reason you need it send multiple smaller request instead of a  single big one 
this shall be fine until the  cut small request are still multiple of the end point max packet size 

Then you'll get intermediate call back/Completion not just the "big one"

ganeshrvel, 2020-10-24:
shoot, that's going to be a heck slower!

diabolo38, 2020-10-25:
if you use multiple assynch request you should get same or real close performances but system wise it's sure worst

mcuee, 2020-10-26:
Please use libusb mailing list for question like this in the future. Thanks.

closed 2020-10-26 by mcuee