libusb/libusb #803: usbi_backend->kernel_driver_active always return 0

alexbour, 2020-10-26:
usbi_backend->kernel_driver_active seems always return 0, whildriver seems to be attached.
my system linux Fedora33

`.............."v1.0-23{11397}http://libusb.info".......................................
[ 3.075310] [00000b77] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 3.075339] [00000b77] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb

[ 3.075347] [00000b77] libusb: debug [get_kernel_version] reported kernel version is 5.8.16-300.fc33.x86_64`

`$ dmesg | grep usb
[    0.377474] usbcore: registered new interface driver usbfs
[    0.377474] usbcore: registered new interface driver hub
[    0.377474] usbcore: registered new device driver usb
[    0.420325] pci 0000:00:1a.0: quirk_usb_early_handoff+0x0/0x669 took 15805 usecs
[    0.436321] pci 0000:00:1d.0: quirk_usb_early_handoff+0x0/0x669 took 15607 usecs
[    0.867355] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.08
[    0.867356] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.867357] usb usb1: Product: EHCI Host Controller
[    0.867358] usb usb1: Manufacturer: Linux 5.8.16-300.fc33.x86_64 ehci_hcd
[    0.867359] usb usb1: SerialNumber: 0000:00:1a.0
[    0.878362] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.08
[    0.878364] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.878364] usb usb2: Product: EHCI Host Controller
[    0.878365] usb usb2: Manufacturer: Linux 5.8.16-300.fc33.x86_64 ehci_hcd
[    0.878366] usb usb2: SerialNumber: 0000:00:1d.0
[    0.880086] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.08
[    0.880087] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.880088] usb usb3: Product: xHCI Host Controller
[    0.880088] usb usb3: Manufacturer: Linux 5.8.16-300.fc33.x86_64 xhci-hcd
[    0.880089] usb usb3: SerialNumber: 0000:00:14.0
[    0.882091] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.08
[    0.882092] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.882092] usb usb4: Product: xHCI Host Controller
[    0.882093] usb usb4: Manufacturer: Linux 5.8.16-300.fc33.x86_64 xhci-hcd
[    0.882093] usb usb4: SerialNumber: 0000:00:14.0
[    0.882633] usbcore: registered new interface driver usbserial_generic
[    0.882636] usbserial: USB Serial support registered for generic
[    0.883547] usbcore: registered new interface driver usbhid
[    0.883547] usbhid: USB HID core driver
[    1.189352] usb 1-1: new high-speed USB device number 2 using ehci-pci
[    1.197352] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    1.205351] usb 3-1: new low-speed USB device number 2 using xhci_hcd
[    1.308723] usb 1-1: New USB device found, idVendor=8087, idProduct=8008, bcdDevice= 0.04
[    1.308724] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.316710] usb 2-1: New USB device found, idVendor=8087, idProduct=8000, bcdDevice= 0.04
[    1.316712] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.392674] usb 3-1: New USB device found, idVendor=040b, idProduct=2000, bcdDevice= 2.05
[    1.392678] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.392681] usb 3-1: Product: USB Keyboard
[    1.392684] usb 3-1: Manufacturer: Generic
[    1.398522] input: Generic USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:040B:2000.0001/input/input2
[    1.450924] hid-generic 0003:040B:2000.0001: input,hidraw0: USB HID v1.10 Keyboard [Generic USB Keyboard] on usb-0000:00:14.0-1/input0
[    1.457081] input: Generic USB Keyboard System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:040B:2000.0002/input/input3
[    1.508677] input: Generic USB Keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:040B:2000.0002/input/input4
[    1.508920] input: Generic USB Keyboard Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:040B:2000.0002/input/input5
[    1.509249] hid-generic 0003:040B:2000.0002: input,hidraw1: USB HID v1.10 Mouse [Generic USB Keyboard] on usb-0000:00:14.0-1/input1
[    1.623358] usb 3-7: new full-speed USB device number 3 using xhci_hcd
[    1.752359] usb 3-7: New USB device found, idVendor=046d, idProduct=c534, bcdDevice=29.01
[    1.752362] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.752365] usb 3-7: Product: USB Receiver
[    1.752366] usb 3-7: Manufacturer: Logitech
[    1.755973] input: Logitech USB Receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:046D:C534.0003/input/input6
[    1.808017] hid-generic 0003:046D:C534.0003: input,hidraw2: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-7/input0
[    1.811149] input: Logitech USB Receiver Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:046D:C534.0004/input/input7
[    1.811460] input: Logitech USB Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:046D:C534.0004/input/input8
[    1.863490] input: Logitech USB Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:046D:C534.0004/input/input9
[    1.863843] hid-generic 0003:046D:C534.0004: input,hiddev96,hidraw3: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-7/input1
[    1.977336] usb 3-8: new high-speed USB device number 4 using xhci_hcd
[    2.103493] usb 3-8: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[    2.103495] usb 3-8: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.103495] usb 3-8: Product: USB 2.0 Hub
[    2.490261] usb 3-8.1: new full-speed USB device number 5 using xhci_hcd
[    2.604428] logitech-djreceiver 0003:046D:C534.0003: hidraw2: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-7/input0
[    2.681251] usb 3-8.1: New USB device found, idVendor=1cbe, idProduct=00fd, bcdDevice= 1.00
[    2.681253] usb 3-8.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.681261] usb 3-8.1: Product: In-Circuit Debug Interface
[    2.681261] usb 3-8.1: Manufacturer: Texas Instruments
[    2.681262] usb 3-8.1: SerialNumber: 0E229ED3
[    2.693552] logitech-djreceiver 0003:046D:C534.0004: hiddev96,hidraw3: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-7/input1
[    2.750811] input: Logitech Wireless Mouse PID:4054 Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:046D:C534.0004/0003:046D:4054.0005/input/input12
[    2.751108] hid-generic 0003:046D:4054.0005: input,hidraw4: USB HID v1.11 Mouse [Logitech Wireless Mouse PID:4054] on usb-0000:00:14.0-7/input1:2
[    2.874267] usb 3-8.3: new full-speed USB device number 6 using xhci_hcd
[    2.874722] input: Logitech Wireless Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:046D:C534.0004/0003:046D:4054.0005/input/input16
[    2.874776] logitech-hidpp-device 0003:046D:4054.0005: input,hidraw4: USB HID v1.11 Mouse [Logitech Wireless Mouse] on usb-0000:00:14.0-7/input1:2
[    3.071257] usb 3-8.3: New USB device found, idVendor=16c0, idProduct=05dc, bcdDevice= 1.00
[    3.071258] usb 3-8.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.071259] usb 3-8.3: Product: VE12Demo Board_ctrl
[    3.071260] usb 3-8.3: Manufacturer: SCASIC_pro_and_VE
[    3.071260] usb 3-8.3: SerialNumber: JUL_2020AU
[    9.468646] usbcore: registered new interface driver cdc_acm`

mcuee, 2020-10-27:
Which device is in question? Can you post the full libusb debug log?

alexbour, 2020-11-01:
mcuee, thank you so much for response. please find required output below
****************************************************************************************************
`.............."-------------------------connecting device----------------".......................................
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 3.024718] [00015dbd] libusb: debug [libusb_get_device_list] 
[ 3.024747] [00015dbd] libusb: debug [discovered_devs_append] need to increase capacity
[340747866]
.............."connecting board[16C0:5DC]]
found  15 candidates".......................................
[340762278]
.............."connecting board[16C0:5DC]]
found  15 candidates".......................................
[ 3.024794] [00015dbd] libusb: debug [libusb_get_device_descriptor] 
[ 3.024802] [00015dbd] libusb: debug [libusb_get_device_descriptor] 
[340790889]
..............".....................board identified................".......................................
[ 3.024819] [00015dbd] libusb: debug [libusb_open] open 2.7
[ 3.024874] [00015dbd] libusb: debug [usbi_add_pollfd] add fd 30 events 4
[340864122]
.............."device 7openned".......................................
[ 3.024892] [00015dbd] libusb: debug [libusb_kernel_driver_active] interface 0
[340884684]
.............."kernel isn't active".......................................
[340895011]
.............."set autodetouch".......................................
[ 3.024922] [00015dbd] libusb: debug [libusb_claim_interface] interface 0
[340961730]
.............."interface claimed".......................................
[ 3.024992] [00015dbd] libusb: debug [libusb_alloc_transfer] transfer 0x19cfb90
[ 3.024997] [00015dbd] libusb: debug [libusb_submit_transfer] transfer 0x19cfb90
[ 3.025003] [00015dbd] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 3.025026] [00015dbd] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.025030] [00015dbd] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.025037] [00015dbd] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.025188] [00015dbd] libusb: debug [handle_events] poll() returned 1
[ 3.025200] [00015dbd] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 3.025205] [00015dbd] libusb: debug [handle_control_completion] handling completion status 0
[ 3.025209] [00015dbd] libusb: debug [disarm_timerfd] 
[ 3.025214] [00015dbd] libusb: debug [usbi_handle_transfer_completion] transfer 0x19cfb90 has callback 0x7feabac2eb10
[ 3.025217] [00015dbd] libusb: debug [sync_transfer_cb] actual_length=4
[ 3.025222] [00015dbd] libusb: debug [libusb_free_transfer] transfer 0x19cfb90
[ 3.025228] [00015dbd] libusb: debug [libusb_alloc_transfer] transfer 0x19cc370
[ 3.025231] [00015dbd] libusb: debug [libusb_submit_transfer] transfer 0x19cc370
[ 3.025234] [00015dbd] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 3.025244] [00015dbd] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.025248] [00015dbd] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.025430] [00015dbd] libusb: debug [handle_events] poll() returned 1
[ 3.025442] [00015dbd] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=36
[ 3.025446] [00015dbd] libusb: debug [handle_control_completion] handling completion status 0
[ 3.025450] [00015dbd] libusb: debug [disarm_timerfd] 
[ 3.025455] [00015dbd] libusb: debug [usbi_handle_transfer_completion] transfer 0x19cc370 has callback 0x7feabac2eb10
[ 3.025459] [00015dbd] libusb: debug [sync_transfer_cb] actual_length=36
[ 3.025464] [00015dbd] libusb: debug [libusb_free_transfer] transfer 0x19cc370
[ 3.025469] [00015dbd] libusb: debug [libusb_alloc_transfer] transfer 0x1ab4a30
[ 3.025473] [00015dbd] libusb: debug [libusb_submit_transfer] transfer 0x1ab4a30
[ 3.025476] [00015dbd] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 3.025486] [00015dbd] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.025490] [00015dbd] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.025682] [00015dbd] libusb: debug [handle_events] poll() returned 1
[ 3.025702] [00015dbd] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 3.025709] [00015dbd] libusb: debug [handle_control_completion] handling completion status 0
[ 3.025715] [00015dbd] libusb: debug [disarm_timerfd] 
[ 3.025722] [00015dbd] libusb: debug [usbi_handle_transfer_completion] transfer 0x1ab4a30 has callback 0x7feabac2eb10
[ 3.025729] [00015dbd] libusb: debug [sync_transfer_cb] actual_length=4
[ 3.025736] [00015dbd] libusb: debug [libusb_free_transfer] transfer 0x1ab4a30
[ 3.025744] [00015dbd] libusb: debug [libusb_alloc_transfer] transfer 0x1aada10
[ 3.025749] [00015dbd] libusb: debug [libusb_submit_transfer] transfer 0x1aada10
[ 3.025753] [00015dbd] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 3.025774] [00015dbd] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[... 634 lines cut ...]
[ 3.172476] [00017d88] libusb: debug [linux_get_device_address] getting address for device: 2-1.4.3 detached: 0
[... cut ...]
[ 6.173398] [00017d88] libusb: debug [discovered_devs_append] need to increase capacity
[ 6.173415] [00017d88] libusb: debug [libusb_get_device_descriptor] 
[ 6.173419] [00017d88] libusb: debug [libusb_get_device_descriptor] 
[ 6.173427] [00017d88] libusb: debug [libusb_open] open 2.7
[ 6.173478] [00017d88] libusb: debug [usbi_add_pollfd] add fd 34 events 4
[ 6.173489] [00017d88] libusb: debug [libusb_kernel_driver_active] interface 0
[ 6.173503] [00017d88] libusb: debug [libusb_claim_interface] interface 0

`

alexbour, 2020-11-01:
the device in question is 16C0:5DC, devaddress=8?

alexbour, 2020-11-01:
code in question:
` bool   usbcomm::connect(void)
 {struct libusb_device_descriptor desc;
     unsigned char uctmp1[128];
     unsigned char uctmp2[128];
              char   cbuf[512];
          libusb_device *dev;
          size_t idx=0;
          int rc=-1,rc1=-1,rc2=-1;

      showstat("-------------------------connecting device----------------");
      if(devlist==nullptr)
      {

     deviceCount = libusb_get_device_list(ctx, &devlist);

     snprintf(cbuf,sizeof(cbuf),"connecting board[%03X:%03X]]\nfound  %lu candidates",VENDORID,PRODUCTID,deviceCount);
     showstat(cbuf);
     if(devlist==NULL||deviceCount==0)errmsg(__LINE__-1,__FILE__,"can't get  device List",deviceCount,true);
     else showstat(cbuf);
      };
     if(pdev_handle!=nullptr)showstat("double initialization");

     ///////////////////////////////////////////////////
     /////////////////////////////////////////////////// /// \brief snprintf
      for (idx = 0; idx<=deviceCount; idx++)
     {
         if( (dev=devlist[idx])==NULL) break;//last device in the list
         rc=libusb_get_device_descriptor(dev, &desc);
        if(rc!=LIBUSB_SUCCESS)errmsg(__LINE__-1,__FILE__,"can't get  device descriptor",rc,true);
      if( desc.idVendor==VENDORID&&desc.idProduct==PRODUCTID) break;
     };//endof for loop

     if((ok=((dev!=NULL)&(desc.idVendor==VENDORID)&(desc.idProduct==PRODUCTID)))) showstat(".....................board identified................");
     else   return errmsg(__LINE__-1,__FILE__,"can't find the  board",0,true);
///////////////////////////////////////////////////
     address=libusb_get_device_address(dev);
     bus=libusb_get_bus_number(dev);
///////////////////////////////////////////////////
        rc=libusb_open(dev, &pdev_handle);
        snprintf(cbuf,sizeof(cbuf),"device %uopenned",address);
        if(rc!=LIBUSB_SUCCESS||pdev_handle==NULL)ok&=errmsg(__LINE__-1,__FILE__,"can't open  device",rc,true);
        else showstat(cbuf);

          ////////////////////////////////////////claim interface/////////////////////////////
          rc=libusb_kernel_driver_active( pdev_handle, INTERFACENUMBER);
          if(force_detach&&!rc) rc=34;
        switch(rc)
        {
        case 0:
            showstat("kernel isn't active");
        break;
        case 34:
             showstat("kernel driver forced detach");
        case 1:   showstat("kernel driver is active");
            rc=libusb_detach_kernel_driver(pdev_handle, INTERFACENUMBER);
         if(rc!=LIBUSB_SUCCESS) errmsg(__LINE__-1,__FILE__,"kernel driver failed to be  detached",rc,true);
          else  showstat("kernel driver got detached");
        break;
       default:
            errmsg(__LINE__-1,__FILE__,"kernel_driver_active failed",rc,true);
        };//endof switch
        if(autodetach)
        {
        rc=libusb_set_auto_detach_kernel_driver(pdev_handle, true);
        if(rc!=LIBUSB_SUCCESS)errmsg(__LINE__-1,__FILE__,"kernel driver failed to be set  autodetached",rc,true);
        else  showstat("set autodetouch");
        };//endof autodetouch
        snprintf(cbuf,sizeof(cbuf),"can't claim interface %d ", INTERFACENUMBER);
        rc=libusb_claim_interface(pdev_handle, INTERFACENUMBER);
        if(rc!=LIBUSB_SUCCESS) ok=errmsg(__LINE__-1,__FILE__,cbuf, rc,true);
        else  showstat("interface claimed");

        ////////////////////////////////////////////endof    claim interface/////////////////////////////////////
               /*
                  snprintf(cbuf,sizeof(cbuf),"can't set configuration %d ",CONFIGNUMBER);
                  rc=libusb_set_configuration(pdev_handle, CONFIGNUMBER);
                   if(rc!=LIBUSB_SUCCESS) ok=errmsg(__LINE__-1,__FILE__,cbuf, rc,true);
         */

    rc1 =libusb_get_string_descriptor_ascii(pdev_handle,desc.iManufacturer,uctmp1,sizeof(uctmp1));
    if(rc1<=0)  errmsg(__LINE__-1,__FILE__,"can't get  manufacturer ascii",rc,true);
    rc2 =libusb_get_string_descriptor_ascii(pdev_handle,desc.iProduct,uctmp2,sizeof(uctmp2));
    if(rc2<=0)  errmsg(__LINE__-1,__FILE__,"can't get  product ascii",rc,true);
   if(!(ok=(pdev_handle!=NULL)&(rc1>0)&(rc2>0)&(rc==0)))   errmsg(__LINE__-1,__FILE__,"can't connect board",deviceCount,true);
   else {
     snprintf(cbuf,sizeof(cbuf),"board[%3u:%3u] connected %s:%s",desc.idVendor,desc.idProduct, uctmp1,uctmp2);
    showstat(cbuf);
    };
 return ok;
 }//endof connect
//////////////////////////////////////////////////////////////////////////////////////////////////////////`

dickens, 2020-11-01:
The `libusb_kernel_driver_active()` returns whether or not a _kernel_ driver is bound to the specified interface.  It does not report whether the interface has been claimed by another process or usbfs instance.  For example, the following sequence, even in the same process, will fail:

```
libusb_device_handle *h1, *h2:
libusb_open(dev, &h1); // assume this succeeds
libusb_claim_interface(h1, 0); // assume this succeeds
libusb_open(dev, &h2); // assume this succeeds
libusb_claim_interface(h2, 0); // this will fail with LIBUSB_ERROR_BUSY
```

alexbour, 2020-11-01:
OK, thankk everyone for your timeand expertise, I found error: somehow my qt window  get initialize twice and call connect() function twice, no wonder second time routine tries to claim iface, it fails miserably.therefore libusb works as expected.
Once again, appreciate your fellow inputs.
Alex.

closed 2020-11-01 by alexbour