libusb/libusb #804: Get device list takes an infinite amount of time with 3rd party controllers

german77, 2020-10-27:
Some third party controllers like [titan one](https://www.consoletuner.com/products/titan-one/), Xtreme usb controler cause this specific function to take and infinite amount of time to return something. As soon as the controller is disconnected the function is able to return the list without any problem.

mcuee, 2020-10-28:
Please provide more details. Is this regarding an application using libusb or libusb itself?

What is the OS and what is the application?

german77, 2020-10-28:
All issues where found in windows. In the yuzu emulator. Libusb is used to communicate with a Gamecube adapter. 

The library works in most PC without a problem but people with certain drivers / 3rd party controllers will cause the application to hang while the controller is connected.

I debugged yuzu and get_device_list was the function that was causing the application to hang.
At least 1 person per week comes up with this problem.

mcuee, 2020-10-30:
yuzu Gamecube adapter codes:
https://github.com/yuzu-emu/yuzu/tree/master/src/input_common/gcadapter

Please help to provide the relevant debug log.

dickens, 2020-12-03:
Still waiting for the debug log.

mcuee, 2020-12-08:
@german77 Please help to provide the debug log. Thanks.

german77, 2020-12-08:
I will when I can contact the person with the problem again. Or find another with the same issue willing to help. Right now I can't get a log because I don't have the issue. With the last change I made there are fewer reports of this problem, the problem still there but now goes unnoticed.

Sorry for making you wait. But I can't do much if I don't have the problem.

mcuee, 2020-12-10:
@german77 No problem. I will keep the issue open.

mcuee, 2021-01-16:
BTW, it seems there is no report of similar issues from Dolphin. The code seems to be common.
https://github.com/dolphin-emu/dolphin/tree/master/Source/Core/InputCommon

mcuee, 2021-01-25:
@german77 If possible, please help to see if the issue is still there with latest git and send the debug log. Thanks.

german77, 2021-02-02:
I haven't seen more users with the same problem lately

closed 2021-02-02 by german77

mcuee, 2022-04-28:
In case there are future reports, please check out Issue #1136 and pull request #1127.
* #1136 
* #1127