libusb/libusb #807: libusbK allows claiming the interface from different applications at the same time

jankae, 2020-11-16:
Some background: I have an application that connects to a USB device. Potentially, there could be multiple instances running, each connected to a unique device. The devices have a serialnumber to differentiate between them in the UI. The program should work like this:
1. Create a list of connected devices with libusb_get_device_list
2. Open (and immediately close afterwards) every device with the correct VID/PID to get the serialnumber
3. Show available serialnumbers in UI
4. If the user selects a device, open it again and claim the interfaces (this should fail if the interfaces are already claimed by another instance of the application)

Step 2 is not possible with the WinUSB driver (if the device is already connected to another instance of the program, opening it to get the serialnumber is not possible), so I switched to libusbK. I now am able to retrieve all serialnumbers, however claiming the interface *always* succeeds, even when it was already claimed by another instance. This leads to serious communication problems because two instances are connected to the same device.

As far as I know, libusb_claim_interface should fail if something else already claimed the interface. This is also what happens if I run my application on Linux.
Here is a minimal example to reproduce the problem:
```c
#include <stdio.h>
#include <libusb-1.0/libusb.h>
#include <windows.h>

int main()
{
    libusb_context *ctx;
    int res = libusb_init(&ctx);
    if (res != 0)
    {
      printf("Error initialising libusb.\n");
      return 1;
    }

    libusb_device_handle* handle = libusb_open_device_with_vid_pid(ctx, 0x0483, 0x4121);
    if (!handle)
    {
      printf("Unable to open device.\n");
      return 1;
    }

    int ret = libusb_claim_interface(handle, 1);
    if (ret < 0) {
        printf("Failed to claim interface: %s\n", libusb_strerror((libusb_error) ret));
    } else {
        printf("Claimed interface\n");
    }

    Sleep(10000);

    libusb_release_interface(handle, 1);
    libusb_close(handle);
    libusb_exit(ctx);
}
```
When starting the first instance of this example it prints "Claimed interface" as expected. However, when I start a second instance while the first is still running, it also prints "Claimed interface". I would have expected "Failed to claim interface: LIBUSB_ERROR_BUSY"

I am using these versions:
* libusb v1.0.23
* libusbK v3.0.7.0 (installed with Zadig)
* Windows 10

diabolo38, 2020-11-16:
Actulay you explained exactly what is possible  with libusbk  VS winusb
But they are couple operation you can do after open but  without claiming the interface, reading descriptor is one of thems
So in scope of finding a device via a serial number or some other string desc  is possible under all Os even window with win usb.

jankae, 2020-11-21:
Okay, it could be that this is actually the normal behavior but doesn't the [API](http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#gaee5076addf5de77c7962138397fd5b1a) say otherwise?
> Returns: LIBUSB_ERROR_BUSY if another program or driver has claimed the interface

Also, could you please elaborate on this:
> So in scope of finding a device via a serial number or some other string desc is possible under all Os even window with win usb.

I can't open the device if another instance of the application already opened it (when using WinUSB) and I didn't find a way to read the serial number without opening it first.

diabolo38, 2020-11-23:
You can normally open the device multiple time but only claim any interface once
That work and I tested it under linux/Mac and it should under window/any backend but I only used libusbk so never tested it personaly.

So to find a device potentially opening it and  reading string  or  over desc shall be  possible wthout claiming device even under window/winusb.

jankae, 2020-11-29:
>You can normally open the device multiple time but only claim any interface once

Yes, that is exactly what I am doing on Linux but can't get to work on Windows:
* With WinUSB I can not open the same device twice (this behavior is documented and apparently a limitation of the WinUSB backend)
* With libusbk I can claim the interface twice (libusb API says that this should not be possible)

diabolo38, 2020-11-29:
unfortunately it doesn't work with winsub :(
 just tried it got same issue second open fail with LIBUSB_ERROR_ACCESS = -3,
log for failure is attached
 
[log.txt](https://github.com/libusb/libusb/files/5612628/log.txt)

mcuee, 2020-12-10:
Probably we can not do much about this right now. I will update the Wiki as of now.

mcuee, 2020-12-10:
Wiki updated.
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions

dickens, 2021-01-19:
This is really a libusbK issue.  The underlying driver is responsible for enforcing mutual exclusion, so I'm going to close this.

closed 2021-01-19 by dickens

mcuee, 2021-01-20:
I agree to close this and I think we are covered now that Wiki has been updated.

mcuee, 2021-01-20:
Please also take note libusbk project is now marked as unmaintained.
https://github.com/mcuee/libusbk

mcuee, 2022-03-04:
libusbk is now maintained again.

mcuee, 2023-10-12:
Please have a look at : https://github.com/libusb/libusb/pull/1323

mcuee, 2023-10-12:
Further discussion should be in #1087.