libusb/libusb #812: .gitattributes causes bocus changes to files and prevents old version checkout with tag

MarkoTurunen, 2020-11-30:
On Linux (not sure about Windows) there are many bocus changes reported right after new clone. Reason for this behavior seems to be quite recent addition of .gitattributes file. This prevents checking out specific release using tag, because git thinks that there are uncommitted changes on workarea.

- `git restore .` does not help
- `git reset --hard` does not help
- `git stash` does not help

The only way to get this work is by deleting `.gitattributes` file and commit deletion.

LudovicRousseau, 2020-11-30:
I also have the problem on macOS.
I just did a `git merge upstream/master` and now I have modified files I can fix.
```
± LANG=C git status 
On branch master
Your branch is ahead of 'origin/master' by 16 commits.
  (use "git push" to publish your local commits)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   dpfp_2013.vcxproj
	modified:   dpfp_2013.vcxproj.filters
	modified:   dpfp_2015.vcxproj
	modified:   dpfp_2015.vcxproj.filters
	modified:   dpfp_2017.vcxproj
	modified:   dpfp_2017.vcxproj.filters
	modified:   dpfp_2019.vcxproj
	modified:   dpfp_2019.vcxproj.filters
	modified:   dpfp_threaded_2013.vcxproj
	modified:   dpfp_threaded_2013.vcxproj.filters
	modified:   dpfp_threaded_2015.vcxproj
	modified:   dpfp_threaded_2015.vcxproj.filters
	modified:   dpfp_threaded_2017.vcxproj
	modified:   dpfp_threaded_2017.vcxproj.filters
	modified:   dpfp_threaded_2019.vcxproj
	modified:   dpfp_threaded_2019.vcxproj.filters
	modified:   sam3u_benchmark_2013.vcxproj
	modified:   sam3u_benchmark_2013.vcxproj.filters
	modified:   sam3u_benchmark_2015.vcxproj
	modified:   sam3u_benchmark_2015.vcxproj.filters
	modified:   sam3u_benchmark_2017.vcxproj
	modified:   sam3u_benchmark_2017.vcxproj.filters
	modified:   sam3u_benchmark_2019.vcxproj
	modified:   sam3u_benchmark_2019.vcxproj.filters

```

it looks like the problem is with end of line encoding. I do not see other changes from `git diff`.

I have no idea why some files do not have the problem like `xusb_2019.vcxproj` in the same directory.

The `.gitattributes` file is quiet old (initial commit in 2011). Maybe the problem is with a new version of git?

diabolo38, 2020-11-30:
I have similar issues on cross platform project mixing dos/unix files , checkout/in in linux or mac killing dos files or creating full file change due to eol.
one solution is to force gitconfig core.autocrlf=false
developer must also take care to configure their editor to use correct cr/lf when configurable.
option to ommit white space diff also help

ev-mp, 2020-11-30:
Hello, the change introduced in v1.0.24  breaks compatibility with Cmake, and more specifically with its **`ExternalProject_Add** API.

I'm one of the maintainers of [Librealsense](https://github.com/IntelRealSense/librealsense) and we encountered this issues as well. In our case, however, `libusb` is added into the project using CMake's `ExternalProject_Add` command so there is no way to work-around in code as the flow is hard-coded internally in Cmake and it performs `git clone .. master -> git checkout <the actual branch I need>`.
So even when non-master branch is requested using Cmake's `ExternalProject_Add` cannot work.

Currently we use forked `libusb`  with the `master` being reverted to previous version (before the vcxproj and filter files were added to CI). 
Hope that a resolution will be provided in a timely manner

MarkoTurunen, 2020-11-30:
> The .gitattributes file is quiet old (initial commit in 2011). Maybe the problem is with a new version of git?

True, I used `git blame` and misread the times there. Probably new version of git then. Anyways as mentioned above, this breaks CMake `ExternalProject_Add` API when using `GIT_TAG`.

commit cd431859 (2020-11-30): fix eol of msvc/*.vcxproj and msvc/*.vcxproj.filters files

Closes #812
Closes #813
closed 2020-11-30 by dickens