libusb/libusb #814: No Bulk IN Endpoint listed

banchanKiroHashtaag, 2020-12-02:
Hi,

I'm new to libusb, and I am writing a program to automatically detect and configure my devices.
I am not sure if this is the right place to get some support with my code, if not please tell me where I can get some support, community or otherwise.

So right now, I am just trying to get the program to detect and receive a simple string message from my device. 

There are 2 Bulk Endpoints listed (determined by the attribute on the endpoint descriptor), but one is eternally busy (errno=16), whereas the other one is an out address (0x02). 

So there is no IN endpoint address listed.

I have also tried to send a string message to the device, but can't seem to get it to work.

Below is my entire code:
```
#include <stdio.h>
#include <string.h>

#include <fcntl.h> // Contains file controls like O_RDWR
#include <errno.h> // Error integer and strerror() function
#include <termios.h> // Contains POSIX terminal control definitions
#include <unistd.h> // write(), read(), close()
#include <libusb-1.0/libusb.h>

#define DEVICE_FOUND 		30000
#define DEVICE_NOT_FOUND 	-30001
#define NO_DEVICE_LIST 		-30002

#define BULK_EP_OUT     	(0x02 | LIBUSB_ENDPOINT_OUT)   
#define BULK_EP_IN      	(0x02 | LIBUSB_ENDPOINT_IN) 

#define DEBUG

typedef struct device {
	libusb_device_handle *handle;
	const struct libusb_endpoint_descriptor *endpoint;
	uint8_t endpoint_in;
	uint8_t endpoint_out;
} device;

int interface_ref = 0;    
int alt_interface, interface_number;
uint8_t endpoint_in, endpoint_out;

device devices[20] = {0};

const struct libusb_endpoint_descriptor* get_endpoint(struct libusb_device *dev, 
	struct libusb_device_handle *handle, 
	int device_index, 
	struct libusb_config_descriptor *config)    
{    
    struct libusb_device_handle *hDevice_req;    
    const struct libusb_endpoint_descriptor *endpoint;    
    int altsetting_index, interface_index=0, ret_active;    
    int i;    

    hDevice_req = handle;    

    for(interface_index=0; interface_index < config->bNumInterfaces; interface_index++)    
    {    
        const struct libusb_interface *iface = &config->interface[interface_index];    
        for(altsetting_index=0; altsetting_index < iface->num_altsetting; altsetting_index++)    
        {    
            const struct libusb_interface_descriptor *altsetting = &iface->altsetting[altsetting_index];    

            int endpoint_index;    
            for(endpoint_index=0; endpoint_index < altsetting->bNumEndpoints; endpoint_index++)    
            {    
                const struct libusb_endpoint_descriptor *ep = &altsetting->endpoint[endpoint_index];    
                endpoint = ep;      
[... 743 lines cut ...]
[ 0.104079] [00006e18] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.104090] [00006e18] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.104095] [00006e18] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.104117] [00006e18] libusb: debug [handle_events] poll() returned 1
[ 0.104126] [00006e18] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=42
[ 0.104131] [00006e18] libusb: debug [handle_control_completion] handling completion status 0
[ 0.104135] [00006e18] libusb: debug [disarm_timerfd] 
[ 0.104140] [00006e18] libusb: debug [usbi_handle_transfer_completion] transfer 0x55cfb3b4f960 has callback 0x7fec3e5abb30
[ 0.104144] [00006e18] libusb: debug [sync_transfer_cb] actual_length=42
[ 0.104150] [00006e18] libusb: debug [libusb_free_transfer] transfer 0x55cfb3b4f960
Not a supported device
[ 0.104160] [00006e18] libusb: debug [libusb_close] 
[ 0.104167] [00006e18] libusb: debug [usbi_remove_pollfd] remove fd 20

1 device(s) found
Waiting for input
Endpoint IN: 0x00
Endpoint OUT: 0x02
[ 0.104191] [00006e18] libusb: debug [libusb_alloc_transfer] transfer 0x55cfb3b4f960
[ 0.104196] [00006e18] libusb: debug [libusb_submit_transfer] transfer 0x55cfb3b4f960
[ 0.104200] [00006e18] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
[ 0.104214] [00006e18] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.104219] [00006e18] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.104226] [00006e18] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
```

dickens, 2020-12-03:
Please provide the descriptors.  You can use `lsusb -v -d 0x0ab1:0x0007`.

mcuee, 2020-12-04:
Endpoint IN: 0x00
This is not valid. 

Are you developing the firmware for your device?

banchanKiroHashtaag, 2020-12-07:
> Endpoint IN: 0x00
> This is not valid unless you are using the default control endpoint (0x00 Control OUT, 0x80 Control IN).
> 
> Are you developing the firmware for your device?

Hi Thanks for the reply. I am not developing a firmware for the device,  I plan to make a Serial Communication program for the device. At this point I am trying to write code to automatically detect the device and take care of all the configurations so that the device is ready to use.

The Endpoint IN was not supposed to have value 0x00. This is only because it could not find an IN Endpoint of Bulk Type, thus the default value of the variable never got changed. One of the sub routines in the program was supposed to automatically scan all the available endpoints and then assign the required endpoints.

Also after doing some reading, I realised, the endpoints that I require are not the Bulk but the Isochronous Endpoints.

banchanKiroHashtaag, 2020-12-07:
> Please provide the descriptors. You can use `lsusb -v -d 0x0ab1:0x0007`.

Hi Dickens, Below is the descriptor I got from running the cmmand
```
Bus 002 Device 002: ID 0ab1:0007 FEIG ELECTRONIC GmbH 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0ab1 FEIG ELECTRONIC GmbH
  idProduct          0x0007 
  bcdDevice            4.09
  iManufacturer           1 FEIG ELECTRONIC GmbH
  iProduct                2 cVEND cD02.01.00-00.37-2-2
  iSerial                 3 11e931ff
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          154
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 CDC Composite (ECM + ACM)
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               7 CDC Serial
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              5 CDC Abstract Control Model (ACM)
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval              32
[... 88 lines cut ...]
      bInterfaceProtocol      0 
      iInterface             11 CDC Ethernet Data
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

mcuee, 2020-12-07:
#define BULK_EP_IN      	(0x02 | LIBUSB_ENDPOINT_IN) 

The above code is not correct. Your device does not have EP02 IN but only EP01 IN.

Anyway, you can not anyhow send a bulk transfer. You have to understand the device communication protocol. In your case, the device is a USB Composite device, with a CDC-ACM USB Serial Port and a CDC-ECM Ethernet port. By right you should not even use libusb but rather the normal serial API and network API to communicate to your device.

If you really want to use libusb and carry out low level communication with your device, then read USB specifications for CDC-ACM and CDC-ECM.

mcuee, 2020-12-07:
I will close this as it is not an libusb issue. You can ask for more help after you read the USB CDC-ACM specification.

closed 2020-12-07 by mcuee

mcuee, 2020-12-07:
Just one super simple example to get you started. 
https://github.com/tytouf/libusb-cdc-example/blob/master/cdc_example.c

BTW, I just remember my first post to libusb mailing list was also trying to communicate to a USB CDC-ACM device under Linux back in 2005. 
https://libusb-devel.narkive.com/nefpTO3u/libubs-devel-cdc-acm-under-linux-with-libusb

banchanKiroHashtaag, 2020-12-07:
Thank you for the suggestions. I will go through the materials and get back through the mailing list with more questions if required.