libusb/libusb #815: Attempting to flash older Samsung S4 android phones get bulk transfer errors -7

misanthropos, 2020-12-02:
This is just an assumption:

I believe there has been a regression in the last 1.5 years. But I can not pinpoint it to either the kernel or libusb or heimdall. It has become impossible to flash a Samsung android S4 mini with alternative recovery.img because of empty bulk transfers (error -7). It worked before. And I found a solution (Forget Windows Use Linux iso image which is about 3 years old)
I have tried the oldest linux kernel available to gentoo (4.20) and different libusb-versions - in vain. 
The FWUL iso file has a 4.19 kernel.

It could be that heimdall needs adaptions to newer libusb, since it has not been touched for years either.

Searching the net I found many others being unable to flash Samsung phones with the same error messages.
Maybe you can give some input on the issue. 

Thank you in advance.

dickens, 2020-12-03:
Please provide a debug log.

mcuee, 2020-12-04:
https://gitlab.com/BenjaminDobell/Heimdall/-/issues/526

It seems to me Heimdall is still under active development.

misanthropos, 2020-12-04:
Check:

https://gitlab.com/BenjaminDobell/Heimdall

No Changes for 3 years. Yes - many issues an PRs but none answered.

misanthropos, 2020-12-04:
@dickens The second link @mcuee supplied shows a similar log. I can create a log too, but it will look the same.

misanthropos, 2020-12-05:
I realized libusb debug output was missing in mentioned log file.

So I tried it with following setup:

- heimdall-1.4.2
- linux kernel 5.8.14-gentoo-x86_64
- libusb-1.0.23-r1

[heimdall.log](https://github.com/libusb/libusb/files/5646686/heimdall.log)

[console.log](https://github.com/libusb/libusb/files/5646689/console.log)

misanthropos, 2020-12-05:
Surprisingly this heimdall.log claimed it worked. So I tried it again with an older recovery image... and it DID work. I have tried for two days to get the alternative recovery image on that device before. And even though it looked like it worked, I still had stock recovery on the phone.

Since it works now, the problem must be something else then. Not sure though, what it actually is.
I had a system update including recompile of usbutils, libusb, libusb-compat in the meantime.

I did the exact same thing on three different linux-distributions (gentoo/debian/ubuntu) and only got it to work on FWUL. Weird.

mcuee, 2020-12-07:
In that case, can we say libusb is in fact working? Can we close this issue? You can always re-open if you have new findings.

misanthropos, 2020-12-07:
Agreed.

closed 2020-12-07 by misanthropos