libusb/libusb #816: Libusb bulk stream examples

chaitanya77185, 2020-12-03:
Hi,

I want to write a host applications which will use bulk streams to send and receive data.
But I didn't find reference how to use the bulk stream api's.

Can you provide a example which use bulk streams to transmit and receive data.

Thanks in advance.

Thanks & Regards,
Krishna Chaitanya.

dickens, 2020-12-03:
This is not an issue with the library, so I am closing this.  Please use the libusb-devel mailling list for such questions.

fenugrec, 2023-02-22:
Can we reopen this ? Documentation for the library is part of the library. We have nice doc pages discussing sync vs async, caveats, other general matters etc. But for streams, all we have is a few functions to allocate, free, and fill stream transfers, with zero information about 

- what is a "bulk stream"
- what is a "stream id" and when is it needed
- hints on when a bulk stream can be useful , or necessary

mcuee, 2023-02-23:
I will agree with this one. It is good to have better example and documentation for this "bulk stream" feature, which seems to be rarely used by the users.

mcuee, 2023-02-23:
Linux kernel documentation about Bulk Stream which is a USB 3 feature.
https://www.kernel.org/doc/Documentation/usb/bulk-streams.txt

Commit for Linux: https://github.com/libusb/libusb/commit/6d8dfe0aee41a3c7dd862c3de488df3ce452d86e
Commit for macOS: https://github.com/libusb/libusb/commit/e16575ceaf3148c389d66686e51af7cc8dff6053 (untested)

Other OS -- not implemented.

mcuee, 2023-02-23:
@fenugrec 

Linux kernel documentation about Bulk Stream which is a USB 3 feature.
https://www.kernel.org/doc/Documentation/usb/bulk-streams.txt

I think together with the[ USB 3 specification](https://www.usb.org/sites/default/files/usb_32_202206.zip) that would answer most of the questions you raised. But then an example is still good to have.

fenugrec, 2023-02-23:
> Bulk Stream which is a USB 3 feature.

That is what I suspected but had not confirmed; thanks for the info. Mentioning that somewhere in the docs would probably be a good idea too.

mcuee, 2026-03-06:
Recent discussion in libusb-devel mailing list.
https://sourceforge.net/p/libusb/mailman/message/59302818/

From: Praveena G
Sent: Thursday, February 19, 2026 3:55 PM
Subject: Seeking Guidance on High-Speed SCSI Command Execution in Linux: libusb vs UAS Driver/IO_uring Performance Implications

Hi Everyone,

I hope this email finds you well. We are currently working on developing a solution in Linux to send SCSI commands in UAS and BOT mode, with our primary goal being to achieve high-speed performance. So far, we have explored a couple of promising approaches: one using libusb and the other leveraging the internal UAS driver stack in combination with io_uring.

We have successfully implemented a proof of concept (POC) using libusb, which allows us to continuously send all SCSI commands. To achieve asynchronous transfer, we are utilizing the `libusb_fill_bulk_stream_transfer()` and `libusb_handle_events_completed()` APIs.

However, we have a query regarding the performance implications of libusb operating in user space. Specifically, for each command, there are three context switches between the kernel and user space to provide callbacks for the command phase, data phase, and status phase. We are concerned that these context switches might slow down the overall performance.

On the other hand, we understand that the underlying UAS driver stack can be directly accessed and used to send commands with a single API call, thereby reducing context switches. Additionally, the io_uring approach, which utilizes a shared ring buffer between user space and the kernel, minimizes system call overhead and could potentially offer faster performance.

Given these considerations, we are curious about the potential advantages of using libusb in this scenario. Could you please help us understand the benefits of libusb and guide us on the best approach to achieve high-speed performance? We would greatly appreciate any suggestions or insights you could provide to help us improve our solution.

Thank you in advance for your assistance and expertise.

Best regards,
Praveena G