libusb/libusb #820: libusb_alloc_streams returns LIBUSB_ERROR_INVALID_PARAM

ramuvsign, 2020-12-09:
Hi All,
When I try to open a stream using libusb_alloc_streams (the below snippet), I see LIBUSB_ERROR_INVALID_PARAM error returned.
was the endpoint address sent wrong? Any feedback and help to address this issue.

Snippet:

    devh = libusb_open_device_with_vid_pid(NULL, VENDOR_ID, PRODUCT_ID);

    ret = libusb_get_config_descriptor(libusb_get_device(devh), 0, &config);

const    struct libusb_endpoint_descriptor *endpoint = &config->interface[1].altsetting[0].endpoint[0];

     ret = libusb_alloc_streams(devh, 2, (unsigned char *)endpoint, 1);
     fprintf(stdout, "No 0f streams allocated %d endpoint address = %d \n", ret,endpoint->bEndpointAddress);
	 endpoint = &config->interface[1].altsetting[0].endpoint[1];
     ret = libusb_alloc_streams(devh, 2, (unsigned char *)endpoint, 1);
     fprintf(stdout, "No 0f streams allocated %d endpoint address = %d \n", ret,endpoint->bEndpointAddress);

**Output Logs with return value -2 :**

[ 0.009089] [0000185b] libusb: debug [libusb_alloc_streams] streams 2 eps 1
No 0f streams allocated -2 endpoint address = 130
[ 0.009108] [0000185b] libusb: debug [libusb_alloc_streams] streams 2 eps 1
No 0f streams allocated -2 endpoint address = 2

mcuee, 2020-12-10:
What is the device? What is the OS used? Do yo have the output of "lsusb -vvv" if you use Linux? If using other OS, you can use corresponding tool to dump the USB descriptors.

dickens, 2020-12-10:
You are not using the API correctly.  The `endpoint` parameter is an address to an array of `unsigned char` where each member contains an endpoint address.  You are passing the pointer to the `struct libusb_endpoint_descriptor`.  The correct usage is as follows:
```
unsigned char endpoint = config->interface[1].altsetting[0].endpoint[0].bEndpointAddress;
ret = libusb_alloc_streams(devh, 2, &endpoint, 1);
```

Or if you prefer an array, then:
```
unsigned char endpoint[1];
endpoint[0] = config->interface[1].altsetting[0].endpoint[0].bEndpointAddress;
ret = libusb_alloc_streams(devh, 2, endpoint, 1);
```

Since this is not a libusb issue, I'm going to close this.

closed 2020-12-10 by dickens