libusb/libusb #823: This is already fed up! Can't you fix this shit in so many years ?!

sevstels, 2020-12-12:
1>libusb.h(783): warning C4200: nonstandard extension used : zero-sized array in struct/union
1>libusb.h(808): warning C4200: nonstandard extension used : zero-sized array in struct/union
1>lib\libusb.h(968): warning C4510: 'libusb_version' : default constructor could not be generated
1>libusb.h(968): warning C4610: struct 'libusb_version' can never be instantiated - user defined constructor required
1>libusb.h(1280): warning C4200: nonstandard extension used : zero-sized array in struct/union
 Cannot generate copy-ctor or copy-assignment operator when UDT contains a zero-sized array

diabolo38, 2020-12-12:
certainly you should use another shity c++ compiler ;) 
 this is a msvc warn ,  so why don't you use winusb directly ? 

At least you should write in correct and not offending words if to get any sensible and intelligent anwers ... 

USB descriptor are unfortunately describe in that way,
they are not fixed sized and theirs not quite alternative to represent them  in a c/c++ smarter way

sevstels, 2020-12-12:
>> certainly you should use another shity c++ compiler ;)
As I understand it, this is the most qualified advice here ...

An insult is repeating this stupid mistake endlessly for many years. And don't try to fix it.

sevstels, 2020-12-12:
libusb\os\windows_winusb.c(3483): error C2146: syntax error : missing ')' before identifier 'PRIuPTR'
libusb\libusb\os\windows_winusb.c(3537): error C2146: syntax error : missing ')' before identifier 'PRIuPTR'

hjelmn, 2020-12-12:
This is not the appropriate way to act toward the developers of an all-volunteer project. You are lucky we even provide support for Windows much less the crappy Microsoft compiler. If you would like this fixed open a PR and one of us will take a look. Otherwise do not hold your breath waiting for anything. I have no interest in Windows support.

closed 2020-12-12 by hjelmn

sevstels, 2020-12-12:
In your answer, defeat is visible ... Windows machines are 90% of the desktop market today. Not to mention all commercial companies in the world, where the use of Windows is almost everywhere! And to ignore it is serious stupidity ...

hjelmn, 2020-12-12:
Except every company I have worked for use primarily Linux with Mac on the desktop. I don't care for your attitude and, again, if you want libusb to work with your broken Microsoft compiler open a PR, fix it locally, or just use WinUSB. I, and the other maintainers, will not waste time fixing what isn't broken. The errors are clearly erroneous as the lines in question are valid C.

```
#if defined(__STDC_VERSION__) && (__STDC_VERSION__ >= 199901L)
#define ZERO_SIZED_ARRAY                /* [] - valid C99 code */
#else
#define ZERO_SIZED_ARRAY        0       /* [0] - non-standard, but usually working code */
#endif /* __STDC_VERSION__ */
```

C99 allows a non-nested struct to end in a zero-sized array with empty braces. Before C99 gcc allowed it but required an explicit 0 size hence the conditional. BTW, libusb no longer supports C89 compilers so support for the old style is no longer needed. It is irrelevant as a support compiler sets ```__STDC_VERSION__``` to ```199901``` or higher.

And defeat? Jeebus. libusb only fairly recently (in its history) started supporting Windows. Its usb model is a mess and doesn't mesh well with user-space USB drivers. Linux, BSD, macOS, etc ALL are far better suited for this USB model.

hjelmn, 2020-12-12:
The user has been blocked for 7 days.

hjelmn, 2020-12-12:
As a reminder to all other users who may run across this issue. libusb is an open-source library maintained by an ever-changing team of volunteer developers. We strive to provide the best software we can with the time available. We welcome pull requests and constructive discussions. Discussions like this one do not help improve libusb and combative users may be blocked from posting or opening pull requests. In this case I am using a temporary ban but I, or any other maintainer, may ban a user indefinitely depending on the circumstances.