libusb/libusb #824: Error: assert (NULL == darwin_cached_devices.next). Regression from 1.0.23 (Commit: 1201bccf8)

cpagravel, 2020-12-13:
I'm using PyUSB on MacOS Catalina (up-to-date as of writing this) and I recently updated my libusb with brew. I found that the issue  this commit: https://github.com/libusb/libusb/commit/1201bccf857aa2c126f6f568f5255b0aa6027ac2#diff-ac52d89176c92629a8fdf7930595e6ec24298ca593f0fbb4942093626bd6a09f 

The program I'm using is a blackbox - I'm not sure how exactly PYusb is being used, I know it is ending abruptly due to the assert here: https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L584

I'm not very familiar with the libusb code. Can someone who is more familiar take another look at the commit I'm pointing to?

cpagravel, 2020-12-13:
Also, if I checkout that commit and build I can reproduce the issue with the following python code (using pyusb).
```
import faulthandler; faulthandler.enable()
import usb.core
usb.core.find()
usb.core.find()
usb.core.find()
```

This will fail on the second call to `usb.core.find()`

The output is:
```
(p3) @chrisgravel-macbookpro[~/dev/nest-harness-tests](u/chrisgravel/gqnq *<>) ✔ python testy.py 
Assertion failed: (NULL == darwin_cached_devices.next), function darwin_init, file os/darwin_usb.c, line 582.
Fatal Python error: Aborted

Current thread 0x0000000110785dc0 (most recent call first):
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 708 in __init__
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 945 in get_backend
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/core.py", line 1258 in find
  File "testy.py", line 32 in <module>
Abort trap: 6
```

cpagravel, 2020-12-13:
I can produce it consistently, but I also found that if I use the debugger it sometimes isn't reproduced.

```
(p3) @chrisgravel-macbookpro[~/dev/nest-harness-tests](u/chrisgravel/gqnq *<>) ✔ LIBUSB_DEBUG=4 python testy.py 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000004] [00000307] libusb: debug [libusb_init] created default context
[ 0.000023] [00000307] libusb: debug [libusb_init] libusb v1.0.23.11474
first init is: 1 
Assertion failed: (NULL == darwin_cached_devices.next), function darwin_init, file os/darwin_usb.c, line 585.
Fatal Python error: Aborted

Current thread 0x0000000106a9adc0 (most recent call first):
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 708 in __init__
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 945 in get_backend
  File "/Users/chrisgravel/.virtualenvs/p3/lib/python3.7/site-packages/usb/core.py", line 1258 in find
  File "testy.py", line 31 in <module>
Abort trap: 6
```

Anything I can do to provide more info?

dickens, 2020-12-13:
Look at the version in your debug output.  You aren't running the latest version.

cpagravel, 2020-12-13:
Ah, let me add some context. With my blackbox software, I can reproduce the issue 100% on 1.0.24, though, as I mentioned I noticed it sometimes does not occur if I use the debugger.

I did a git bisect to find the commit that seemed to cause the issue in libusb. On this particular commit, I can reproduce the issue without using the blackbox software. I realize there are 110 commits between this one and 1.0.24 and it's not necessarily the case that this commit is the issue since it could be a combination of things. I can say that I don't have an issue until this commit.

cpagravel, 2020-12-13:
I can get the logs from 1.0.24 in a moment.

cpagravel, 2020-12-13:
Logs for 1.0.24
```
(p3) @chrisgravel-macbookpro[~]( *) ✔ LIBUSB_DEBUG=4 python relay_board.py  
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000018] [006088be] libusb: debug [libusb_init] created default context
[ 0.000024] [006088be] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000185] [006088be] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.000191] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000206] [006088be] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.002094] [006088be] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002104] [006088be] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002107] [006088be] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.002109] [006088be] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002111] [006088be] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.002113] [006088be] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002115] [006088be] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.002119] [006088be] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.002121] [006088be] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8007
[ 0.002123] [006088be] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.002125] [006088be] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.007293] [006088be] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.007303] [006088be] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.007307] [006088be] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.007374] [006088be] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.007378] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.007393] [006088be] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x1000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.007396] [006088be] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.007432] [006088be] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.007436] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.007569] [006088be] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x14000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.007577] [006088be] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.007625] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3ac4a5fc
[ 0.007659] [006088be] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3ac4a5fc/0x80100000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.007663] [006088be] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3ac4a5fc
[ 0.008533] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.008709] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.008856] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.008991] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009123] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009135] [006088be] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.009175] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3b5cd0aa
[ 0.009199] [006088be] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3b5cd0aa/0x80200000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.009203] [006088be] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3b5cd0aa
[ 0.009466] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009641] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009885] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.010175] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.010345] [006088be] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.010358] [006088be] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.010402] [006088be] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3af61bfc
[ 0.010429] [006088be] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3af61bfc/0x80700000 against cached device with sessionID/locationID 0x0/0x80000000
[... cut ...]
  File "/Users/chrisgravel/dev/nest-engprod/nest_device_manager/nest_device/__init__.py", line 108 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "relay_board.py", line 14 in <module>
Abort trap: 6
```

dickens, 2020-12-13:
Interesting result.  The message "darwin event thread exiting" is triggered by the call to `CFRunLoopSourceSignal()` in `darwin_exit()`.  After the call to `pthread_join()`, we call `darwin_cleanup_devices()` which, if you look, clears the `darwin_cached_devices` list.  Could you add two debug messages and run it again?

1) In `darwin_cleanup_devices()` right before clearing `darwin_cached_devices`
2) In `darwin_init()` right before `assert (NULL == darwin_cached_devices.next);`.  In this one please print the value of `darwin_cached_devices.next`

cpagravel, 2020-12-13:
I want to add that I noticed the issue goes away if I manually call the init function twice.

E.g. in libusb1.py (re: https://github.com/pyusb/pyusb/blob/master/usb/backend/libusb1.py#L717)

If I change:
```
class _LibUSB(usb.backend.IBackend):
    @methodtrace(_logger)
    def __init__(self, lib):
        usb.backend.IBackend.__init__(self)
        self.lib = lib
        self.ctx = c_void_p()
        _check(self.lib.libusb_init(byref(self.ctx)))
```
into
```
class _LibUSB(usb.backend.IBackend):
    @methodtrace(_logger)
    def __init__(self, lib):
        usb.backend.IBackend.__init__(self)
        self.lib = lib
        self.ctx = c_void_p()
        _check(self.lib.libusb_init(byref(self.ctx)))
        _check(self.lib.libusb_init(byref(self.ctx)))
```

Will add the debug messages, one moment.

cpagravel, 2020-12-13:
Output after debug messages.

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000018] [00614afb] libusb: debug [libusb_init] created default context
[ 0.000023] [00614afb] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000026] [00614afb] libusb: debug [darwin_init] darwin_cached_devices.next: 0x0
[ 0.000167] [00614afb] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.000172] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000187] [00614afb] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.002099] [00614afb] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002108] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002110] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.002112] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002114] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.002116] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002118] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.002120] [00614afb] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.002121] [00614afb] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8007
[ 0.002125] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.002127] [00614afb] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.002184] [00614afb] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.002191] [00614afb] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002193] [00614afb] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002238] [00614afb] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.002242] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.002257] [00614afb] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x1000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.002260] [00614afb] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.002291] [00614afb] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.002294] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.002307] [00614afb] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x14000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.002310] [00614afb] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.002341] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3ac4a5fc
[ 0.002362] [00614afb] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3ac4a5fc/0x80100000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.002365] [00614afb] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3ac4a5fc
[ 0.003133] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003270] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003388] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003502] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003613] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003624] [00614afb] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.003655] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3b5cd0aa
[ 0.003677] [00614afb] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3b5cd0aa/0x80200000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.003680] [00614afb] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3b5cd0aa
[ 0.003783] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.003981] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.004171] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.004304] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.004421] [00614afb] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.004441] [00614afb] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.004509] [00614afb] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3af61bfc
[... cut ...]
  File "/Users/chrisgravel/dev/utility/reflection_utils.py", line 35 in identify_all_supported_classes
  File "/Users/chrisgravel/dev/__init__.py", line 108 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "relay_board.py", line 14 in <module>
Abort trap: 6
```

dickens, 2020-12-13:
Yup, does not make sense.  One more request please, change the end of `darwin_cleanup_devices()` to the following:

```
usbi_dbg("clearing darwin_cached_devices (%p) {%p,%p}", &darwin_cached_devices, darwin_cached_devices.prev, darwin_cached_device.next);
darwin_cached_devices.prev = darwin_cached_devices.next = NULL;
usbi_dbg("darwin_cached_devices after {%p,%p}", darwin_cached_devices.prev, darwin_cached_device.next);
```

cpagravel, 2020-12-13:
No problem. I'm happy to help.
```
(p3) @chrisgravel-macbookpro[~/dev/auxiliary_devices](u/chrisgravel/relay_board *) ✔ LIBUSB_DEBUG=4 python relay_board.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000015] [0062121f] libusb: debug [libusb_init] created default context
[ 0.000020] [0062121f] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000022] [0062121f] libusb: debug [darwin_init] darwin_cached_devices.next: 0x0
[ 0.000150] [0062121f] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.000155] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000169] [0062121f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.008677] [0062121f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.008710] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.008713] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.008715] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.008717] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.008719] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.008721] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.008722] [0062121f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.008724] [0062121f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8007
[ 0.008726] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.008746] [0062121f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.009832] [0062121f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.009849] [0062121f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.009851] [0062121f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.009983] [0062121f] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.009989] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.010006] [0062121f] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x1000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.010033] [0062121f] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.010114] [0062121f] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.010118] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.010331] [0062121f] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x14000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.010338] [0062121f] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.010735] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3ac4a5fc
[ 0.010760] [0062121f] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3ac4a5fc/0x80100000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.010763] [0062121f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3ac4a5fc
[ 0.012760] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.012972] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.013094] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.013246] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.013554] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.013598] [0062121f] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.013704] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3b5cd0aa
[ 0.013753] [0062121f] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x3b5cd0aa/0x80200000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.013756] [0062121f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x3b5cd0aa
[ 0.014030] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.014273] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.014393] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.014618] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.014891] [0062121f] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.014908] [0062121f] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.014946] [0062121f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x3af61bfc
[... cut ...]
  File "/Users/chrisgravel/dev/utility/reflection_utils.py", line 35 in identify_all_supported_classes
  File "/Users/chrisgravel/dev/__init__.py", line 108 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "relay_board.py", line 14 in <module>
Abort trap: 6
```

cpagravel, 2020-12-13:
Looks like it is getting changed in `darwin_destroy_device`

cpagravel, 2020-12-13:
```
[ 0.022023] [00636336] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.026830] [00636304] libusb: debug [darwin_cleanup_devices] clearing darwin_cached_devices (0x10bac6438) {0x7fba0051b780,0x7fb9f04a2c50}
[ 0.026846] [00636304] libusb: debug [darwin_cleanup_devices] darwin_cached_devices update4 {0x0,0x0}
[ 0.026871] [00636304] libusb: debug [libusb_unref_device] destroy device 20.7
[ 0.026878] [00636304] libusb: debug [libusb_unref_device] destroy device 20.4
[ 0.026882] [00636304] libusb: debug [libusb_unref_device] destroy device 20.1
[ 0.026886] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices before deref {0x0,0x0}
[ 0.026941] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after deref {0x7fba00528210,0x0}
[ 0.026948] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after nullify {0x7fba00528210,0x0}
[ 0.026952] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices before deref {0x7fba00528210,0x0}
[ 0.027000] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after deref {0x7fb9f04a2c50,0x0}
[ 0.027007] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after nullify {0x7fb9f04a2c50,0x0}
[ 0.027011] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices before deref {0x7fb9f04a2c50,0x0}
[ 0.027063] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after deref {0x10bac6438,0x10bac6438}
[ 0.027072] [00636304] libusb: debug [darwin_destroy_device] darwin_cached_devices after nullify {0x10bac6438,0x10bac6438}
[ 0.027113] [00636304] libusb: debug [libusb_init] created default context
[ 0.027122] [00636304] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.027127] [00636304] libusb: debug [darwin_init] darwin_cached_devices (0x10bac6438) {0x10bac6438,0x10bac6438}
Assertion failed: (NULL == darwin_cached_devices.next), function darwin_init, file os/darwin_usb.c, line 590.
Fatal Python error: Aborted
```

cpagravel, 2020-12-13:
I'm 90% certain the issue is this function:
```
static inline void list_del(struct list_head *entry)
{
	entry->next->prev = entry->prev;
	entry->prev->next = entry->next;
	entry->next = entry->prev = NULL;
}
```

hjelmn, 2020-12-13:
Looking at the commit that apparently caused the regression. dickens@ can you give some more context? The destructor use was intentional even if it left some stale resources. Don't remember looking at that commit when it went in (doesn't have my sign off). Was there a bug associated with the issue?

mcuee, 2020-12-13:
FYI no regression here on my Mac Mini M1.
```

mcuee@mcuees-Mac-mini ~ % python3
Python 3.9.1 (default, Dec 10 2020, 11:11:14) 
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import faulthandler; faulthandler.enable()
>>> import usb.core
>>> usb.core.find()
<DEVICE ID 1915:1025 on Bus 002 Address 012>
>>> usb.core.find()
<DEVICE ID 1915:1025 on Bus 002 Address 012>
>>> usb.core.find()
<DEVICE ID 1915:1025 on Bus 002 Address 012>
>>> usb.core.find()
<DEVICE ID 1915:1025 on Bus 002 Address 012>
>>> usb.core.find()
<DEVICE ID 1915:1025 on Bus 002 Address 012>
```

dickens, 2020-12-13:
> Looking at the commit that apparently caused the regression. @dickens can you give some more context? The destructor use was intentional even if it left some stale resources until exit. Don't remember looking at that commit when it went in (doesn't have my sign off). Was there a bug associated with the issue?

Deferring the resource cleanup until the destructor confuses memory analysis tools.  That is the primary motivation behind the change.  It's not exactly "nice" of libusb to hold onto heap allocations after you've told it to exit anyhow.

dickens, 2020-12-13:
Ok, finally figured this out.  The problem is that pyusb is holding references to libusb devices after calling `libusb_exit()`.  When `libusb_unref_device()` is called later, one or more of these devices are still pointing to `darwin_cached_devices` and that is how it is getting modified back to a non-NULL value.  A simple reproducer in C:

```
libusb_init(NULL);
libusb_device **list,  *dev;
if (libusb_get_device_list(NULL, &list) < 0)
	handle_error();
dev = libusb_ref_device(list[0]);
libusb_free_device_list(list, 1);
libusb_exit(NULL);
libusb_unref_device(dev);
libusb_init(NULL); // <- assertion failure here
```

While libusb might be able to handle this more gracefully, I think this is technically a bug in pyusb.  At the time you call `libusb_exit()`, there should be nothing going on with the library (e.g active transfers, open devices or even _referenced_ devices).  I don't think we've documented the last point anywhere, but if you think about it that makes sense.

I can enhance the library to spit out a warning message if a device is still referenced when `libusb_exit()` is called, but I don't think we should be going down the path of modifying backends to gracefully handle resource deallocations when they've already been instructed to exit.  I'll wait for input from the other maintainers before proceeding further.

mcuee, 2020-12-13:
@jonasmalacofilho 
Please help to take a look at pyusb side. Thanks.
https://github.com/pyusb/pyusb/issues/342

jonasmalacofilho, 2020-12-13:
@cpagravel

What version of PyUSB is that?  The traceback doesn't match the latest/1.1.0 release, and it already included a fix for something like what @dickens described to be the problem.

cpagravel, 2020-12-13:
My pyusb version was out of date. Using the latest release seems to handle the memory issue. Sorry for the confusion, I tried to update pyusb, but I didn't use the `--force` flag to force updating.

cpagravel, 2020-12-13:
+1 for adding a warning instead the asset. My experience was that after brew updated my libusb I saw the abort trap code and the code stopped functioning. I attempted to update my Python lib (forgot to use —force) and quickly assumed the issue was a regression in libusb. This will happen to other people who update libusb. Pip won’t bother updating packages in many cases, so inexperienced users (like myself) may have some frustration from this. I’ve spent a lot of time investigating before raising the issue here. I made an oversight; I may not be the only one who makes this oversight. Thanks for sparing time on this. +1 for removing the hard block and adding a warning.

dickens, 2020-12-13:
To clarify, I’m not proposing the removal of the assertion.  It is undefined behavior for devices to be accessed after `libusb_exit()` is called.  I am proposing to add a warning message in `libusb_exit()` if it detects that one or more device references are still held by the application, similar to how it warns when devices are still open.

dickens, 2020-12-17:
See commit 1a08aa84d96397a3840a75abe66051f5360c2c84.  I've made the error visible as part of this change.

hjelmn, 2020-12-17:
@dickens That commit needs work. We should never use the __atomic operations. They are defined in gcc and clang, yes, but they are not standard. libusb should be using c11 atomics ```#include <stdatomic.h>``` only.

hjelmn, 2020-12-17:
Did anyone review that commit?

hjelmn, 2020-12-17:
Fixing that now. Please get a review next time :). We require C11 so stdatomic is available.

dickens, 2020-12-17:
> @dickens That commit needs work. We should never use the __atomic operations. They are defined in gcc and clang, yes, but they are not standard. libusb should be using c11 atomics `#include <stdatomic.h>` only.

While we require C11, we have not settled on requiring the C11 headers.  See [your comment on another C11-related change](https://github.com/libusb/libusb/pull/682#issuecomment-578595193).  C11 compiler support and C11 header support are not the same thing.  The `_Thread_local` identifier is not standard either, so to avoid dependence on the C11 headers I followed the exact same approach.

dickens, 2020-12-17:
> Did anyone review that commit?

Well, no _human_ reviewed the commit but a number of CI builds did.  We are a little short on active maintainers at the moment :wink:

hjelmn, 2020-12-17:
> While we require C11, we have not settled on requiring the C11 headers. See [your comment on another C11-related change](https://github.com/libusb/libusb/pull/682#issuecomment-578595193). C11 [...]

_Thread_local is standard (see  ISO/IEC 9899:201x 6.4.1). WG14 has taken all identifiers of the form ```_[A-Z].* and __.*```. That is the reason bool is ```_Bool``` not bool unless you include stdbool.h. See ISO/IEC 9899:201x section 7.1.3:

```
All identifiers that begin with an underscore and either an uppercase letter or another underscore are always reserved for any use.
```

gcc took some liberties taking ```__atomic``` from what I can tell. But they were following the convention set by Intel with ```__sync``` (why is Intel always involved when something is done wrong).

As for the header issue. We decided to avoid requiring threads.h because we didn't see a benefit to using the thread_local alias. For atomics, however, we could use _Atomic long and just add, subtract, etc and they would automatically be atomic. The calls are almost irrelevant but since they are in stdatomic.h and that is in gcc 4.9+ we should use it.

hjelmn, 2020-12-17:
> > Did anyone review that commit?
> 
> Well, no _human_ reviewed the commit but a number of CI builds did. We are a little short on active maintainers at the moment 😉

Too true. I am around though for reviewing darwin/macOS changes. At this point the only active maintainer from libusb 0.1 on. Feel free to ping me for a review by opening a PR and tagging me.

dickens, 2020-12-22:
Where do we want to leave this?  It is not a libusb issue and the library will now emit a warning message for each device with leaked references during `libusb_exit()`.  I think we can close this.

mcuee, 2020-12-23:
I agree. Close this.

closed 2020-12-23 by mcuee