libusb/libusb #825: [regression 1.0.24] can't communicate with iPhone/iPad on Linux

corsac-s, 2020-12-13:
Hi,
this bug was originally reported to Debian (https://bugs.debian.org/977231) but I've reproduced and bisected using libusb repository so I'm reporting directly as well.

To summarize the downstream bug: communication with iPhone/iPad don't work with 1.0.24: the device appear in /sys/kernel/debug/usb/devices but doesn't show in lsusb, and the userland stack linked against libusb (ideviceutils, usbmuxd etc.) doesn't see any device either.

I've bisected between 1.0.22 and 1.0.24 and here's the result:

```e2be556bd262cd260bebe3fc724509692537eba3 is the first bad commit
commit e2be556bd262cd260bebe3fc724509692537eba3
Author: Chris Dickens <christopher.a.dickens@gmail.com>
Date:   Wed Apr 29 12:39:35 2020 -0700

    linux_usbfs: Parse config descriptors during device initialization
    
    Do the work ahead of time and cache the results so that fetching config
    descriptors becomes a trivial operation.
    
    Signed-off-by: Chris Dickens <christopher.a.dickens@gmail.com>

 libusb/os/linux_usbfs.c | 263 ++++++++++++++++++++++++++----------------------
 libusb/version_nano.h   |   2 +-
 2 files changed, 145 insertions(+), 120 deletions(-)
bisect run success
```
And here's the bisect log:
```
git bisect start
# bad: [c6a35c56016ea2ab2f19115d2ea1e85e0edae155] libusb 1.0.24
git bisect bad c6a35c56016ea2ab2f19115d2ea1e85e0edae155
# good: [0034b2afdcdb1614e78edaa2a9e22d5936aeae5d] libusb 1.0.22
git bisect good 0034b2afdcdb1614e78edaa2a9e22d5936aeae5d
# good: [48200f2cc23633899270e7c666ac7e948ead8df1] core: Fix build on linux
git bisect good 48200f2cc23633899270e7c666ac7e948ead8df1
# bad: [26611eaa494ed9e077b5b0e1f999f5ae377de958] Windows: Translate ERROR_NO_SUCH_DEVICE to LIBUSB_TRANSFER_NO_DEVICE
git bisect bad 26611eaa494ed9e077b5b0e1f999f5ae377de958
# good: [09d0312fbb466ce2457df94e314ba1348039138d] Xcode: Update project file
git bisect good 09d0312fbb466ce2457df94e314ba1348039138d
# good: [71a3fb3faadb79c2ccff8f23f0520e4015f547f3] Android: fixes unknown warning option from ndk build
git bisect good 71a3fb3faadb79c2ccff8f23f0520e4015f547f3
# good: [e873677b9196b191d6cdbdf9783c6d6a18379249] descriptor: Minor improvements to the parse_descriptor() function
git bisect good e873677b9196b191d6cdbdf9783c6d6a18379249
# good: [8476804b283cfed82aec5e45301b51be3ad68e99] descriptor: Remove usbi_get_config_index_by_value()
git bisect good 8476804b283cfed82aec5e45301b51be3ad68e99
# good: [14a302a2f55cb2e619158854f94845f2ca2c8214] sunos: Fix a number of compiler warnings
git bisect good 14a302a2f55cb2e619158854f94845f2ca2c8214
# bad: [e2be556bd262cd260bebe3fc724509692537eba3] linux_usbfs: Parse config descriptors during device initialization
git bisect bad e2be556bd262cd260bebe3fc724509692537eba3
# good: [e9eec3a680cad3b2c9c5213fb7d60148cf6900da] core: Narrow the types passed to certain backend functions
git bisect good e9eec3a680cad3b2c9c5213fb7d60148cf6900da
# first bad commit: [e2be556bd262cd260bebe3fc724509692537eba3] linux_usbfs: Parse config descriptors during device initialization
```
I couldn't just revert the offending commit. I guess you might not have an iDevice ready to test, so if you need me to do stuff don't hesitate to ask.

aurel32, 2020-12-13:
I would add that the following error is reported when running lsusb with `LIBUSB_DEBUG=99`:

```
warning [parse_config_descriptors] config length mismatch wTotalLength 39 real 30
error [parse_config_descriptors] descriptor is not a config desc (type 0x00)
debug [libusb_unref_device] destroy device 1.11

```

sbraz, 2020-12-13:
Hi,
I'm seeing the same thing with an Eaton UPS. I can confirm that e2be556bd262cd260bebe3fc724509692537eba3 is the first bad commit.
Output from  `LIBUSB_DEBUG=99 lsusb`:
```
[ 0.002177] [00003888] libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
[ 0.002184] [00003888] libusb: warning [parse_config_descriptors] config length mismatch wTotalLength 34 real 25
[ 0.002204] [00003888] libusb: error [parse_config_descriptors] descriptor is not a config desc (type 0x03)
[ 0.002205] [00003888] libusb: debug [libusb_unref_device] destroy device 1.5
```

From `usb-devices`:
```
T:  Bus=01 Lev=01 Prnt=01 Port=10 Cnt=01 Dev#=  5 Spd=1.5 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  6
P:  Vendor=0463 ProdID=ffff Rev=01.00
S:  Manufacturer=EATON
S:  Product=Ellipse ECO
S:  SerialNumber=000000000
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=20mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```

dickens, 2020-12-13:
The device’s descriptors appear broken.  The commit in question brings this to light during device enumeration rather than when descriptors are asked for.

Can you run `LIBUSB_DEBUG=4 lsusb -v` using the older version of libusb and provide the output?  It would also be nice to see the binary dump of the descriptors, which can be captured by running `cat /dev/bus/usb/<bus>/<address> > descriptors.bin` where \<bus> and \<address> are the numbers reported by lsusb.

corsac-s, 2020-12-13:
Here is the requested information.

[libusb.txt](https://github.com/libusb/libusb/files/5685256/libusb.txt)
[descriptor.txt](https://github.com/libusb/libusb/files/5685269/descriptor.txt)
[descriptor.bin.txt](https://github.com/libusb/libusb/files/5685271/descriptor.bin.txt)

sbraz, 2020-12-13:
Here's [lsusb's output](https://dpaste.com/67GLJDREU) and [the descriptor's dump](https://github.com/libusb/libusb/files/5685575/descriptor.bin.txt).

dickens, 2020-12-13:
Found and fixed.  Thanks for the report and provided data.

closed 2020-12-13 by dickens

corsac-s, 2020-12-14:
Thanks a lot, I can confirm it works (and @aurel32 already uploaded the fix to Debian).h

mcuee, 2021-01-15:
Just copy the root cause for the fix here.

Commit e2be556 ("linux_usbfs: Parse config descriptors during device
initialization") introduced a regression for devices with multiple
configurations. The logic that verifies the reported length of the
configuration descriptors failed to count the length of the
configuration descriptor itself and would truncate the actual length by
9 bytes, leading to a parsing error for subsequent descriptors.

kasperk81, 2021-05-30:
Glad to know this issue is fixed and it's working!
Can we get a new release so distro packages are updated in this quarter?

hjelmn, 2021-05-30:
That is the plan. If the process is not started by the end of the week I intend to cut an RC on Friday.

kasperk81, 2021-05-30:
Issue is the released version for past six months is completely broken due to this bug (can't even do the initial handshake). Would be nice to backport this fix to release version (patch version bump), so RC can take its time. Distros do not jump to RC normally.

hjelmn, 2021-06-01:
@kasperk81 The distros tend to do the work themselves when it comes to back porting patches. Not sure any of us has the time to get a 1.0.24.1 out. I will see what I can do to accelerate the 1.0.25 timeline.

mcuee, 2021-06-01:
Backport should be done by distros normally, as far as I know. We do not do backport in this project.

corsac-s, 2021-06-01:
Speaking for Debian, I try to do some backporting, but it would really help to have a more frequent release cycle.

kasperk81, 2021-06-01:
i also submitted a patch to alpine linux yesterday and it's merged. agree with @corsac-s on frequency.. especially when critical bug is fixed just four days after the release.

mcuee, 2021-06-01:
Unfortunately libusb is lacking active maintainer resources, even though it can be said to be a critical project for Linux distros. 

Real jobs are mainly done by Chris (mainly Windows and Linux) and Nathan (mainly macOS) now.

mcuee, 2021-06-01:
> That is the plan. If the process is not started by the end of the week I intend to cut an RC on Friday.

That will be great. As far as I am concerned, we only need to merge #856 (which also fixed #858) to have the RC.

hjelmn, 2021-06-03:
@mcuee Done. I think we are ready for an RC now.

mcuee, 2021-06-03:
@hjelmn Great. Yes we are ready for RC.

commit f6d2cb56 (2020-12-13): linux_usbfs: Fix parsing of descriptors for multi-configuration devices

Commit e2be556bd2 ("linux_usbfs: Parse config descriptors during device
initialization") introduced a regression for devices with multiple
configurations. The logic that verifies the reported length of the
configuration descriptors failed to count the length of the
configuration descriptor itself and would truncate the actual length by
9 bytes, leading to a parsing error for subsequent descriptors.

Closes #825