libusb/libusb #826: Did find the .dylib file, but wrong architecture

MonaNekoWR, 2020-12-13:
It's just a python code as follows:

```
import ctypes
ctypes.cdll.LoadLibrary(find('libusb-1.0')) 
```
OSError: dlopen(/usr/local/lib/libusb-1.0.dylib, 6): no suitable image found.  Did find:
        /usr/local/lib/libusb-1.0.dylib: mach-o, but wrong architecture
        /usr/local/lib/libusb-1.0.0.dylib: mach-o, but wrong architecture

This .dylib file is built with MacOS Big Sur M1
It's also showed in Terminal that
```
(base) ... ~ % cd /usr/local/lib
(base) ... % lipo -info libusb-1.0.0.dylib
Non-fat file: libusb-1.0.0.dylib is architecture: arm64
```
I have no idea about what's wrong with the .dylib file...
Could you give me some advice about how to deal with this problem.

LudovicRousseau, 2020-12-13:
I guess you need to rebuild libusb for x86_64 in addition to arm64.
This is NOT an issue in libusb.

closed 2020-12-13 by LudovicRousseau

hjelmn, 2020-12-13:
Yup. Just rebuild. libusb works perfectly on the M1.

MonaNekoWR, 2020-12-14:
> I guess you need to rebuild libusb for x86_64 in addition to arm64.
> This is NOT an issue in libusb.

That dose work. Thank you very much for replying so fast.
For our beginner, I think it's a good ideal to add some description about HOW TO INSTALL LIBUSB IN MacOS M1.

Just like:
1. Install Xcode Ver12.2
2.Clone the newest libusb from git
3.Open the Xcode project from /libusb-1.0.24/Xcode/libusb.xcodeproj
4.Choose the build option to "Any Mac" which means both ARM64 and X86_64
5.Edit the Run Scheme to Release and then build the project.
6.Show the "libusb-1.0.0.dylib" in Finder then copy it to /usr/local/lib and copy the "libusb.h" to /usr/local/include/libusb-1.0

Thank you again for solving my problem.

MonaNekoWR, 2020-12-14:
> Yup. Just rebuild. libusb works perfectly on the M1.

Actually I just spent lot of time fighinting with the "./configure make make install" in Terminal
Consider why pip install libusb doesn't work.
Try to debug the source code of pyusb step by step but find the NoBackendError
Also waste time trying to build a ARM64 dll file with Visual Studio in Windows OS.

However, I learnt a lot about how does the unix system work.
And finally, every thing works well! Thank you very much! @LudovicRousseau @hjelmn

LudovicRousseau, 2020-12-14:
You can add a page on the project Wiki https://github.com/libusb/libusb/wiki.
But I guess most macOS developers will know what to do to handle/generate fat binaries.

MonaNekoWR, 2020-12-14:
[Does libusb support MacBook Pro (Apple Sillicon, M1, 2020)](https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_MacBook_Pro_Apple_Sillicon_M1_2020)

Anyway, I added it to FAQ. Maybe it just helps the user who use pylib.

@LudovicRousseau

hjelmn, 2020-12-14:
I will update the xcode project and add a configure option for building universal.