libusb/libusb #827: Cannot open include file unistd.h and List.h

machmar, 2020-12-13:
So I finally got the demo project to compile without issues. I ended up including all the files from libusb-1.0.24/libusb because I was lost for over two days.(this is downloaded from your website) But now it tells me `Cannot open include file 'unistd.h': No such file or directory` and the file is in `haiku_usb_backend.cpp`. It also can't find `List.h`. 

I have tried starting the project over and including the library like it said in the `README.txt`. But then I get errors like `unresolved external symbol _libusb_set_debug@8 referenced in function _main` and more. (this is downloaded from your GitHub releases)

I should also say I'm attempting to use this with Visual Studio 2019 and that I have no Idea what I'm doing as I have ever only coded for AVR chips.

dickens, 2020-12-13:
It sounds as though your project's configuration is not correct.  The missing files are because you are compiling the wrong files (we already provide Visual Studio solutions) and the unresolved symbols are because you aren't linking correctly against the library.  

I'm going to close this issue as this is for reporting problems with libusb.  Please use the libusb-devel mailing list for support questions.

closed 2020-12-13 by dickens

machmar, 2020-12-14:
Oh okay. Thanks anyway.