libusb/libusb #829: MinGW64 release dll depends on MSYS2

bluecmd, 2020-12-20:
Hi,

I am trying to use MinGW64/dll/libusb-1.0.dll from libusb-1.0.24. Sadly it seems to rely on MSYS2 for some reason.
The 32-bit does not have this dependency.

```
$ ldd MinGW64/dll/libusb-1.0.dll
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffc8d950000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffc8c870000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffc8b420000)
        msys-2.0.dll => /usr/bin/msys-2.0.dll (0x180040000)
        advapi32.dll => /c/WINDOWS/System32/advapi32.dll (0x7ffc8c2e0000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffc8caf0000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ffc8c960000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffc8c740000)
        CRYPTBASE.DLL => /c/WINDOWS/SYSTEM32/CRYPTBASE.DLL (0x7ffc8a930000)
        bcryptPrimitives.dll => /c/WINDOWS/System32/bcryptPrimitives.dll (0x7ffc8b1f0000)

$ ldd MinGW32/dll/libusb-1.0.dll
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffc8d950000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffc8c870000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffc8b420000)
        libusb-1.0.dll => /c/openocd/libusb/MinGW32/dll/libusb-1.0.dll (0x6b800000)
```

diabolo38, 2020-12-20:
I suspect that is because you build it under msys-2
i build with and old msys (not 2) and minwgw64 8.2 and i have nothing such 
it just depends on the 3 standrad os dll 
KERNEL32.DLL
MSVCRT.DLL
USER32.DLL

bluecmd, 2020-12-20:
I didn't build it, I used the release hosted/provided here on GitHub - sorry if that wasn't clear :-).

The bug is about the release artifact/7z file.

mcuee, 2020-12-21:
The Windows binary is more for convenience and it may not work for all use cases, this is especially true for the static library, but it can also happen for the DLL.

In this case, what is your use case here? You may want to build the binary by yourself to see if you can get rid of the MSYS2 dependencies.

I use MSYS2 to build libftdi binaries and I always have the MSYS 2 dependencies for the binaries.
https://sourceforge.net/projects/picusb/files/

bluecmd, 2020-12-21:
Indeed, they are very convenient. I want to build an 64-bit version of OpenOCD without any big dependencies. I can successfully build a 32-bit version, but since the 64-bit libusb depends on MSYS2 the final output of course does that as welll.

I can of course build it myself, but what's the point of providing a "mingw64" dll that's dependent on MSYS2? In my eyes that makes little sense, hence this bug report.

diabolo38, 2020-12-21:
oops sorry didn't seen you took the bin 
I found the same  dependencies issue for dll  personally rebuilt on msys2 it depend on msys2.dll 👎 
i also don't like much such dependencies and the pain managing and distributing thems if it can be avoided 
that is a reason i use old msys (small/portbale offlline  instalabe) for most windows work ,  this  issues is not encouraging me toward msys2 move :(

dickens, 2020-12-21:
I've rebuilt the 64-bit MinGW DLL and uploaded a new 7z archive.  The MSYS2 dependency should be gone.

closed 2020-12-21 by dickens