libusb/libusb #831: [Regression 1.0.24] Device is not enumerated on Linux

Lekensteyn, 2020-12-26:
After upgrading from 1.0.23-3 to 1.0.24-1 on Arch Linux, `lsusb` fails to show an iPhone 12 Pro. As usbmuxd depends on libusb for enumeration, that also failed to find any devices. git bisect pointed to:

e2be556bd262cd260bebe3fc724509692537eba3 is the first bad commit

```
commit e2be556bd262cd260bebe3fc724509692537eba3
Author: Chris Dickens <christopher.a.dickens@gmail.com>
Date:   Wed Apr 29 12:39:35 2020 -0700

    linux_usbfs: Parse config descriptors during device initialization
    
    Do the work ahead of time and cache the results so that fetching config
    descriptors becomes a trivial operation.
    
    Signed-off-by: Chris Dickens <christopher.a.dickens@gmail.com>

 libusb/os/linux_usbfs.c | 263 ++++++++++++++++++++++++++----------------------
 libusb/version_nano.h   |   2 +-
 2 files changed, 145 insertions(+), 120 deletions(-)
```

Automated the bisect with `git bisect run ./bisect.sh` and
<details><summary>bisect.sh</summary>

```
#!/bin/sh
git describe --tags
autoreconf -ivf || exit 128
mkdir -p build
cd build
../configure && make -j10 || exit 128

LD_LIBRARY_PATH=libusb/.libs ldd /bin/lsusb | grep -F .libs || { echo "Bad!"; exit 128; }
out=$(LD_LIBRARY_PATH=libusb/.libs lsusb)
echo "$out"
echo .
echo "$out" | grep Apple
```
</details>

Further info:
- `lsusb -d05ac: -v` output: [lsusb-d05ac-v.txt](https://github.com/libusb/libusb/files/5743509/lsusb-d05ac-v.txt)
- Issue is still present on git master (v1.0.24-7-g073f229fd11f).
- Reproduced on both Linux 5.9.13 and 5.10.2.

dickens, 2020-12-26:
This was fixed in f6d2cb561402c3b6d3627c0eb89e009b503d9067.  Refer to #825

closed 2020-12-26 by dickens

Lekensteyn, 2020-12-27:
Thank you! I could confirm that the issue was fixed on git master, but after switching branches I accidentally forgot to rebuild libusb.

Happy holidays! :tada: