libusb/libusb #836: darwin: clock_gettime is used even if deployment target is <10.12, leads to crash

parafin, 2021-01-02:
Commit 26b16eb65a61b37b64ce8962b5b7d927c3c54a7d introduces this issue by switching from OSX_USE_CLOCK_GETTIME, which is defined using `#if MAC_OS_X_VERSION_MIN_REQUIRED >= 101200`, to HAVE_CLOCK_GETTIME, which is generated by ./configure without taking into account the current deployment target. Only config.h for Xcode keeps the old logic, but it's not used when building from console, e.g. by macports. The result is that when one builds on macOS >= 10.12 but targets versions < 10.12, the resulting binary won't work on older OSes with the following error message:
```
Exception Type: EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000

Application Specific Information:
dyld: in dlsym()

Dyld Error Message:
Symbol not found: _clock_gettime
Referenced from: /Applications/darktable.app/Contents/MacOS/../Resources/lib/libusb-1.0.0.dylib
Expected in: /usr/lib/libSystem.B.dylib
```
I'm not sure how to better solve this, otherwise I would have proposed a PR.

dickens, 2021-01-15:
Since autoconf is unaware of deployment targets, you have to take care of it manually.  You can invoke the configure script as `./configure ac_cv_func_clock_gettime=no` to force it off.

The other possibility is to complicate the configure script to test the deployment target instead of checking for `clock_gettime()` on Mac.

parafin, 2021-01-15:
Second option sounds better to me, since first one is at best can be called a workaround - it's still a regression compared to how it worked before.

hjelmn, 2021-01-15:
I would prefer not to complicate the script since Apple ended support for 10.11.x in 2019. It was released back in 2015. I am not opposed to the check if it could be kept simple.

@parafin How are you setting the deployment target when building? configure should have correctly detected that is not available. Are you using the 10.11 SDK?

parafin, 2021-01-15:
Apple stopped shipping older SDKs at some point with Xcode. I'm using Xcode 11.3.1 which includes SDK 10.15.2. I'm building with macports and deployment target is set in macports.conf. I'm not sure how it's exactly handled. configure doesn't detect that it's missing, since it is present in SDK and on the host OS. This situation (newer SDK and older deployment target) usually results in weak linking I think.

I don't think you're considering stopping support for macOS <= 10.11, since otherwise you would have just removed the alternative implementation for clock_gettime altogether. But without fixing this bug you're basically breaking the only reasonable way to cross-build applications for those versions. Newer Xcode versions might be required for some compiler features, so downgrading it is not a good option.

An option to pass ac_cv_func_clock_gettime=no to configure script is a bad one, since the list might grow in the future. This stuff should just work, without shifting maintenance cost to every user of the library (for whom it matter of course). Moreover it did work fine, and also some other macOS features are correctly guarded with MAC_OS_X_VERSION_MIN_REQUIRED preprocessor checks. So please consider fixing this bug properly. I think checking the deployment target when testing for clock_gettime is simple and already proven as reliable. So why not go this way?

hjelmn, 2021-01-15:
You can get the SDK from an older version of Xcode. I do from time to time to check with different SDKs. For 10.11 download Xcode 7 or Xcode 8 (earlier than 8.3.3). Right-click on the old Xcode and select "Show package contents". The SDK is in Contents/Developer/Platforms/MacOSX.platform/Developers/SDKs. You can put this SDK in the same place in the current Xcode version and it will be detected.

And, I don't think of this as a bug. The natural course of software development will, by definition, naturally stop support for older software as developers no longer have access to those environments. Older versions of libusb will continue to work on these older systems. That said, this one does seem to be an unfortunate side-effect of code cleanup so it makes sense to either 1) see why clock_gettime presence is not being detected properly on your platform, or 2) see if a simple change would address it.

Looking at /usr/include/time.h it does protect clock_gettime with:

```
#if !defined(_DARWIN_FEATURE_CLOCK_GETTIME) || _DARWIN_FEATURE_CLOCK_GETTIME != 0
#if __DARWIN_C_LEVEL >= 199309L
#if __has_feature(enumerator_attributes)
#define __CLOCK_AVAILABILITY __OSX_AVAILABLE(10.12) __IOS_AVAILABLE(10.0) __TVOS_AVAILABLE(10.0) __WATCHOS_AVAILABLE(3.0)
#else
#define __CLOCK_AVAILABILITY
#endif

__CLOCK_AVAILABILITY
int clock_gettime(clockid_t __clock_id, struct timespec *__tp);
```

So I think configure is not being run with the proper deployment target.

hjelmn, 2021-01-15:
Oh, thats fun. ```-mmacosx-version-min=10.11``` is not preventing ```clock_gettime``` detection. That isn't right. Digging into it now.

Using the 10.11 SDK would work without this piece working (which should be).

parafin, 2021-01-15:
This is a bug if libusb supports macOS < 10.12. And it seems to try to support it, since there is code for older macOS versions. Some people are stuck with older macOS versions because they run it on old hardware, so it would be a pity for them to stop getting updates of the software they use. Of course it will inevitably happen eventually, but I hope not because of such trivial and already understood problem like this one. Anyway, enough about that, I don't think anyone opposes much to fixing this.

Using older SDK isn't actually a viable option either, because it will make it impossible to use newer API using weak linking (and proper checks that called functions are not null).

`-mmacosx-version-min=10.11` enables weak linking for functions appeared after 10.11, it's MAC_OS_X_VERSION_MAX_ALLOWED who forbids their usage.

parafin, 2021-01-15:
So have about adding this to configure check:
```
#ifdef __APPLE__
#define MAC_OS_X_VERSION_MAX_ALLOWED MAC_OS_X_VERSION_MIN_REQUIRED
#endif
```
before any includes?

parafin, 2021-01-15:
Hmm, actually I can't get MAC_OS_X_VERSION_MAX_ALLOWED (or __MAC_OS_X_VERSION_MAX_ALLOWED) to work. I now remembered that I tried in the past and similarly failed.

hjelmn, 2021-01-15:
@parafin Same here. This is weird. I swear just the ```-mmacosx-version-min=``` used to work for guarding against new features.

hjelmn, 2021-01-15:
@dickens Has a the necessary changes. Looks like there is no choice to go the complicated route. At some point we will probably clean out that check but it should get you running.

hjelmn, 2021-01-15:
And, FWIW, I need to do some backend cleanup. I think there is still some really old macOS stuff in there.

dickens, 2021-01-15:
How about the solution in dickens@c032a10?  If there becomes a future need we can refactor to create a more generic version check.  Sucks that the toolchain and headers can't figure this out for us.

hjelmn, 2021-01-15:
@dickens Agreed. Thought those macOS version guards would work here. Maybe they did at one point.

I usually just used -isysroot to force a version but it has been some time since I bothered with that. So, maybe it never worked as expected :).

parafin, 2021-01-15:
> How about the solution in [dickens/libusb@c032a10](https://github.com/dickens/libusb/commit/c032a10)?

I've checked this solution and it works, thanx!

> Sucks that the toolchain and headers can't figure this out for us.

Amen to that, Apple toolchain sucks.

> And, FWIW, I need to do some backend cleanup. I think there is still some really old macOS stuff in there.

Just FYI, I'm currently targeting macOS 10.7 for [darktable](https://github.com/darktable-org/darktable/) builds. I don't remember when I've last seen a report from a user staying on that version, but 10.8 I've seen recently (reporting this bug actually).

commit 08e0029f (2021-01-15): configure.ac: Consider Mac OS X target version for clock_gettime()

The AC_CHECK_FUNCS() macro checks whether a program compiled on the host
will link against the function, but it does not help when targeting
older systems where that function is not available. Additionally, the
way that autoconf compiles the program ignores any attributes or
preprocessor guards that installed headers might have applied to that
function.

Resolve this by verifying that the OS X target version is 10.12 or newer
before checking for clock_gettime().

Closes #836
closed 2021-01-15 by dickens