libusb/libusb #837: Problems sending to and receiving data from the IC-R8600

righthalfplane, 2021-01-05:
Something strange going on - the idVendor vendor comes up as 0x0c26, but the iProduct comes up as 0x0023 on Windows and as 0x0022 on macOS and ubuntu 18.04. I wrote a little program to set one of the  IC-R8600 modes - It works perfectly on Windows 7 and Windows 10, but it errors on the MacOS High Sierra and Ubuntu 18.04 . If just set the iProduct to 0x0022, I get the "MacOS - output 1". If just set the iProduct to 0x0022 and also do the libusb_set_interface_alt_setting also, I get the "MacOS - output 2" - it just times output, but everything else looks Ok.

`
#include <stdio.h>
#include <stdlib.h>
#include <libusb-1.0/libusb.h>

// cc -o usbTry3 usbTry3.c -lusb-1.0

int main()
{
	libusb_context *ctx;
	libusb_device_handle *handle;

	libusb_init(&ctx);
	//libusb_set_debug(ctx, 3);
	libusb_set_option(ctx, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG);
	handle = libusb_open_device_with_vid_pid(ctx, 0x0c26, 0x0023);
	//handle = libusb_open_device_with_vid_pid(ctx, 0x0c26, 0x0022);
	if(handle == NULL){printf("libusb_open_device_with_vid_pid failed\n");exit(1);}
	//libusb_set_configuration(handle, 1);
	
	int e=libusb_claim_interface(handle,0);
	if(e != 0){printf("libusb_claim_interface error %d\n",e);return 1;}

	//e=libusb_set_interface_alt_setting(handle,0,1);
	//if(e != 0){printf("libusb_set_interface_alt_setting error %d\n",e);return 1;}

	{
		unsigned char cmd[] = { 0xFE, 0xFE, 0x96, 0xE0,  0x1A, 0x13, 0x00, 0x01,  0xFD, 0xFF };
		
		int transferred;
		transferred=0;
		e=libusb_bulk_transfer(handle, 0X02,cmd,(int)sizeof(cmd), &transferred,2000);
	    if(e != 0)printf("libusb_bulk_transfer1 %d\n",e);
	    printf(" transferred %d\n",transferred);
		transferred=0;
		e=libusb_bulk_transfer(handle, 0X88,cmd,(int)sizeof(cmd), &transferred,2000);
	    if(e != 0)printf("libusb_bulk_transfer2 %d\n",e);
	    printf(" transferred %d\n",transferred);

	}

	libusb_release_interface(handle, 0);
	libusb_close(handle);
	libusb_exit(ctx);
	return 0;
}
`

Windows output -
`
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.003507] [00000a34] libusb: debug [libusb_get_device_list]
[ 0.007991] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.008121] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.008317] [00000a34] libusb: debug [get_api_type] driver(s): RtlWlanu_OldIC
[ 0.008419] [00000a34] libusb: debug [get_api_type] upper filter driver(s): vwifibus
[ 0.008515] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [18]
[ 0.008660] [00000a34] libusb: debug [get_api_type] driver(s): WinUsb
[ 0.008727] [00000a34] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.008772] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [2D]
[ 0.008906] [00000a34] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.008978] [00000a34] libusb: debug [get_api_type] matched driver name against HID API
[ 0.009023] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [40]
[ 0.009104] [00000a34] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.009137] [00000a34] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.009154] [00000a34] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[ 0.009245] [00000a34] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_10DE&DEV_1ADA&SUBSYS_C7521462&REV_A1\4&37732CF4&0&0208' bus number 1
[... 200 lines cut ...]
libusb_bulk_transfer2 -7
 transferred 0
[ 4.103269] [00000307] libusb: debug [libusb_release_interface] interface 0
[ 4.105326] [00000307] libusb: debug [libusb_close] 
[ 4.105366] [00000307] libusb: debug [libusb_exit] 
[ 4.105371] [00000307] libusb: debug [libusb_exit] destroying default context
[ 4.105374] [00000307] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 4.105377] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.105380] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 0ms
[ 4.105388] [00000307] libusb: debug [handle_events] poll() returned 0
[ 4.105391] [00000307] libusb: debug [libusb_unref_device] destroy device 0.8
[ 4.105395] [00000307] libusb: debug [libusb_unref_device] destroy device 0.4
[ 4.105398] [00000307] libusb: debug [libusb_unref_device] destroy device 0.3
[ 4.105401] [00000307] libusb: debug [libusb_unref_device] destroy device 0.2
[ 4.105405] [00000307] libusb: debug [libusb_unref_device] destroy device 0.1
[ 4.105408] [00000307] libusb: debug [libusb_unref_device] destroy device 29.4
[ 4.105411] [00000307] libusb: debug [libusb_unref_device] destroy device 29.2
[ 4.105414] [00000307] libusb: debug [libusb_unref_device] destroy device 29.1
[ 4.105417] [00000307] libusb: debug [libusb_unref_device] destroy device 29.0
[ 4.105420] [00000307] libusb: debug [libusb_unref_device] destroy device 29.0
[ 4.105423] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 4.105483] [00001507] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[dir:SdrGlut-780/misc/usb-example] dir% 

`

mcuee, 2021-01-05:
Please post the usb descriptor of the device (similar to "lsusb -vvv" under Linux). 
You can get "lsusb" formular from Homebrew for macOS which produce similar output to Linux "lsusb -vvv". 
Or you can post the output of USBView or similar utility under Windows, say usbtreeview.
https://www.uwe-sieber.de/usbtreeview_e.html

mcuee, 2021-01-05:
Also this github repo is for issue reporting and not for technical report. Please post your question to the mailing list, including the USB descriptors.

mcuee, 2021-01-06:
Basically you can not just use your windows code in Mac/Linux as the device is obviously different from Windows -- that is why the product ID is different. You have to figure out a way to send some command to to the device to make it switching to the "Windows" operating mode. You have to reverse engineering if you do not know the communication protocol.

I will close the issue for now. You can ask the question in libusb mailing list after you have done some more research about the device.

closed 2021-01-06 by mcuee

mcuee, 2021-01-16:
The issue is now discussed in the mailing list.