libusb/libusb #840: The new version of linux kernel no longer supports usbfs, how to use libusb in embedded arm development board

lijiatuYY, 2021-01-09:
I plan to use libusb-1.0.24 in the embedded arm development board, but the program after cross-compilation fails in the first step libusb_init, and the return value is -99. The reason may be that I am using the 4.9.84 kernel. According to the information found, libusb uses usbfs to access the device, but the new kernel from 4.9.8 no longer supports usbfs, so how should I use it in 4.9.84 What about libusb library.

mcuee, 2021-01-09:
Who told you that? usbfs is still in the latest kernel.
https://elixir.bootlin.com/linux/v5.10/source/drivers/usb/core/devio.c

What you are seeing is different. Please pose the debug log.

mcuee, 2021-01-09:
Also this github is for issue reporting. For questions, please go to libusb mailing list.

mcuee, 2021-01-10:
Close for now. Please use libusb mailing list for further support. Thanks.

closed 2021-01-10 by mcuee