libusb/libusb #844: Access violation in libusb v1.0.24 when using libusbK driver

dmitry-zakablukov, 2021-01-11:
Good day!

The following test code crashes with 0xC0000005 error code if libusb v1.0.24 is used for a device with libusbK driver:
```
#include "libusb.h"

#define VID 0x0951
#define PID 0x1600

libusb_device_handle* open_device( libusb_context* ctx, int vid, int pid )
{
    libusb_device_handle* handle = nullptr;

    libusb_device** devices = nullptr;
    ssize_t count = libusb_get_device_list( ctx, &devices );

    for( ssize_t index = 0; index < count; ++index )
    {
        libusb_device_descriptor desc{};
        libusb_get_device_descriptor( devices[ index ], &desc );

        if( desc.idVendor == VID && desc.idProduct == PID )
        {
            libusb_open( devices[ index ], &handle );
            break;
        }
    }

    libusb_free_device_list( devices, 1 );
    return handle;
}

int main()
{
    libusb_context* ctx = nullptr;
    libusb_init( &ctx );

    libusb_device_handle* handle = open_device( ctx, VID, PID );

    unsigned char buffer[ 256 ]{};

    // HERE: the following function call will crash the program
    libusb_control_transfer( handle,
        LIBUSB_ENDPOINT_IN | LIBUSB_REQUEST_TYPE_STANDARD | LIBUSB_RECIPIENT_DEVICE,
        LIBUSB_REQUEST_GET_DESCRIPTOR,
        LIBUSB_DT_STRING << 8,
        0,
        buffer,
        sizeof( buffer ),
        1000 );

    libusb_close( handle );
    libusb_exit( ctx );

    return 0;
}
```

Note that there is no crash if libusb v1.0.23 is used or a device has the WinUSB driver.

I know the problem is in `windows_iocp_thread() / usbi_signal_transfer_completion()` functions: access violation occurs in https://github.com/libusb/libusb/blob/master/libusb/io.c#L1734

When a `libusb_control_transfer()` is called, the libusbK driver is initialized, see https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L2531
This driver initialization triggers a return from `GetQueuedCompletionStatus()` in `windows_iocp_thread()`, but the data in `overlapped` variable is not the desired one (it is not the libusb library who prepared this data). This results in garbage pointed by `itransfer` variable here: https://github.com/libusb/libusb/blob/master/libusb/os/windows_common.c#L438

The log shows this garbage access:
```
[ 0.020483] [00002ab4] libusb: debug [libusb_get_device_descriptor]
[ 0.020572] [00002ab4] libusb: debug [libusb_alloc_transfer] transfer 0000029EE5AD9838
[ 0.020660] [00002ab4] libusb: debug [libusb_submit_transfer] transfer 0000029EE5AD9838
[ 0.020741] [00002ab4] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.020820] [00002ab4] libusb: debug [libusb_claim_interface] interface 0
[ 0.020929] [00004c9c] libusb: debug [windows_iocp_thread] transfer 000000ADED2FC258 completed, length 24
[ 0.020980] [00002ab4] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.021037] [00004c9c] libusb: debug [windows_iocp_thread] transfer 000000ADED2FC2F8 completed, length 1
[ 0.021097] [00002ab4] libusb: debug [windows_assign_endpoints] no endpoints found for interface 0
```

There was no 000000ADED2FC258 transfer, it is a garbage data.

I hope this information is enough to fix the problem.

mcuee, 2021-01-13:
@dickens Please take a look at well. Thanks. If possible, we may want to have 1.0.25 release to address #831 and#838.

dmitry-zakablukov, 2021-01-14:
@mcuee , @dickens , I've fixed the problem, please take a look at PR #848.

dickens, 2021-01-14:
> @mcuee , @dickens , I've fixed the problem, please take a look at PR #848.

Thanks for the PR but unfortunately your approach isn’t going to work.  It isn’t safe to access the memory surrounding the `OVERLAPPED` pointer without first validating that the pointer originated from within libusb.  I’m working on a proper fix, hopefully by the end of the week.

closed 2021-01-16 by dickens

mcuee, 2021-06-08:
Not so sure if #912 is related or not.

cnlohr, 2021-09-22:
Any way to get a release candidate of 1.0.25?

mcuee, 2021-09-22:
> Any way to get a release candidate of 1.0.25?
Please post here. 
https://github.com/libusb/libusb/issues/777

cnlohr, 2021-09-22:
This error is not limited to libusbK, but also exists with the backend of libusb-win32 v1.2.6.0.  Running a device with that backend using zadig will cause the same crash.

mcuee, 2021-09-22:
Yes that is correct. But for libusb-win32 device driver, latest git is still broken.

cnlohr, 2021-09-22:
Understood.

mcuee, 2021-09-22:
Ref: https://github.com/libusb/libusb/issues/761
git head is okay with WinUSB and libusbk driver, but not libusb0.sys (libusb-win32 device driver).
1.024 is okay with WinUSB but not libusbk driver and libusb0.sys (libusb-win32 device driver).

commit b51c743e (2021-01-16): Windows: Filter out non-libusb I/O completions

The I/O completion port thread gets notified of every I/O operation that
completes for a given HANDLE, but not all I/O operations originate from
within libusb. For example, libusbK performs a number of I/O control
calls during initialization and specific device operations, such as
setting an interface alternate setting. These non-libusb operations need
to be ignored as the OVERLAPPED associated with such operations is not
tied to a libusb_transfer structure.

Resolve this situation by using the libusb device handle as the I/O
completion port's un [... cut]