libusb/libusb #845: libusb_fill_iso_transfer callback overflow

fayfive, 2021-01-11:
I'm using libusb to write a camera driver. 
When I am using libusb_fill_iso_transfer,the callback libusb_transfer->iso_packet_desc[i]->status sometime will be overflow.
Is there something wrong? I'm using 720P camera, but libusb_set_iso_packet_lengths can not be malloc larger.

#define num_iso_pack 32
#define per_pack_size 2048
void cb_xfr(struct libusb_transfer *xfr)
{
    int ret = -1;
    if (xfr->type == LIBUSB_TRANSFER_TYPE_ISOCHRONOUS)
    {
        for (int i = 0; i < xfr->num_iso_packets; i++)
        {
            struct libusb_iso_packet_descriptor *pack = &xfr->iso_packet_desc[i];
            if (pack->status != LIBUSB_TRANSFER_COMPLETED)
            {
                LOG("Error: pack %d status %d\n", i, pack->status);
                continue;
            }
            LOG("pack%d length:%u, actual_length:%u\n", i, pack->length, pack->actual_length);
             uint8_t *pktbuf = (uint8_t *)libusb_get_iso_packet_buffer_simple(xfr, i);
             LOG_HEX(pktbuf, 12);
        }
        ret = libusb_submit_transfer(xfr);
        if (ret < 0)
        {
            LOG("%s\n", libusb_strerror((libusb_error)ret));
            if (LIBUSB_ERROR_NO_DEVICE == ret)
            {
                return;
            }
        }
    }
}

uint8_t buf[num_iso_pack * per_pack_size] = {0};
    xfr = libusb_alloc_transfer(num_iso_pack);
    if (!xfr)
    {
        LOG("xfr NULL\n");
    }
libusb_fill_iso_transfer(xfr, device_handle, 0x81, buf, sizeof(buf), num_iso_pack, cb_xfr, NULL, 5000);
    libusb_set_iso_packet_lengths(xfr, per_pack_size);
while (running)
    {
        timeout.tv_sec = 1;
        timeout.tv_usec = 0;
        ret = libusb_handle_events_timeout(NULL, &timeout);
        if (LIBUSB_SUCCESS == ret)
        {
            continue;
        }
        else if (LIBUSB_ERROR_TIMEOUT == ret)
        {
            continue;
        }
        else
        {
            LOG("%s\n", libusb_strerror((libusb_error)ret));
        }
    }

mcuee, 2021-01-12:
Please use libusb mailing list for technical support.

fayfive, 2021-01-12:
@mcuee
Hello, I'm new here. I have confirmed the libusb mailing list. How can I submit the question? 
linux kernel is 3.10,0, 
usb camera is here.

Connection Status	Device connected
Current Configuration	1
Speed	High (480 Mbit/s)
Device Address	14
Number Of Open Pipes	1
Device Descriptor
Offset	Field	Size	Value	Description
0	bLength	1	12h	
1	bDescriptorType	1	01h	Device
2	bcdUSB	2	0200h	USB Spec 2.0
4	bDeviceClass	1	EFh	Miscellaneous
5	bDeviceSubClass	1	02h	Common Class
6	bDeviceProtocol	1	01h	Interface Association Descriptor
7	bMaxPacketSize0	1	40h	64 bytes
8	idVendor	2	046Dh	Logitech, Inc.
10	idProduct	2	0825h	
12	bcdDevice	2	0012h	0.12
14	iManufacturer	1	00h	
15	iProduct	1	00h	
16	iSerialNumber	1	02h	"2C043370"
17	bNumConfigurations	1	01h	
Device Qualifier Descriptor
Offset	Field	Size	Value	Description
0	bLength	1	0Ah	
1	bDescriptorType	1	06h	Device Qualifier
2	bcdUSB	2	0200h	USB Spec 2.0
4	bDeviceClass	1	EFh	Miscellaneous
5	bDeviceSubClass	1	02h	Common Class
6	bDeviceProtocol	1	01h	Interface Association Descriptor
7	bMaxPacketSize0	1	40h	64 bytes
8	bNumConfigurations	1	01h	
9	bReserved	1	00h	
Configuration Descriptor 1 Bus Powered, 500 mA
Offset	Field	Size	Value	Description
0	bLength	1	09h	
1	bDescriptorType	1	02h	Configuration
2	wTotalLength	2	09A2h	
4	bNumInterfaces	1	04h	
5	bConfigurationValue	1	01h	
6	iConfiguration	1	00h	
7	bmAttributes	1	80h	Bus Powered
4..0: Reserved		...00000 	
5: Remote Wakeup		..0..... 	No
6: Self Powered		.0...... 	No, Bus Powered
7: Reserved (set to one)
(bus-powered for 1.0)		1....... 	
8	bMaxPower	1	FAh	500 mA
Interface Association Descriptor Video Interface Collection
Offset	Field	Size	Value	Description
0	bLength	1	08h	
1	bDescriptorType	1	0Bh	Interface Association
2	bFirstInterface	1	00h	
3	bInterfaceCount	1	02h	
4	bFunctionClass	1	0Eh	Video
5	bFunctionSubClass	1	03h	Video Interface Collection
6	bFunctionProtocol	1	00h	
7	iFunction	1	00h	
Interface Descriptor 0/0 Video, 1 Endpoint
Offset	Field	Size	Value	Description
0	bLength	1	09h	
1	bDescriptorType	1	04h	Interface
2	bInterfaceNumber	1	00h	
3	bAlternateSetting	1	00h	
4	bNumEndpoints	1	01h	
5	bInterfaceClass	1	0Eh	Video
[... 1563 lines cut ...]
Offset	Field	Size	Value	Description
0	bLength	1	0Bh	
1	bDescriptorType	1	24h	Audio Streaming Format Type
2		9	02 01 01 02 10 01 80 BB 
00 	
Endpoint Descriptor 86 6 In, Isochronous, 125 us
Offset	Field	Size	Value	Description
0	bLength	1	09h	
1	bDescriptorType	1	05h	Endpoint
2	bEndpointAddress	1	86h	6 In
3	bmAttributes	1	05h	Isochronous, Asynchronous, Data
1..0: Transfer Type		......01 	Isochronous
3..2: Sync Type		....01.. 	Asynchronous
5..4: Usage Type		..00.... 	Data
7..6: Reserved		00...... 	
4	wMaxPacketSize	2	00C4h	196 bytes
6	bInterval	1	01h	125 us
7	bRefresh	1	00h	
8	bSynchAddress	1	00h	
Audio Streaming Isochronous Audio Data Endpoint Descriptor
Offset	Field	Size	Value	Description
0	bLength	1	07h	
1	bDescriptorType	1	25h	Audio Streaming Isochronous Audio Data Endpoint
2		5	01 01 00 00 00 	
This report was generated by USBlyzer

fayfive, 2021-01-12:
Sometime, I can receive the mjpeg and save it, but it is not complete. I can see 2/5 picture with ffmplay -f mjpeg

mcuee, 2021-01-12:
https://sourceforge.net/projects/libusb/lists/libusb-devel
Once you subscribe to the mailing list, you can send the email to the mailing list.

mcuee, 2021-01-13:
Now the OP has posted in the mailing list. Close for now.

closed 2021-01-13 by mcuee