libusb/libusb #847: Potential libusb_set_interface_alt_setting improvement for macOS

mcuee, 2021-01-14:
As mentioned by #838, there is a macOS specific behavior if libusb_set_interface_alt_setting failed with a pipe stall. 

```
kresult = (*(cInterface->interface))->SetAlternateInterface (cInterface->interface, altsetting);
  /* If a device only supports a default setting for the specified interface, then a STALL
     (kIOUSBPipeStalled) may be returned. Ref: USB 2.0 specs 9.4.10.
     Mimick the behaviour in e.g. the Linux kernel: in such case, reset all endpoints,
     and hide errors.Current implementation resets the entire device, instead of single
     interface, due to historic reasons. */
  if (kresult != kIOReturnSuccess) {
    usbi_warn (HANDLE_CTX (dev_handle), "SetAlternateInterface: %s", darwin_error_str(kresult));
    darwin_reset_device (dev_handle);
  }
```

> From Nathan:
The reset was not present in libusb 0.1.12 so I specifically added it when implementing the libusb 1.0 backend. It was undoubtedly to work around some issue seen during development. If there was some sort of bug in that version of IOUSBFamily it is likely long gone (since at least 10.9.0 given the IOUSBFamily rewrite).

tormodvolden, 2021-01-14:
I have some suggestions at https://github.com/tormodvolden/libusb/tree/darwin_alt_intf
Not yet a merge requests because it wasn't tested much, and there is something going wrong in darwin_clear_halt(): it seems to me that it finds the correct pipeRef's but ClearPipeStall returns "pipe ref not recognized". Maybe I use the wrong parameter or an off-by-one. Ran out of time but maybe someone else spots it. Also not sure if get_endpoints() is needed any longer.

hjelmn, 2021-01-14:
get_endpoints should be required as the pipes may change.

mcuee, 2021-01-15:
One idea I mentioned in #838.
>We can probably check the current alternative setting and do nothing (most of the cases) if it is already at the correct alternative setting. Majority the device should only have the defaul alt setting 0. And for those which needs to be set a different alt settings, then the device should anyway supports the call propely.

mcuee, 2021-01-15:
BTW, I remember there are devices having problem with Clear_HALT feature request as well.

mcuee, 2021-01-15:
Of course there is another opinion that libusb should not do any check and the application should check. 

However, I think under Windows due to the limitation of WinUSB/libusbK.sys, we actually check for libusb_set_configuration. So probably we can do similar things here for libusb_set_interface_alt_setting() under macOS.

mcuee, 2021-01-15:
From #838 by tormodvolden.
----
If this is the reason behind the device_reset(), it is maybe enough to call darwin_clear_halt().

I don't have a Pycom board to test on, but I tried on a PL2303 USB-serial adapter (which was reported by macfreek to behave the same in this regard) on Linux after unbinding the kernel driver from the USB interface. Indeed there is a PIPE error, and CLEAR_FEATURE requests to each of the three endpoints in the interface are sent [by the kernel](https://elixir.bootlin.com/linux/v5.10.7/source/drivers/usb/core/message.c#L1588), and there is no error from libusb.

If I try to set an alternate interface different than zero (not available on this interface) using pyusb, there will be an error but it seems pyusb is doing some [checks](https://github.com/pyusb/pyusb/blob/49e896b3de240ffa1b66779adf1d00d38cf6a108/usb/core.py#L205) and will not send an invalid SET_INTERFACE USB request.

tormodvolden, 2021-01-15:
To document the observation in the quote above, here is what goes on the wire (Linux, PL2303):
```
333	5.600698	host	3.43.0	USB	64	SET INTERFACE Request (0, 0)
334	5.600812	3.43.0	host	USB	64	SET INTERFACE Response (-EPIPE)
335	5.600851	host	3.43.0	USB	64	CLEAR FEATURE Request (ENDPOINT HALT, 129)
336	5.600931	3.43.0	host	USB	64	CLEAR FEATURE Response (success)
337	5.600947	host	3.43.0	USB	64	CLEAR FEATURE Request (ENDPOINT HALT, 2)
338	5.601021	3.43.0	host	USB	64	CLEAR FEATURE Response (success)
339	5.601096	host	3.43.0	USB	64	CLEAR FEATURE Request (ENDPOINT HALT, 131)
340	5.601177	3.43.0	host	USB	64	CLEAR FEATURE Response (success)
```
Is there an equivalent to usbmon on macOS to see what is actually being sent?

mcuee, 2021-01-15:
https://developer.apple.com/forums/thread/95380
https://developer.apple.com/forums/thread/124875
As per the thread, the USB traffic function is no longer available after 10.15, but it is said to be able to work if SIP is disabled.

mcuee, 2021-01-15:
Just a side thing, it seems Apple has made big changes after 10.11 for USB.
https://developer.apple.com/support/kernel-extensions/

IOUSBFamily
IOUSBFamily has been deprecated and headers removed from SDK since macOS El Capitan (10.11). All clients should move to IOUSBHostFamily or USBDriverKit, where appropriate and outlined below.

USB Networking
The combination of using IONetworkingFamily KPIs as well as any USB KPI (IOUSBHostFamily or IOUSBFamily) is deprecated. USB KPIs have alternatives in USBDriverKit . Clients of IONetworkingFamily KPIs should use NetworkingDriverKit instead.

USB Serial
The combination of using any IOSerialFamily KPI as well as any USB KPI (IOUSBHostFamily or IOUSBFamily) is deprecated. USB KPIs have alternatives in USBDriverKit. Clients of IOSerialFamily KPIs should use SerialDriverKit or USBSerialDriverKit instead.

USB Vendor Specific IPC
Any kernel extension using USB KPI (IOUSBHostFamily or IOUSBFamily) that does not tie into the IO Family subsystem and uses IOUserClient, or subclasses, for IPC is deprecated. USB KPIs have alternatives in USBDriverKit or consider IOUSBHost. DriverKit also has mechanisms for communication channels.

mcuee, 2021-01-15:
https://developer.apple.com/documentation/usbdriverkit
USBDriverKit seems to be the main one.

tormodvolden, 2021-01-16:
I am running into what might be other issues when I test this out with pyusb. Or there is really some resetting in the USB stack needed after the pipe error. I can run dev.set_interface_altsetting(0, 0) once and as described above ClearPipeStall returns "pipe ref not recognized". But if I try it again I get Access denied (insufficient permissions). Calling clear_halt on a pipe now also returns "pipe ref not recognized" and "Other error". If I do dev.set_configuration(1) again, which is said to do a "light-weight" reset but AFAICS sends nothing over the wire in my case, it will work another time. Calling clear_halt on a pipe in the "good" state sends CLEAR_FEATURE requests as expected.

I am using Wireshark (3.4.2) which seem to work, but I see URB unknown status 0xe0005000 instead of -EPIPE as on Linux. Maybe the capture or decoding is buggy because the source/destination shows up as device->host on all entries.

tormodvolden, 2021-01-16:
This is what dev.set_interface_altsetting(0, 0) (with my patches) looks like on macOS 10.13.6:
```
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       19.4.0                host                  USB      40     SET INTERFACE Request
      2 0.000166       19.4.0                host                  USB      32     SET INTERFACE Response
      3 0.000265       16.1.0                host                  USBHUB   40     CLEAR_TT_BUFFER Response
      4 0.000419       16.1.0                host                  USBHUB   32     CLEAR_TT_BUFFER Response
      5 0.000506       16.1.0                host                  USBHUB   40     CLEAR_TT_BUFFER Response
      6 0.000659       16.1.0                host                  USBHUB   32     CLEAR_TT_BUFFER Response
```
I don't know why these CLEAR_TT_BUFFER are generated. They are not generated on a clear_halt.

mcuee, 2021-01-16:
What if you try it on a good device which does not generate the pipe stall. Maybe you can compare the wireshark capture.

The IOkit/Driverkit documentation is really bad with the new macOS.

mcuee, 2021-01-16:
But if you read that, it is not necessary to issue clear halt command for the Control Endpoint. As mentioned in #838 by Chris, it is not necessary for the Control Endpoint.
>From Chris: clearing a pipe stall is only applicable to bulk/interrupt endpoints. A pipe stall in the case of a failed set interface request would be returned by the control pipe, which never needs to be cleared.

https://developer.apple.com/documentation/usbdriverkit/iousbhostpipe/3182649-clearstall
>Discussion
When a bulk or interrupt USB endpoint encounters an I/O error other than a timeout, it transitions to a halted state that you must clear if you want to perform additional I/O on the endpoint. Call this method to clear the halted condition for the endpoint. This method aborts all pending I/O on the endpoint and resets its data toggle.

If an intermediate hub is between the device and your driver, this method also sends a CLEAR_TT_BUFFER control request (USB 2.0, section 11.24.2.3) to the hub.

tormodvolden, 2021-01-16:
As I mentioned in the other ticket, the reason I want to send CLEAR_FEATURE to each endpoint like Linux do, is /not/ to clear the control endpoint from the pipe error. It is to emulate as much as possible what would be done if the SET_INTERFACE would be supported by the device. So if the user wants to libusb_set_alt_interface but we see that request fails but would anyway only have reset the endpoints, we reset the endpoints and consider this successful.

I think comparing the wireshark captures is tricky, what we see on macOS is not as close to the "wire" as on Linux.

mcuee, 2021-01-16:
I see. Thanks.

Maybe it is easier to use test program as per the USBDriverKit documentation and then see the error codes.
https://developer.apple.com/documentation/usbdriverkit/iousbhostinterface
https://developer.apple.com/documentation/usbdriverkit/iousbhostinterface/3182596-selectalternatesetting

tormodvolden, 2021-01-17:
I finally found a good solution/workaround: Instead of the full darwin_set_configuration() after the pipe error, I tried removing the SetConfiguration() call (which didn't seem to create any USB request anyway). And the remaining darwin_release_interface / darwin_claim_interface dance is enough to make it work.

What I see in Wireshark now pretty nicely matches what I see on Linux, apart from the extra CLEAR_TT_BUFFER requests that from above Linux kernel patch proposal link seem legitimate.

I don't know if this is because of a libusb or darwin bug (or undocumented feature).

In case my branch get merged (which I highly recommend for 1.0.25 since this is much better than the current darwin_reset_device() workaround) we should keep this ticket open, for adding a check that the interface has only one alternate interface. If it has several, the pipe failure should make our library API function fail, and not pretend it was successful.

mcuee, 2021-01-17:
We will keep this ticket open if the improvement is not merged. Doing "darwin_release_interface / darwin_claim_interface" sounds good to me and it is less intrusive than the current darwin_reset_device.

On the other hand, I still think may idea is worth considering, that is to do nothing and return success if the device is already at the correct alt setting.
> We can probably check the current alternative setting and do nothing (most of the cases) if it is already at the correct alternative setting. Majority the device should only have the defaul alt [...]

tormodvolden, 2021-01-17:
The addition of darwin_clear_halt makes it now behave like on Linux. If a device wouldn't handle it, we would have seen problems on Linux.

Because setting the alternate interface even when it is already the current setting has the standard-mandated side effect of clearing all endpoints, I think we should not quietly skip it like you suggest. Many devices have only one setting but support the SET_INTERFACE request nonetheless.

mcuee, 2021-01-18:
I just read the USB 2.0 Specification 9.4.10. Actually it does not mention the "the standard-mandated side effect of clearing all endpoints". But if you say Linux kernel is doing that, then I agree with you not to skip it silently.

https://www.usb.org/document-library/usb-20-specification
> Set Interface
This request allows the host to select an alternate setting for the specified interface.

Some USB devices have configurations with interfaces that have mutually exclusive settings. This request
allows the host to select the desired alternate setting. If a device only supports a default setting for the
specified interface, then a STALL may be returned in the Status stage of the request. This request cannot be
used to change the set of configured interfaces (the SetConfiguration() request must be used instead).

If the interface or the alternate setting does not exist, then the device responds with a Request Error. If
wLength is non-zero, then the behavior of the device is not specified.

Default state: Device behavior when this request is received while the device is in the Default state
is not specified.

Address state: The device must respond with a Request Error.

Configured state: This is a valid request when the device is in the Configured state.

tormodvolden, 2021-01-18:
Thanks for looking up this. It is actually in 9.1.1.4:

mcuee, 2021-01-19:
Thanks. Good point. Setting the data toggle to DATA0 is kind of important.

Back in 2009, there was a discussion thread on this topic: [Libusb-devel] Set Alternative Configuration.
https://sourceforge.net/p/libusb/mailman/libusb-devel/thread/49E8AAF9.8060808@gmail.com/

mcuee, 2021-01-19:
Another interesting thread in 2011. I was dealing with Microchip USB devices with buggy FW (and some say USB SIE implementation) where Data Toggle handling may not be correct and my brutal force method is to call libusb_reset_device to deal with such devices at the end of the program.

https://marc.info/?l=libusb-devel&m=129415457123625&w=4

> From Alan Stern
>Before 2.6.28, Linux would always send a Set-Interface(0) request when 
a libusb file was closed or an interface was unclaimed.  Starting with 
2.6.28, Linux would do this only if the interface wasn't already in 
altsetting 0.

>The USB-2.0 spec is surprisingly ambiguous about what should happen in 
these situations.  Section 9.1.1.5 says:

>	Configuring a device or changing an alternate setting causes
	all of the status and configuration values associated with
	endpoints in the affected interfaces to be set to their default
	values. This includes setting the data toggle of any endpoint
	using data toggles to the value DATA0.

> It's not clear what should happen when a Set-Interface request 
specifies the _same_ altsetting as the interface is already in.

mcuee, 2021-01-19:
https://marc.info/?l=libusb-devel&m=129416609706824&w=4

From Alan Stern
>And for what it's worth, I still don't know
what the spec's intended behavior is when a Set-Interface request
leaves the altsetting the same instead of changing it.  The Linux USB
stack assumes the data toggles will all be reset, but it appears that
devices are within their rights not to do so.

>It's an odd omission.

mcuee, 2021-01-19:
@tormodvolden So you can say USB 2.0 specification is clear when **changing an alternate setting** but it is not clear if **not changing**. 

Tim Roberts mentioned that Windows almost never have Data Toggle problem because it seldom sends unsolicited Set Configuration or Set Interface command. Linux has also changed after 2.6.28 by reducing the chances. I am not so sure about macOS.

mcuee, 2021-01-19:
So to be pedantic, in this case, the device is already at alt setting 0 and the call is setting alt setting 0 and not changing the alt setting to other value, the device can return a pipe stall (no change of data toggle), or the device may reset the data toggle to DATA0.

mcuee, 2021-01-19:
Still in any case, I think it is okay by following Linux behavior.

mcuee, 2021-07-05:
Close this since https://github.com/libusb/libusb/pull/852 has been merged.

closed 2021-07-05 by mcuee