libusb/libusb #849: libusb_submit_transfer() return LIBUSB_ERROR_NO_DEVICE on MacOS

laran002, 2021-01-14:
I am having some problems with libusb on MacOS Mojave 10.14.6. The same code works fine on linux and Windows and I am connecting to a vendor class device.. I am thinking this may be related to the handling of vendor class on MacOS.

Now to the point,
- I setup a device, opens a handle and sends a control transfer using libusb_submit_transfer(). all working fine
- When I then submit another transfer making a control read libusb_submit_transfer() return LIBUSB_ERROR_NO_DEVICE

I have read somewhere about macos not configuring vendor class completely, is this still an issue, or is there something else that is know that may cause my problem?

mcuee, 2021-01-14:
Please post more details about your device, your codes and the debug log.

laran002, 2021-01-18:
Please see the complete log, from initialization through the control transfer that fails. I believe the last four lines may spread some light on the problem.

[ 0.021391] [0020e904] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.021475] [0020e904] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.021500] [0020e904] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.021929] [0020e904] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.949639] [0020e847] libusb: debug [libusb_get_device_list]  
[ 0.951037] [0020e847] libusb: debug [libusb_open] open 20.7
[ 0.951345] [0020e847] libusb: debug [darwin_open] device open for access
[ 0.951490] [0020e847] libusb: debug [libusb_open] open 20.3
[ 0.951721] [0020e847] libusb: debug [darwin_open] device open for access
[ 0.952327] [0020e847] libusb: debug [libusb_get_device_descriptor]  
[ 0.952376] [0020e847] libusb: debug [libusb_get_device_descriptor]  
[ 0.952387] [0020e847] libusb: debug [libusb_get_device_descriptor]  
[ 0.952395] [0020e847] libusb: debug [libusb_get_device_descriptor]  
[ 0.952415] [0020e847] libusb: debug [libusb_alloc_transfer] transfer 0x101cc6d38
[ 0.952421] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101cc6d38
[ 0.952519] [0020e847] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.952534] [0020e847] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 0.952770] [0020e903] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.952845] [0020e904] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.952871] [0020e904] libusb: debug [handle_event_trigger] event triggered
[ 0.952878] [0020e904] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.952885] [0020e904] libusb: debug [usbi_handle_transfer_completion] transfer 0x101cc6d38 has callback 0x101aa2470
[ 0.952890] [0020e904] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.952953] [0020e847] libusb: debug [libusb_free_transfer] transfer 0x101cc6d38
[ 0.952968] [0020e904] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.952978] [0020e904] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.952986] [0020e904] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.953011] [0020e847] libusb: debug [libusb_alloc_transfer] transfer 0x101dd9d68
[ 0.953018] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101dd9d68
[ 0.953075] [0020e847] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.953086] [0020e847] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 0.953113] [0020e903] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.953158] [0020e904] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.953176] [0020e904] libusb: debug [handle_event_trigger] event triggered
[ 0.953182] [0020e904] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.953189] [0020e904] libusb: debug [usbi_handle_transfer_completion] transfer 0x101dd9d68 has callback 0x101aa2470
[ 0.953194] [0020e904] libusb: debug [sync_transfer_cb] actual_length=24
[ 0.953221] [0020e847] libusb: debug [libusb_free_transfer] transfer 0x101dd9d68
[ 0.953266] [0020e904] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.953279] [0020e904] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.953281] [0020e847] libusb: debug [libusb_get_device_descriptor]  
[ 0.953313] [0020e847] libusb: debug [libusb_alloc_transfer] transfer 0x101cc6d38
[ 0.953327] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101cc6d38
[ 0.953292] [0020e904] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 0.953401] [0020e847] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.953413] [0020e847] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 0.953619] [0020e903] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.953669] [0020e904] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.953681] [0020e904] libusb: debug [handle_event_trigger] event triggered
[ 0.953686] [0020e904] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status 0
[ 0.953693] [0020e904] libusb: debug [usbi_handle_transfer_completion] transfer 0x101cc6d38 has callback 0x101aa2470
[ 0.953698] [0020e904] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.953785] [0020e904] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.953797] [0020e847] libusb: debug [libusb_free_transfer] transfer 0x101cc6d38
[ 0.953812] [0020e904] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.953836] [0020e847] libusb: debug [libusb_alloc_transfer] transfer 0x101cc6d38
[ 0.953868] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101cc6d38
[ 0.953875] [0020e904] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[... cut ...]
[ 0.957566] [0020e904] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 1.222622] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101cc6d38
[ 1.222631] [0020e847] libusb: debug [ep_to_pipeRef] converting ep address 0x80 to pipeRef and interface
[ 1.222636] [0020e847] libusb: warning [ep_to_pipeRef] no pipeRef found with endpoint address 0x80.
[ 1.222639] [0020e847] libusb: error [submit_control_transfer] endpoint not found on any open interface
[ 1.222660] [0020e847] libusb: debug [libusb_free_transfer] transfer 0x101cc6d38

laran002, 2021-01-18:
I should mention that I am working with a home built device based on atmel Vendor class. Perhaps I can try this using another device, just to verify this is not related to my device.

If someone can point me to a suitable device/class that I can use and suggest some basic control requests I can try out with, that would be helpful.

dickens, 2021-01-18:
```
[ 1.222622] [0020e847] libusb: debug [libusb_submit_transfer] transfer 0x101cc6d38
[ 1.222631] [0020e847] libusb: debug [ep_to_pipeRef] converting ep address 0x80 to pipeRef and interface
[ 1.222636] [0020e847] libusb: warning [ep_to_pipeRef] no pipeRef found with endpoint address 0x80.
[ 1.222639] [0020e847] libusb: error [submit_control_transfer] endpoint not found on any open interface
```
The failure seems to be because the `endpoint` field of the transfer is non-zero, which means it is trying to target a control pipe other than the default one which is zero.  I’m guessing the device does not have multiple control pipes.

mcuee, 2021-01-19:
Most of the device will only have one Control Endpoint (EP0).

mcuee, 2021-01-19:
@laran002 Please post your codes and device descriptor as well. Thanks.

laran002, 2021-01-19:
Yes, I have the LIBUSB_ENDPOINT_IN bit set in the transfer endpoint. This is the only way I have found it possible to distinguish between in and out transfer in the transfer complete callback function. This does not appear to be a problem on Windows, nor Linux.

I have found a solution that works for me. That is to submit the transfer with EP (0x00) and then set the LIBUSB_ENDPOINT_IN bit directly after submission. That way pipeRef conversion get the expected 0x00 EP and I get the 0x80 in my callback.

I am not familiar with the pipeRef conversion, although if one want same behavior on all platforms I can give it a shot. I am not sure what behavior is expected thouh.

mcuee, 2021-01-19:
In this case, your application has a bug. When you specify Endpoint Address, direction bit is a must. So you have to use 0x80 for CONTROL IN and 0 for CONTROL OUT.

mcuee, 2021-01-19:
Close this for now.

mcuee, 2021-01-19:
Please also look into the example codes in libusb repo to see how to use libusb.

closed 2021-01-19 by mcuee

laran002, 2021-01-19:
Well, It is when I am using the direction bit that LibUSB fails to submit the transfer, as dickens points out above.

As I interpret the discussion above, this warning should never happen since the endpoint to be opened is 0x00, with the CONTROL_IN bit set, equals 0x80.
[ 1.222636] [0020e847] libusb: warning [ep_to_pipeRef] no pipeRef found with endpoint address 0x80

This code is from ep_to_pipeRef(.) in Darwin_usb.c. Should a device with a control endpoint actually indicate num_endpoints =2 (0x00, and 0x80) or just the single control endpoint (0x00) assuming it is "bidirectional"?
`  for (iface = 0 ; iface < USB_MAXINTERFACES ; iface++) {
    cInterface = &priv->interfaces[iface];

    if (dev_handle->claimed_interfaces & (1U << iface)) {
      for (i = 0 ; i < cInterface->num_endpoints ; i++) {
        if (cInterface->endpoint_addrs[i] == ep) {
          *pipep = i + 1;

          if (ifcp)
            *ifcp = iface;

          if (interface_out)
            *interface_out = cInterface;

          usbi_dbg ("pipe %d on interface %d matches", *pipep, iface);
          return LIBUSB_SUCCESS;
        }
      }
    }
  }`

laran002, 2021-01-19:
Yes, the Examples are great. I always use them as a reference.

dickens, 2021-01-19:
The default control pipe is bi-directional.  The direction of the transfer is specified in the bmRequestType field of the setup packet.

laran002, 2021-01-19:
ok, got it. Thanks!

All works fine when leaving libusb_transfer.endpoint = 0x00 after filling the transfer.

My only concern is that that the behavior differs between the platforms. Is it worth adding a warning if control ep has the direction bit set on the other platforms as well?

hjelmn, 2021-01-19:
Do all platforms support multiple control endpoints? I added the support to the macOS backend after encountering a device with two control endpoints.

In this case it definitely is a user error as the endpoint specified did not exist. Not sure what else could be expected?

diabolo38, 2021-01-19:
Multiple control ep  is very uncommon but  as far i remember the spec does not put any limit , any ep can be control.

mcuee, 2021-01-19:
To me this is a user problem so we do not need to do anything else. That is why I chose to close the ticket.

mcuee, 2021-01-19:
I remember old discussion when libusb-1.0 was started and the conclusion is to fail those wrong Control In request if the user forgot to use the direction bit.

mcuee, 2021-01-22:
Basically it is like this.

libusb-1.0 has always required the user to specify the endpoint address, not just the endpoint number.

libusb-0.1 does not require the direction bits actually. So libusb-compat-0.1 was changed to be compatible with libusb-0.1.

libusb-win32 has always required the user to specify the endpoint address, including the direction bits.

mcuee, 2021-01-22:
Ref from libusb-compat-0.1 codes.

```
https://github.com/libusb/libusb-compat-0.1/blob/master/libusb/core.c
API_EXPORTED int usb_bulk_read(usb_dev_handle *dev, int ep, char *bytes,
	int size, int timeout)
{
	if (!(ep & USB_ENDPOINT_IN)) {
		/* libusb-0.1 will strangely fix up a read request from endpoint
		 * 0x01 to be from endpoint 0x81. do the same thing here, but
		 * warn about this silly behaviour. */
		usbi_warn("endpoint %x is missing IN direction bit, fixing");
		ep |= USB_ENDPOINT_IN;
	}

	return usb_bulk_io(dev, ep, bytes, size, timeout);
}
```