libusb/libusb #850: Regression in libusb 1.0.24

bodograumann, 2021-01-14:
After an update from libusb 1.0.23-3 to 1.0.24-2 on Arch linux, my SmartCard reader does not work anymore.
I confirmed that it works again after a downgrade.
I am using gnupg with pcscd / ccid and get the following errors:

>Jan 14 14:16:13 MYHOSTNAME pcscd[5858]: 00008827 ccid_usb.c:598:OpenUSBByName() Can't get config descriptor on 1/3: LIBUSB_ERROR_NOT_FOUND
>Jan 14 14:16:13 MYHOSTNAME pcscd[5858]: 00000367 ifdhandler.c:150:CreateChannelByNameOrChannel() failed
>Jan 14 14:16:13 MYHOSTNAME pcscd[5858]: 00000020 readerfactory.c:1105:RFInitializeReader() Open Port 0x200000 Failed (usb:0a5c/5832:libudev:3:/dev/bus/usb/001/003)
>Jan 14 14:16:13 MYHOSTNAME pcscd[5858]: 00000003 readerfactory.c:376:RFAddReader() Broadcom Corp 5880 [Broadcom NFP] (0123456789ABCD) init failed.

Device: Bus 001 Device 003: ID 0a5c:5832 Broadcom Corp. 5880

Kernel: Linux 5.10.7-arch1-1 x86_64
Software:
- gnupg 2.2.26-1
- pcsclite 1.9.0-1
- ccid 1.4.33-1

tormodvolden, 2021-01-14:
For general debugging, please run the program with the debug environment variable set:
`env LIBUSB_DEBUG=4 pcscd -f -d
`
Of course if you could use git bisect to find which change broke it, it would be great.

dickens, 2021-01-14:
I suspect this is the same problem as #825.  Can you please confirm by testing the latest code from the repo?

mcuee, 2021-01-15:
https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/88
lsusb -vvv result of the particular device.
```
Bus 001 Device 005: ID 0a5c:5834 Broadcom Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0a5c Broadcom Corp.
  idProduct          0x5834 
  bcdDevice            1.01
  iManufacturer           1 Broadcom Corp
  iProduct                2 5880
  iSerial                 3 0123456789ABCD
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          269
    bNumInterfaces          4
    bConfigurationValue     0
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
==> Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              4 Broadcom USH w/touch sensor
      ** UNRECOGNIZED:  10 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
[... 161 lines cut ...]
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

bodograumann, 2021-01-15:
@tormodvolden Ok, I have put the log here: https://gist.github.com/bodograumann/a8e723aeb160e62d42eff292f4bff4d8
@dickens I think I could build it locally, but how would I go about running pcscd with the locally-built version?

tormodvolden, 2021-01-15:
"make install" will probably (depending on options to ./configure) install the libraries to /usr/local/lib, and this will override distribution libraries under /usr/lib. You can also set LD_LIBRARY_PATH if you install it somewhere else. You can verify the correct library was loaded in the debug info, e.g.
`[ 0.000041] [00009e73] libusb: debug [libusb_init] libusb v1.0.24.11584`

bodograumann, 2021-01-15:
The problem still persists on master. bisecting lead to this:
```
e9eec3a680cad3b2c9c5213fb7d60148cf6900da is the first bad commit
commit e9eec3a680cad3b2c9c5213fb7d60148cf6900da
Author: Chris Dickens <christopher.a.dickens@gmail.com>
Date:   Tue Apr 28 22:04:11 2020 -0700

    core: Narrow the types passed to certain backend functions
    
    Backend functions dealing with interfaces and alternate settings should
    use a type whose range represents that of valid values for interfaces
    and alternate settings. Switch to use uint8_t instead of int so that
    backends do not have to cast values or do range checks.
    
    Signed-off-by: Chris Dickens <christopher.a.dickens@gmail.com>

 libusb/core.c                   |  31 ++++++------
 libusb/libusbi.h                |  14 +++---
 libusb/os/darwin_usb.c          |  49 +++++++++----------
 libusb/os/darwin_usb.h          |   2 +-
 libusb/os/haiku_usb.h           |  10 ++--
 libusb/os/haiku_usb_backend.cpp |  10 ++--
 libusb/os/haiku_usb_raw.cpp     |   8 ++--
 libusb/os/linux_usbfs.c         | 102 +++++++++++++++++++++-------------------
 libusb/os/netbsd_usb.c          |  28 ++++++-----
 libusb/os/null_usb.c            |   8 ++--
 libusb/os/openbsd_usb.c         |  24 +++++-----
 libusb/os/sunos_usb.c           |  42 ++++++++---------
 libusb/os/windows_common.c      |  12 +++--
 libusb/os/windows_common.h      |  10 ++--
 libusb/os/windows_usbdk.c       |  10 ++--
 libusb/os/windows_winusb.c      |  88 +++++++++++++++++-----------------
 libusb/os/windows_winusb.h      |   8 ++--
 libusb/version_nano.h           |   2 +-
 18 files changed, 230 insertions(+), 228 deletions(-)
```

dickens, 2021-01-15:
Can you capture a debug log?

tormodvolden, 2021-01-15:
He did above, here are some interesting parts:
```
 0.014716] [00009e73] libusb: debug [libusb_get_device_descriptor]  
[ 0.014720] [00009e73] libusb: debug [libusb_open] open 1.3
[ 0.014734] [00009e73] libusb: debug [usbi_add_event_source] add fd 11 events 4
[ 0.014751] [00009e73] libusb: error [op_get_active_config_descriptor] device unconfigured
```

dickens, 2021-01-15:
Ah I missed that, thanks.  Can you post the descriptors of the device?  e.g. run `lsusb -v`

bodograumann, 2021-01-15:
@dickens Sure, here you go: https://gist.github.com/bodograumann/a8e723aeb160e62d42eff292f4bff4d8#file-lsusb-out

dickens, 2021-01-15:
Ok, here's the problem:
```
Bus 001 Device 003: ID 0a5c:5832 Broadcom Corp. 5880
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0a5c Broadcom Corp.
  idProduct          0x5832 
  bcdDevice            1.01
  iManufacturer           1 Broadcom Corp
  iProduct                2 5880
  iSerial                 3 0123456789ABCD
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x010d
    bNumInterfaces          4
    bConfigurationValue     0 <----- USB spec violation
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
```
The fact that it worked before was actually a mistake.  See the comment [here](https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L865).  My commit made the behavior for the sysfs case match the design intent, so I don't know if that fits the pure definition of regression.  However I understand that it is desirable for libusb to support buggy devices, especially if they managed to work in the past.  Let me ponder this for a bit.

mcuee, 2021-01-25:
In general I am not that inclined to support non-compliant devices. 

However, I agree to support buggy devices if they managed to work in the past. And in this particular case, it is even mentioned in libusb API documentation so we will have to support.
http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga785ddea63a2b9bcb879a614ca4867bed
>A configuration value of -1 will put the device in unconfigured state. The USB specifications state that a configuration value of 0 does this, however buggy devices exist which actually have a configuration 0.

commit f38f09da (2021-02-08): linux_usbfs: Gracefully handle buggy devices with a configuration 0

The USB spec states that a configuration value of 0 is reserved and is
used to indicate the device in not configured (e.g. is in the address
state). Unfortunately some devices do exist that violate this and use 0
as the bConfigurationValue of the configuration descriptor.

Improve how the Linux backend handles such non-conformant devices by
adding special handling around the configuration value 0. Most devices
will not require this special handling, but for those that do there is
no way to distinguish between the device being [... cut]
closed 2021-02-08 by dickens