libusb/libusb #855: Segfault at libusb_exit()

ctj-nooiwa, 2021-01-19:
Segfault occurs at libusb_exit() in my environment.
I can reproduce it and test program is follow.
```
$ cat libusb.c
#include <stdio.h>
#include <libusb-1.0/libusb.h>
#include <sys/types.h>
#include <unistd.h>
#include <pthread.h>
// gcc -lusb-1.0 -lpthread libusb.c -o libusb

void* libusb()
{
	libusb_context *ctx = NULL;
	while (1) {
		libusb_init(&ctx);
		libusb_exit(NULL);
	}
	return NULL;
}

int main(int argc, char **argv)
{
	int t;
	pthread_t threads[20];

	for (t = 0; t < 20; t++) {
		pthread_create(&threads[t], NULL, libusb, NULL);
	}
	sleep(100000);
	return 0;
}
```
Segfault occurs after a few seconds on Fedora 32(Intel i7-9700K CPU, libusbx-1.0.23-1 with
a5624b2: "core: Do not attempt to destroy a default context that doesn't exist").
```
$ gdb ./libusb core.libusb.3331047
[...]
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fe14640a8dc in list_del (entry=0x7fe128018088) at libusbi.h:163
163		entry->next->prev = entry->prev;
(gdb) bt
#0  0x00007fe14640a8dc in list_del (entry=0x7fe128018088) at libusbi.h:163
#1  libusb_exit (ctx=0x7fe128017e60) at core.c:2450
#2  0x000000000040117c in libusb () at libusb.c:14
#3  0x00007fe1463eb432 in start_thread () from /lib64/libpthread.so.0
#4  0x00007fe1463199d3 in clone () from /lib64/libc.so.6
```
I created the patch. usbi_get_context() need to hold default_context_lock.
usbi_dbg() also call usbi_get_context().
Please check my patch.
```
diff --git a/libusb/core.c b/libusb/core.c
index 051b4f7..737d4b0 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -2367,13 +2367,12 @@ void API_EXPORTED libusb_exit(libusb_context *ctx)
        struct timeval tv = { 0, 0 };
        int destroying_default_context = 0;
 
-       usbi_dbg(" ");
-
-       ctx = usbi_get_context(ctx);
-
        /* if working with default context, only actually do the deinitialization
         * if we're the last user */
        usbi_mutex_static_lock(&default_context_lock);
+       usbi_dbg(" ");
+       ctx = usbi_get_context(ctx);
+
        if (ctx == usbi_default_context) {
                if (!usbi_default_context) {
                        usbi_dbg("no default context, not initialized?");
```

hjelmn, 2021-01-19:
Not sure this is a valid bug. The code is erroneous. You are creating a non-default context (passing a non-NULL value to ``libusb_init```) but then attempting to destroy the default context (```libusb_exit(NULL)```). ```libusb_exit``` has no return value so this is the only way to indicated libusb_exit has failed.

hjelmn, 2021-01-19:
I can test to see if this still occurs with a valid version of the reproducer.

hjelmn, 2021-01-19:
Nope. Looks fine with both

```c
libusb_init(&ctx);
libusb_exit(ctx);
```

and

```c
libusb_init(NULL);
libusb_exit(NULL);
```

in the loop.

hjelmn, 2021-01-19:
And to clarify, crashing on the wrong context is the only thing you can expect. You gave an invalid context. Yes, libusb could *not* crash and print out a message indicating what happened but that may still be  missed.

mcuee, 2021-01-20:
I agree to treat this as invalid.

ctj-nooiwa, 2021-01-20:
Thank you for quick response. I understand.
I hope someone notice the wrong usage on this page.

dickens, 2021-01-20:
Though the usage detailed in this issue is wrong, libusb is also wrong about how it handles the "default context".  For libusb the "default context" is really the first context.  Thus the original code of
```
libusb_init(&ctx);
libusb_exit(NULL);
```
will work fine because it treats `ctx` as the "default context" since it is the first to be created.  I've always had an issue with this and may now finally be motivated enough to do something about it.

Suppose that two independent libraries within an application use libusb.  Library A runs first and initializes libusb as follows:
```
libusb_init(&ctx);
```
It gets the newly allocated context back in `ctx` and saves this variable for later use.

Now suppose Library B runs and initializes libusb as follows:
```
libusb_init(NULL);
```

Because Library A ran first, the context pointed to by `ctx` is actually assigned as the default context and Library B is actually going to reference/use Library A's context even though, in my opinion,  it should not.  I think libusb should adopt the following semantics with regard to contexts:

1. Calling `libusb_init()` with `NULL` as the context pointer will either create the default context or take a reference to the already existing default context.
2. Calling `libusb_init()` with a non-NULL context pointer will always create a new context, and this context will never be accessible by any other means other than through the returned pointer.

mcuee, 2021-01-20:
Re-open this issue.

mcuee, 2021-01-20:
@dickens The improvement proposal looks nice.

ctj-nooiwa, 2021-01-20:
@dickens 
Thank you for your replay. Your proposal is great.
I was just thinking that the library should not cause a segfault, even if the user misuses it.

mcuee, 2021-01-21:
@ctj-nooiwa Actually I do not agree with you on "the library should not cause a segfault, even if the user misuses it".

ctj-nooiwa, 2021-01-21:
@mcuee 
Thnak you for your reply.
But, why do you think it? I want to know the reason.
Are there any libraries that indicate misuse by segfault ?

In my opinion,
- It is a bug to use usbi_get_context () without holding default_context_lock.
- The libusb no need to indicate misuse. of course, the warning message output is kind.
- If libusb_exit has return value, this segfault is not resolve.

but, I'm going to follow the community policy.

mcuee, 2021-01-21:
Just an example here. Some of the old stuff lingers in my mind so maybe I am biased.
https://marc.info/?t=132468783600004&r=1&w=4

One school of thoughts "Crashing is an inappropriate response to a misuse of a library, especially for what I would consider a system level library".

The other school of thoughts "the library clearly tells you not to do that, crashing is an accepted response if you are not following what the library tells you".

But of course I can see over the years such patches are getting more and more accepted by libusb to check for misuse. I am all for it as long as the developers see it as good.

Still fundamentally I am more for the first school of thought.

hjelmn, 2021-01-21:
Generally speaking most software I have worked on has either intentionally crashed or not avoided crashing on misuse. I would have preferred libusb had returned that status in libusb_exit but that would require and backwards-incompatible API change. I am with @mcuee in that crashing is probably fine here since you asked libusb to delete something that you never created.

I also agree with @dickens on that libusb is handling the default context incorrectly. There should be no default context unless you explicitly call ```libusb_init(NULL)```. We should focus on changing that. It may break some code but I don't think we ever claimed that ```libusb_init(&ctx); libusb_exit(NULL);``` would work. I always assumed it was clear that is not the correct usage.

ctj-nooiwa, 2021-01-22:
@hjelmn 
Thank you for your reply.
> There should be no default context unless you explicitly call libusb_init(NULL). We should focus on changing that.

I think so too.  9a5c688 is very nice.

mcuee, 2021-06-05:
I think this can be closed now that #856 is merged.

closed 2021-06-05 by mcuee