libusb/libusb #858: libusb_handle_events_timeout_completed not waiting on timeout

matallui, 2021-01-21:
This started happening after upgrading from v1.0.23 to v1.0.24.

Calling:
```
libusb_handle_events_timeout_completed(ctx, tv, NULL)
```
with tv = 1 second

Before: it would wait 1 second before returning

After: returns immediately, which in our case causes our thread to use 100% CPU

mcuee, 2021-01-22:
Please post more details, the OS version and libusb debug log (set LIBUSB_DEBUG=4).

dickens, 2021-01-22:
I think I know the answer to this.  I suspect that a hotplug callback has been unregistered.  A bug was introduced when the event abstraction code was merged that leaves the `USBI_EVENT_HOTPLUG_CB_DEREGISTERED` set in the pending event flags.  This is addressed in #856.

mcuee, 2021-01-22:
@matallui Please try out #856 patch to see if the issue goes away. Thanks.

matallui, 2021-01-22:
Yes! That change seems to have fixed the issue. Thanks you guys!

mcuee, 2021-02-07:
Pending #856 to be merged.

mcuee, 2021-06-03:
Close this now that #856 has been merged.

closed 2021-06-03 by mcuee