libusb/libusb #860: libusb_interrupt_transfer return-1

wangdaao, 2021-01-24:
When I use libusb1.0.24 to read and write with the mouse, I always return - 1. The error message is as follows:
libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] access denied。

wangdaao, 2021-01-24:
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.006621] [00006f90] libusb: debug [libusb_get_device_list]  
[ 0.006748] [0000671c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.016820] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [77]
[ 0.017156] [00006f90] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.017185] [00006f90] libusb: debug [get_api_type] matched driver name against HID API
[ 0.017189] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.017283] [00006f90] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017308] [00006f90] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017311] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [55]
[ 0.017404] [00006f90] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017429] [00006f90] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017432] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [56]
[ 0.017544] [00006f90] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.017575] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.017679] [00006f90] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.017704] [00006f90] libusb: debug [get_api_type] matched driver name against HID API
[ 0.017707] [00006f90] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.017856] [00006f90] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A2AF&SUBSYS_50071458&REV_00\3&11583659&0&A0' bus number 1
[ 0.018642] [00006f90] libusb: debug [winusb_get_device_list] found existing device for session [56]
[ 0.018691] [00006f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018702] [00006f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 107 bytes)
[ 0.018712] [00006f90] libusb: debug [init_device] (bus: 1, addr: 34, depth: 1, port: 10): 'USB\VID_1FF7&PID_0001\00000400-511D4815-58F14000-02B09709'
[ 0.018770] [00006f90] libusb: debug [winusb_get_device_list] found existing device for session [55]
[ 0.018797] [00006f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018805] [00006f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.018811] [00006f90] libusb: debug [init_device] (bus: 1, addr: 32, depth: 1, port: 5): 'USB\VID_1C4F&PID_0026\5&2108AD5D&0&5'
[ 0.018869] [00006f90] libusb: debug [winusb_get_device_list] found existing device for session [54]
[ 0.018894] [00006f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018901] [00006f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.018908] [00006f90] libusb: debug [init_device] (bus: 1, addr: 41, depth: 1, port: 7): 'USB\VID_046D&PID_C077\5&2108AD5D&0&7'
[ 0.019036] [00006f90] libusb: debug [winusb_get_device_list] found existing device for session [58]
[ 0.019062] [00006f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019070] [00006f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.019076] [00006f90] libusb: debug [init_device] (bus: 1, addr: 42, depth: 1, port: 8): 'USB\VID_0000&PID_0538\5&2108AD5D&0&8'
[ 0.019132] [00006f90] libusb: debug [winusb_get_device_list] found existing device for session [57]
[ 0.019157] [00006f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019164] [00006f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 32 bytes)
[ 0.019170] [00006f90] libusb: debug [init_device] (bus: 1, addr: 23, depth: 1, port: 3): 'USB\VID_23A9&PID_EF18\5&2108AD5D&0&3'
[ 0.019258] [00006f90] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&92B3C53&0&0' reports 20 ports
[ 0.019298] [00006f90] libusb: warning [init_root_hub] could not get node connection information (V2) for root hub 'USB\ROOT_HUB30\4&92B3C53&0&0' port 11: [87] 参数错误。
[ 0.019304] [00006f90] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&92B3C53&0&0'
[ 0.019531] [00006f90] libusb: debug [winusb_get_device_list] setting composite interface for [56]:
[ 0.019536] [00006f90] libusb: debug [set_composite_interface] interface[0] = \\?\HID#VID_1FF7&PID_0001&MI_00&COL03#7&375D38D&0&0002#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.019552] [00006f90] libusb: debug [winusb_get_device_list] setting HID interface for [58]:
[ 0.019554] [00006f90] libusb: debug [set_hid_interface] interface[0] = \\?\HID#VID_0000&PID_0538#6&1047CE7A&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.019567] [00006f90] libusb: debug [winusb_get_device_list] setting HID interface for [54]:
[ 0.019569] [00006f90] libusb: debug [set_hid_interface] interface[0] = \\?\HID#VID_046D&PID_C077#6&F2E2A79&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.019598] [00006f90] libusb: debug [winusb_get_device_list] setting composite interface for [55]:
[ 0.019600] [00006f90] libusb: debug [set_composite_interface] interface[1] = \\?\HID#VID_1C4F&PID_0026&MI_01&COL02#7&29899891&0&0001#{4D1E55B2-F16F-11CF-88CB-001111000030}
[... cut ...]
[ 1.991752] [00006f90] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 1.991757] [00006f90] libusb: debug [hid_submit_bulk_transfer] matched endpoint 81 with interface 0
[ 1.991760] [00006f90] libusb: debug [hid_submit_bulk_transfer] reading 5 bytes (report ID: 0x00)
[ 1.991774] [00006f90] libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] access denied。
[ 1.991776] [00006f90] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.991780] [00006f90] libusb: debug [libusb_free_transfer] transfer 0591F038

mcuee, 2021-01-24:
Do not use libusb with an mouse. Read the libusb wiki. 

And please use libusb mailing list for questions.

closed 2021-01-24 by mcuee

wangdaao, 2021-01-24:
But I read the mouse, keyboard, U disk and other USB devices are the same situation

mcuee, 2021-01-24:
Again, read the Wiki. You need to install the supported driver (eg: WinUSB) in order to use libusb under Windows.

wangdaao, 2021-01-24:
Yeah, I've got it. Thank you