libusb/libusb #861: libusb cannot open device in windows

jimiking, 2021-01-24:
when i use libusb for listdevice i not have problem. but when i want send data i get error CANNOT OPEN DEVICE

i want fix this error with programmity( not install custom driver with zidag and etc)

i use windows 7-10 (32-64 bit)
and want send data samsung galaxy devices
thank

mcuee, 2021-01-25:
Read the Wiki. You need to install the supported driver (eg: WinUSB) in order to use libusb under Windows.
https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows

And please use libusb mailing list for questions like this.

closed 2021-01-25 by mcuee

mcuee, 2021-01-25:
If you want to integrate driver installation into your application, you can look into libwdi/zadig here.
https://github.com/pbatard/libwdi/wiki/FAQ#How_can_I_use_libwdi_to_install_a_specific_USB_device_with_an_installer