libusb/libusb #862: Android JNI USB backend

xloem, 2021-01-28:
It's a little hard to quickly review, but it looks like libusb attempts to directly access device files on Android.

A compatible way to support Android would be to use the JNI to do it via Android's supported APIs.

I found somebody started this work in 2012 at https://github.com/SpecLad/libusb-android/blob/android/android/LibusbSupport/src/org/libusb/UsbHelper.java .  There are likely other starts around the web.

In Python, Android/USB is functionally smoothly nowadays via the https://github.com/jacklinquan/usb4a project.

avantgardnerio, 2021-01-28:
If you open the device via the regular Android API, you can get the FileDescriptor and use it in libusb with `libusb_wrap_sys_device()`. A bit of a hack, but it works.

mcuee, 2021-01-29:
Please help to review the pull request here.
https://github.com/libusb/libusb/pull/830

I do not know much about the inner working of Android but the above pull request will update the Android Readme file and provide an example of using libusb_wrap_sys_device().

mcuee, 2021-01-29:
I have also updated the new Wiki page based on the current situation. libusb_wrap_sys_device() is not really considered as a hack. I think it is probably the better way now to use libusb under Android than using unsupported forks. But of course patches are welcome to have better libusb support under Android.

mcuee, 2021-01-29:
I have deleted the following entry from the wiki pages as they are unsupported.
- https://github.com/SpecLad/libusb-android (uses Java interface, not updated since 2012)
- https://github.com/kuldeepdhaka/libusb/tree/android-open2
- https://gitlab.com/madresistor/libusb/tree/android

mcuee, 2021-02-05:
Close this for now.
You can continue the discussion in ticket #717.

If you think #717 is different, please re-open this one.

closed 2021-02-05 by mcuee