libusb/libusb #863: Filter out non-libusb I/O completions breaks Winusb HID API

evosystems, 2021-01-28:
commit b51c743e4210756a98f4f60c69a34745e4b27a55 that solved #844 breaks winusb HID API

In windows_winusb.c
 1. in all hid functions usbi_get_device_handle_priv should also be replaced with get_winusb_device_handle_priv
 2. in hid_open() this change is also necessary - "windows_open(dev" replaced with "windows_open(dev_handle"
There might be some other modifications necessary ...

gsmdenis, 2021-01-30:
have same problem , very bad

mcuee, 2021-01-30:
@gsmdenis Thanks for the report but please do not use words like "very bad".

mcuee, 2021-01-30:
BTW, recommendation is to use HIDAPI for Windows.

Actually I have an RFC to remove HID backend from libusb: https://github.com/libusb/libusb/issues/758

mcuee, 2021-01-30:
In this case, many thanks to @evosystems to provide the root cause analysis and @gsmdenis to provide the confirmation. Hopefully the fix will be simple.

Still in any case, no change with the recommendation to move from libusb to HIDAPI for Windows users.

evosystems, 2021-01-30:
Thanks @mcuee 
In the initial post, I did not really pinpoint the exact root cause of the problem. Meanwhile, I did more tests, and seems the problem is somehow related to calling usbi_signal_transfer_completion(itransfer); 

This is normally called in windows_iocp_thread after checking that the IO operation originated in libusb but is also called in windows_force_sync_completion() for a successful winusbx_submit_control_transfer and hid_submit_control_transfer.
What I did was to search the handle_priv->active_transfers list for the transfer and only act on it if found as the windows_iocp_thread might have actually did that already.

So i just added 
`bool found;
	libusb_device_handle* dev_handle = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer)->dev_handle;
	struct windows_device_handle_priv* handle_priv;
	usbi_mutex_lock(&dev_handle->lock);
	handle_priv = usbi_get_device_handle_priv(dev_handle);
	found = false;
	
	list_for_each_entry(transfer_priv, &handle_priv->active_transfers, list, struct windows_transfer_priv) {
		if (overlapped == &transfer_priv->overlapped) {
			// This OVERLAPPED belongs to us, remove the transfer from the device handle's list
			list_del(&transfer_priv->list);
			found = true;
			break;
		}
	}
	
	if (!found) {
		usbi_dbg("ignoring overlapped %p for handle %p (device %u.%u)",
			overlapped, dev_handle, dev_handle->dev->bus_number, dev_handle->dev->device_address);
		usbi_mutex_unlock(&dev_handle->lock);
		return;
	}

to `windows_force_sync_completion before calling usbi_signal_transfer_completion(itransfer);
Hope this will help too.

dickens, 2021-02-01:
This should be resolved.  Apologies for the inconvenience!

commit a108bb8e (2021-02-01): Windows: Fix HID and synchronous transfer completion

Commit b51c743e42 ("Windows: Filter out non-libusb I/O completions")
causes problems with the HID backend functions and synchronous transfer
completions. Address these by fetching the correct pointer in the HID
functions and posting an I/O completion for synchronous transfers.

Closes #863
closed 2021-02-01 by dickens