libusb/libusb #864: SYSFS_DEVICE_PATH issue

yshteinm, 2021-01-30:
In Android 10 (and I think 9 too) the follow doesn't work but works on Android 8.

```
static int sysfs_get_device_list(struct libusb_context *ctx)
{
	DIR *devices = opendir(SYSFS_DEVICE_PATH);
```

mcuee, 2021-01-30:
https://github.com/libusb/libusb/pull/830
Please take a look at the above pull request and example.

yshteinm, 2021-01-30:
Sorry, I am not quite sure which part to look. There are few comments there. I would appreciate if you please can be more specific about steps to take or specific comment to look?

mcuee, 2021-01-30:
Please check the example in the pull request.
https://github.com/libusb/libusb/pull/830/commits/fb9c4082a0d1a1f787ab62d124e2e9fa7bda6acc

mcuee, 2021-02-05:
@yshteinm Does the above address your question? If yes I will close this issue.

yshteinm, 2021-02-05:
So far I have not managed to make it work. You probably can close the issue if it works for other. Anyone can confirm it works?

mcuee, 2021-02-07:
I will close this issue and please use #717 to have the futher discussion.

closed 2021-02-07 by mcuee