libusb/libusb #866: Getting the serial number without opening the device handle

mcuee, 2021-02-02:
Reference:
There are a few past discussions and recent discussions on getting the serial number without opening the device handle

Past discussion: 
https://marc.info/?t=142939771300004&r=1&w=4
https://sourceforge.net/p/libusb/mailman/message/33894596/

Recent discussion: Getting a serial number without opening a handle
Feb 2019: Accessing string descriptors when can't do libusb_open
https://marc.info/?t=155008592500001&r=1&w=4

Feb 2021: Detecting Multiple Devices Of Same Type
https://marc.info/?t=161158709600002&r=1&w=4

xloem, 2021-02-13:
I visited hjmallon's repository at https://github.com/hjmallon/libusb but did not find the fork mentioned at https://marc.info/?l=libusb-devel&m=155025463511502&w=4

mcuee, 2021-02-14:
I do not think the one materialized in the end. 

libusbp from David Grayson claims to have this feature.
https://github.com/pololu/libusbp

hjmallon, 2021-02-15:
@xloem I didn't think anyone was very interested. 😄 Let me do some git archeology and PR what I have this week!

hjmallon, 2021-02-15:
Here is my change from 2 years ago. I have rebased it on master but this has caused some things to need changing. If you are interested in helping test is (I have no Windows or linux machines atm) please comment on the PR.

https://github.com/libusb/libusb/pull/875

mliberty1, 2024-07-07:
I agree that this is an important feature.  I am using PR https://github.com/libusb/libusb/pull/875 in my code, and it would be great to have this upstreamed.

It seems like the API has been the sticking point in integrating this feature.  These USB string descriptors are processed by the OS, so the USB UTF-16 and ASCII formats are not guaranteed.  Perhaps a UTF-8 format would be better?  How about this API?

```
/** \ingroup libusb_desc
 * The OS string type for libusb_get_os_string().
 */
enum libusb_os_string_type_e {
   LIBUSB_OS_STRING_TYPE_MANUFACTURER,
   LIBUSB_OS_STRING_TYPE_PRODUCT,
   LIBUSB_OS_STRING_TYPE_SERIAL_NUMBER,
};

/** \ingroup libusb_desc
 * Retrieve a device string from the operating system.
 * This function works when the device is still closed since it relies
 * on the operating system to fetch the USB string descriptor.
 * Since the USB string descriptor could be processed by the OS,
 * this function returns a UTF-8 encoded string.  The string will
 * be returned untranslated or in the default OS language 
 * when supported by the OS and USB device.
 *
 * \param dev_handle a device handle
 * \param string_type the string type to retrieve
 * \returns the Unicode UTF-8 encoded string.
 * \see libusb_get_string_descriptor()
 * \see libusb_get_string_descriptor_ascii()
 */
const char * libusb_get_os_string(libusb_device *dev, enum libusb_os_string_type_e string_type);  // return UTF-8 or NULL
```

On the first call for each string_type, the string is cached in the dev instance.  It is freed automatically when the dev instance is freed.

The initial implementation can return a valid string for `LIBUSB_OS_STRING_TYPE_SERIAL_NUMBER` and NULL for all others.

Youw, 2024-07-08:
> It seems like the API has been the sticking point in integrating this feature.

I recall the same.

> How about this API?

I really like the idea. Open a PR - I'll help getting it promoted.

---
Some thoughts on the API:

1)

> `libusb_get_os_string`

This seem more like `libusb_get_device_string`.

Same about the enum name/values.

2)

Lets be more consistent on the API, and return `int` as the size of a string or a return code, and have the `const char *string, int length` as the out parameter of the function.

mliberty1, 2024-07-08:
Hi @Youw - Thanks for the quick feedback!

1.
Regarding the name, libusb is a classic C-style object-oriented API.  Unfortunately, the naming convention does not indicate to which class the function / method belongs.  Without looking at the first argument type, you don't know that `libusb_get_configuration` takes a `libusb_device_handle` while `libusb_get_config_descriptor` takes a `libusb_device`.  A more normal way to have a clean namespace would be to have a separate prefix, like `libusb_devh_get_configuration` and `libusb_dev_get_config_descriptor`.

For this function, we need to follow the current libusb naming style while not claiming the entire libusb namespace.   With the current libusb naming convention, I think that this is the `get_os_string` method for  `libusb_device`.  So, `libusb_get_device_string` seems reasonable to differentiate `libusb_device` from a string function on `libusb_context` or `libusb_device_handle`.  Would `libusb_get_device_os_string` be even clearer?

2.
Returning a negative error code or length makes sense.  I can see supporting `LIBUSB_ERROR_NO_DEVICE` and `LIBUSB_ERROR_INVALID_PARAM` for invalid string types.  For valid string types that are not found or not supported on that backend, it is probably better to return a length 0 empty string or NULL rather than `LIBUSB_ERROR_NOT_SUPPORTED`.  What do you think?

3. 
The `libusb` API is not consistent on caller-allocated buffers vs library-allocated buffers.  Most of the descriptor operations are library-allocated with a corresponding free, like `libusb_free_config_descriptor`.  The `libusb_get_string_descriptor` calling convention requires the user to call it twice, once to get the length and again to get the value after allocating the correctly sized buffer.  I personally find this more annoying than a separate free function.

My initial proposal was to use a library-allocated buffer that is then cached in libusb_device and freed with the libusb_device.  The potential downside of this approach is use-after-free if the caller holds onto the string after `libusb_unref_device` causes libusb_device to free.  So, even if libusb_device caches the string, it is probably safer to copy on call than risk the use-after-free.  The potential of leaking a few bytes because the user does not call free is better than risking a potential crash.  So, it's probably better to make an explicit free rather than implicit.  How about this:

```
/** \ingroup libusb_desc
 * The OS string type for libusb_get_device_os_string().
 */
enum libusb_device_os_string_type_e {
   LIBUSB_DEVICE_OS_STRING_TYPE_MANUFACTURER,
   LIBUSB_DEVICE_OS_STRING_TYPE_PRODUCT,
   LIBUSB_DEVICE_OS_STRING_TYPE_SERIAL_NUMBER,
};

/** \ingroup libusb_desc
 * Retrieve a device string from the operating system.
 * This function / method works when the device is still closed since it relies
 * on the operating system to fetch the USB string descriptor.
 * Since the USB string descriptor could be processed by the OS,
 * this function returns a UTF-8 encoded string.  The string will
 * be returned untranslated or in the default OS language 
 * when supported by the OS and USB device.
 *
 * \param dev_handle a device handle
 * \param string_type the string type to retrieve
 * \param data The UTF-8 encoded string.  
 *    NULL if the return value is less than zero.
 *    Must be freed with libusb_free_device_os_string() after use.
 * \returns an error code or the string length in bytes excluding the null terminator.
 * \see libusb_get_string_descriptor()
 * \see libusb_get_string_descriptor_ascii()
 */
int libusb_get_device_os_string(libusb_device *dev, enum libusb_device_os_string_type_e string_type, char ** data);

/**\ingroup libusb_desc
 * Free a device OS string.
 *
 * \param data The string returned by libusb_get_device_os_string().
 */
void libusb_free_device_os_string(char * data);
```

Youw, 2024-07-08:
1)

You're right that overall naming is not strictly consistent or explicit or 100% self-descriptive.
On the other hand - there has never been obvious problems with it (AFAIK).

I don't really like the `_os_` part of the name here, as it more describes "how do I get it" (using some OS API) and not "what it really is" (some known characteristic of the USB device as a string).
When I suggested `_device_` I wasn't trying to imply that the argument of the function is `libusb_device`, but rather that we're trying to have a characteristic of the "USB Device" as an abstract entity.

Just some thoughs, having:
```c
enum libusb_device_string_type_e {
   LIBUSB_DEVICE_STRING_TYPE_MANUFACTURER,
   LIBUSB_DEVICE_STRING_TYPE_PRODUCT,
   LIBUSB_DEVICE_STRING_TYPE_SERIAL_NUMBER,
};
```
Would also make possible some potential (maybe convenient for the use) overload (no I don't think we need it right now, it is more as an example):
```
int libusb_get_string_from_string_descriptors(libusb_device_handle *dev_handle, enum libusb_device_string_type_e, uint16_t langid, char *string, int length);
```
I.e. the convenience is to not require getting device descriptor and string indexes explicitly for the user.
And the `enum libusb_device_string_type_e` name makese sense here.

2)

> For valid string types that are not found or not supported on that backend, it is probably better to return a length 0 empty string or NULL rather than LIBUSB_ERROR_NOT_SUPPORTED

Sounds reasonable.

3)

> once to get the length and again to get the value after allocating the correctly sized buffer

This should be hardly an issue, since any USB descriptor, including String descriptor is limited by 255 bytes of size, so by my taste the easiest/recommended way to be:
```c
char buffer[256];
int ret = libusb_get_device_os_string(dev, LIBUSB_DEVICE_OS_STRING_TYPE_MANUFACTURER, buffer, sizeof(buffer));
```
And even if it is to be used with something higher-level, like C++, it is still best (to my taste) way to do:
```cpp
if (ret < 0) {
    // handle error, maybe throw an exception
}

return std::string(buffer, size_t(ret));
```
With convenient usage, minimum overhead and low chance accidentally misuse the API or cause an undefined/unspecified behavior/etc.

mliberty1, 2024-07-08:
1. sounds good!  `libusb_get_device_string` it is!
2. we agree.
3. Taking advantage of the maximum string length for USB descriptors makes sense.  I will make sure the documentation contains this recommendation / example for how best to use the function.

I will put together a pull request.  Not sure how quick this will go and how much time I will have, but hopefully I will have something in the next week.