libusb/libusb #868: Automating permissions on Android

xloem, 2021-02-05:
After https://github.com/libusb/libusb/issues/862#issuecomment-773726901 it looks like Android is somewhat supported by libusb, but the user still needs to write some JNI and Java to map between the devices listed in the android api and the actual device handles or files.  I'm opening this ticket just as a way to track and discuss the idea of including that code within libusb itself, as some forks have done in the past.

It should all be doable within C++ if the user passes a JNIEnv* pointer in, I think.

mcuee, 2021-02-09:
Please refer to #717 as well.
https://github.com/libusb/libusb/issues/717

xloem, 2021-02-11:
I did some work towards this.
Making java calls from inside libusb to enumerate devices: https://github.com/xloem/libusb/blob/android/libusb/os/linux_usbfs.c#L493
Adding an option to provide a JNIEnv pointer: https://github.com/xloem/libusb/blob/android/libusb/os/linux_usbfs.c#L472

I'm guessing the total work to implement this would be maybe half the work of implementing a new platform.  The permissions situation means the JNI should be used to enumerate configurations, etc, and the formats are all different from the linux ones.  Making a new driver backend for android, and factoring out any similarities with the existing linux backend, would be an organised way to let permissions only be requested when the device is actually opened.

Requesting permissions can take a long delay, where the user can consider an asynchronous dialog and confirm or reject it.

mcuee, 2021-06-05:
Please refer to #830 first for an propoased updated README if you are interested in using libusb under Android.

Then please refer to #874 for better Android support.

mcuee, 2021-07-26:
There is another approach as mentioned by @normanr here.
https://github.com/libusb/libusb/pull/874#issuecomment-873322802

==========
> What normanr is describing appears to be at https://github.com/green-green-avk/libusb-1.0.23-android-helper-service-patch/blob/master/libusb/os/linux_android_helper.c . It's unfortunate that alex initialized a new repository rather than forking libusb to track changes, but it is not too hard to diff them.

There is a properly forked branch available at https://github.com/green-green-avk/libusb/tree/v1.0.23-android-libusbmanager (found via https://github.com/termux/termux-api/issues/349#issuecomment-701735271)

> Termux provides access to some of the java usb api via termux-api: https://github.com/termux/termux-api/blob/master/app/src/main/java/com/termux/api/UsbAPI.java .
> 
> So the present protocols differ between the terminal applications, but both already exist and could be added to the main repo with some work.

Correct. The termux ones seem to be very thin and only supports opening a usb device via an fd, where-as the helper-service properly supports enumeration and hot plug. In fact I wrote a bridge to shim the termux api as the helper-service api (https://github.com/normanr/termux-libusb-bridge).

I have a personal preference that the libusb library shouldn't require any java/jni things, and the java/jni support code could be refactored to work as a helper in the same way as termux/helper-service do, but I'm not a libusb maintainer, so please don't take my preference as a requirement for this PR.

mcuee, 2021-07-26:
Relevant discussion: termux pyusb enhancement
https://github.com/pyusb/pyusb/pull/287

mcuee, 2026-03-12:
Latest PR #1164 
* #1164 

Hopefully more users can help to move the above PR forward.

I will need to learn how to test on Android side...

mcuee, 2026-05-31:
BTW, there is an official libusb NDK github action here.
https://github.com/libusb/libusb/blob/master/.github/workflows/android.yml

You can download the build binaries from each run.
Example: latest run on the master branch.
https://github.com/libusb/libusb/actions/runs/26518291496
```
libusb-android-arm64-v8a-api21
97.9 KB	
sha256:74ca589b3694558ccdade85e8d7312b75cea4ba468e4c9910ded897c4bd0650c

libusb-android-armeabi-v7a-api21
83.3 KB	
sha256:d9524bfcb94d26626ffbe60d9a382b12e18e411652317e3beab52f26d10e7ebd

libusb-android-x86_64-api21
97.6 KB	
sha256:1bdc2706af1fb39155208d02891437772c141bdb393c2bb403eb1291ccf6f9c6
```