libusb/libusb #869: Fail to build on older hosts

otavio, 2021-02-06:
We're having issues building 1.0.24 on old Linux releases. We found it using the [Rust bindings](https://github.com/a1ien/libusb1-sys) in a Linux and Windows driver.

We reported it [here](https://github.com/a1ien/libusb1-sys/issues/28) and it seems to be caused by changes in the include order of `sys/socket.h`. The hook can be seen at https://github.com/libusb/libusb/commit/aaff15d48d1b8555aabf012b06bf39bf8aa4768a#diff-a2ae0398f3555101d572c5c02de627b27517a5b1fb5655dfb896aa300159e0b9L40-L41

It need to be preserved as:
```
#include <sys/socket.h>
#include <linux/netlink.h>
```

mcuee, 2021-02-08:
How old the Linux version you are trying to support?

The link below refers to another link dated 2011.
https://github.com/a1ien/libusb1-sys/issues/28
Link to the 2011 post.
https://www.spinics.net/lists/netdev/msg171723.html

Take note old Linux version support has been removed in 1.0.24.

otavio, 2021-02-08:
In our case, we used Ubuntu 16.04 as the build environment.

a1ien, 2021-02-08:
It's strange in ubuntu 16.04 looks like correct header https://packages.ubuntu.com/xenial/linux-libc-dev. Can you upload /usr/include/linux/netlink.h
what version of `linux-libc-dev` (`apt-cache policy linux-libc-dev`)?

commit 6a394511 (2021-02-08): linux_netlink: Change order of header includes per netlink(7)

Closes #869
closed 2021-02-08 by dickens