libusb/libusb #870: Windows : libusb_get_device_list returns -1 when i plug the device

andrejmz, 2021-02-06:
Hi

I am trying to retrieve data from an oscilloscope.
Before plugging the oscilloscope i can list the other devices fine:

[image]

But once i plug the oscilloscope  libusb_get_device_list returns -1

mcuee, 2021-02-06:
Please provide the debug list by setting LIBUSB_DEBUG=4, before and after plug in the oscilloscope. 

Please also provide the info about the oscilloscope (USB descriptor) by posting the output from USBView.

andrejmz, 2021-02-06:
Before i plug the device:

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.038893] [00007a6c] libusb: debug [libusb_get_device_list]
[ 0.049863] [00007a6c] libusb: debug [winusb_get_device_list] allocating new device for session [5F]
[ 0.050860] [00007a6c] libusb: debug [winusb_get_device_list] allocating new device for session [60]
[ 0.050860] [00007a6c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.050860] [00007a6c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.050860] [00007a6c] libusb: debug [winusb_get_device_list] allocating new device for session [4F]
[ 0.051859] [00007a6c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.051859] [00007a6c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.051859] [00007a6c] libusb: debug [winusb_get_device_list] allocating new device for session [50]
[ 0.051859] [00007a6c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A3AF&SUBSYS_50071458&REV_00\3&11583659&0&A0' bus number 1
[ 0.052855] [00007a6c] libusb: debug [winusb_get_device_list] found existing device for session [60]
[ 0.052855] [00007a6c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.052855] [00007a6c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.053852] [00007a6c] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 9): 'USB\VID_05E3&PID_0608\5&201B4142&0&9'
[ 0.053852] [00007a6c] libusb: debug [winusb_get_device_list] found existing device for session [50]
[ 0.053852] [00007a6c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.053852] [00007a6c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
[ 0.053852] [00007a6c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 2): 'USB\VID_046D&PID_C52B\5&201B4142&0&2'
[ 0.053852] [00007a6c] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&17DA066&0&0'
[ 0.053852] [00007a6c] libusb: debug [winusb_get_device_list] found existing device for session [4F]
[ 0.053852] [00007a6c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.053852] [00007a6c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
[ 0.053852] [00007a6c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 3): 'USB\VID_04F2&PID_0981\5&201B4142&0&3'
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [4F]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[2] = \\.\HID#VID_04F2&PID_0981&MI_02#7&31AD62BF&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] = \\.\HID#VID_046D&PID_C52B&MI_01&COL01#7&2AD0F367&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_046D&PID_C52B&MI_01&COL03\7&2AD0F367&0&0002
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [4F]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] = \\.\HID#VID_04F2&PID_0981&MI_01&COL02#7&85222EE&0&0001#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_046D&PID_C52B&MI_01&COL02\7&2AD0F367&0&0001
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[0] = \\.\HID#VID_046D&PID_C52B&MI_00#7&17A0D729&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_046D&PID_C52B&MI_01&COL04\7&2AD0F367&0&0003
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [4F]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[0] = \\.\HID#VID_04F2&PID_0981&MI_00#7&30BCD0B7&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [4F]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_04F2&PID_0981&MI_01&COL01\7&85222EE&0&0000
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[ 0.054849] [00007a6c] libusb: debug [set_composite_interface] interface[2] = \\.\HID#VID_046D&PID_C52B&MI_02&COL01#7&266459E&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.054849] [00007a6c] libusb: debug [winusb_get_device_list] setting composite interface for [50]:
[... cut ...]
[ 0.046874] [0000792c] libusb: debug [cache_config_descriptors] cached config descriptor 9 (bConfigurationValue=1, 67 bytes)
[ 0.046874] [0000792c] libusb: error [usbi_sanitize_device] too many configurations
[ 0.047872] [0000792c] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.047872] [0000792c] libusb: debug [libusb_unref_device] destroy device 1.11
[ 0.047872] [0000792c] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.047872] [0000792c] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.047872] [0000792c] libusb: debug [libusb_unref_device] destroy device 1.0

Best regards

mcuee, 2021-02-06:
Looks like the device descriptors may have an issue. Unlikely a device will have 10 configurations. Please post the output of USBView.
https://ftdichip.com/utilities/#microsoft-usbview

andrejmz, 2021-02-06:
USb view unplugged:

[image]

USb view plugged:

[image]

andrejmz, 2021-02-06:
Too bad that the one device i am interested in is the one sc***ing everything ;-)

andrejmz, 2021-02-06:
Any suggestion ?

mcuee, 2021-02-07:
Please post the right panel information of the device. 
By the way, if your device is using usbser.sys (USB to serial converter driver for USB CDC ACM device) then you should not use libusb but rather you should use the serial API.

andrejmz, 2021-02-07:
Ok i'll try the serial api thanks !!

mcuee, 2021-02-07:
The device has a problem with the descriptor "bNumConfigurations:   0x0A". So please contact the device vendor to fix the issue. 

For your application, please use serial API.

dickens, 2021-02-08:
FYI the error returned by `libusb_get_device_list()` may be resolved by 851ff404003a26c1e3428b9a1332227173cc124b.

andrejmz, 2021-02-08:
Yes

But it seems the device works fine with the software provided with it, despite everything. 
How is it doing it i don't know.

Best regards

mcuee, 2021-02-09:
Windows driver usually only care about the default configuration. So in this case, your device will be set the first configuration.

mcuee, 2021-02-09:
@andrejmz Just want to confirm with you, are you using the latest 1.0.24 release when you generate the debug log.

mcuee, 2021-02-24:
Close for now. Please reopen when you provide more info.

closed 2021-02-24 by mcuee