libusb/libusb #871: libusb_bulk_transfer fails if called inside thread and not using WaitForSingleObject

smx12, 2021-02-10:
I'm writing a DLL that needs to have a non-blocking function to perform bulk transfers. So I create the function start() which creates a thread. The thread functions calls libusb_bulk_transfer but the functions behaves erratically, Here is some pseudo-code of what I'm doing:

`USB_RESULT DLL_EXPORT start(void)
{
  /* Create Thread */
  hThread = CreateThread(NULL, 0, sampleThread ...);

  if(hThread == NULL)
    return ERROR...

  stopSampling = false;

  WaitForSingleObject(hThread, INFINITE); // I need to add this or else the bulk transfer wont work properly
}

USB_RESULT DLL_EXPORT stop(void)
{
   stopSampling = true;
}

DWORD WINAPI samplesThread(void* data)
{

  do
  {
    // Send device command to send samples
    r = libusb_bulk_transfer(gDeviceHandle, OUT_ENDPOINT_ADDR, buffer, 64, &wr, 0);
    if(r == OK)
    {
       // Store samples in vector
       r = libusb_bulk_transfer(gDeviceHandle, IN_ENDPOINT_ADDR, gVector, SAMPLES_BUFFER_SIZE, &wr, 0);
       if(r != OK) || (wr != SAMPLES_BUFFER_SIZE)
       {
          break
       }
    }
    else
    {
       break;
    }

  }while(stopSampling == false);

  return 0;
}`

I need to call WaitForSingleObject so that bulk_transfer runs fine but by doing so I'm making start() blocking. Is there any way to solve this?

mcuee, 2021-02-11:
Try using async API and that should help. Sync API is blocking in nature.

smx12, 2021-02-11:
Thanks for your reply, yeah I tried using async and I have the same problems. Is there some example about using async that I could use? maybe I'm doing something wrong.

mcuee, 2021-02-11:
https://github.com/libusb/libusb/tree/master/examples
You can take a look at dpfp.c to see if that gives you some ideas.

Or the simpler sam3u_benchmark.c which is a single threaded application.

mcuee, 2021-02-11:
This libusb api documentation page may help as well.
http://libusb.sourceforge.net/api-1.0/libusb_mtasync.html

closed 2021-02-11 by mcuee