libusb/libusb #873: No linking of the 32 bit libusb library on Windows 64

serj053, 2021-02-12:
Dear coders , I have an operating system Windows 10 , 64-bit, who can answer me why the 32-bit library is not linked on a 64-bit machine. I tried both the static library and the dynamic library linker returns an error like this- "undefined reference to 'libusb_get_ss_endpoint_companion_descriptor'". With respect to all .

mcuee, 2021-02-13:
Which compiler are you using?

Take note the Windows binary provided are more for convenience use and it may not cover all the compilers out there. You may have to build the binary by yourself using you compiler.

dickens, 2021-02-15:
I suspect your link command and/or library path is incorrect.  In any case, I can confirm that the provided libraries are exposing the symbol for `libusb_get_ss_endpoint_companion_descriptor()`, so I'm going to close this issue.  If you find new information suggesting a bug, feel free to re-open this.

closed 2021-02-15 by dickens