libusb/libusb #877: Access Denied

smikesmike, 2021-02-18:
Hi,

I've got a Bluetooth dongle, which is not visible in `hidapi`. I am using a MacbookPro with `Big Sur v11.2.1`

Here is the `lsusb` output:

                CSR8510 A10:

                  Product ID: 0x0001
                  Vendor ID: 0x	  (Cambridge Silicon Radio Ltd.)
                  Version: 88.91
                  Speed: Up to 12 Mb/s
                  Location ID: 0x14334000 / 7
                  Current Available (mA): 500
                  Current Required (mA): 100
                  Extra Operating Current (mA): 0

Do you have an idea how I could communicate with the dongle? If I use `libusb` I get an `access denied`.

BR
Marc

mcuee, 2021-02-18:
Please log the issue under HIDAPI.

mcuee, 2021-02-18:
FYI, libusb under macOS does not support USB HID device at all.

closed 2021-02-18 by mcuee