libusb/libusb #880: Windows: libusb_control_transfer

Lulz322, 2021-02-23:
**Yo, i want to recreate control transfer request to a mice as seems below**

[image]

I used this code:
```
        dev_handle = libusb_open_device_with_vid_pid(ctx, USB_VENDOR_ID, USB_PRODUCT_ID);

	libusb_control_transfer(dev_handle, 0x40, 0x01, 0x0100, 0x010a, NULL, 0, 0);
``` 
On linux it works correctly, but on windows there are a few errors:
```
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
libusb: warning [hid_submit_control_transfer] unsupported HID control request
libusb: warning [hid_submit_control_transfer] unsupported HID control request
libusb: warning [hid_submit_control_transfer] unsupported HID control request
libusb: error [composite_submit_control_transfer] no libusb supported interfaces to complete request
```

How to fix that?

diabolo38, 2021-02-23:
I kind of remember some back/end (HID/winusb) may limit control transfer toward interface 
and here if think your are doing a control targeting the device (req type => 0x40 for interface it would be 0xx1)

you can try changing your control toward interface or use different back-end 
I may have used  such dev control in the past with libusbk back end but am nto 100% sure

mcuee, 2021-02-24:
You can not use libusb Windows with a mouse. This is mentioned in the wiki.
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions
HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

If you want to try it, you can use WinUSB driver for your mouse and then try again. But then it will lose the mouse function.

closed 2021-02-24 by mcuee