libusb/libusb #881: Segmentation fault on 1.0.24

noppej, 2021-03-02:
Hi,  We experienced a segmentation fault that can be reproduced on 1.0.24 simply by running the `examples/listdevs.c` from this repo. I first experienced in on OSX Catalina but another user reproduced it on OSX Big Sur also. 

My XCode reports ... 
```
JackiMac cannot run dpfp_threaded.
Domain: IDEFoundationErrorDomain
Code: 3
Recovery Suggestion: JackiMac runs macOS 10.15.7, which is lower than dpfp_threaded’s minimum deployment target of 11.1. Change your project’s minimum deployment target or upgrade JackiMac’s version of macOS.
--
```

mcuee, 2021-03-03:
Is it only with the Xcode project? If that is the case, you can adjust by yourself.

Does this also happen if you use Homebrew? I am using Homebrew and no issues under my Mac Mini M1 Big Sur, either native M1 or Rosetta 2 mode.

Yatekii, 2021-03-03:
No, this is a bug within the lib apparently. Not the example. You can se it in the mentioned issue. The example/XCode project were just a means to verify that it is not an issue with the Rust crate.

mcuee, 2021-03-03:
Please provide the debug log by set LIBUSB_DEBUG=4.

mcuee, 2021-03-03:
Just an example run here with libusb 1.0.24 under Mac Mini M1 fully updated, using homebrew.

```

/Users/mcuee/build/libusb/libusb-1.0.24/build
mcuee@mcuees-Mac-mini build % ./examples/listdevs
1915:1025 (bus 2, device 10) path: 2.5.6
04d8:0033 (bus 2, device 9) path: 2.5.3
1a40:0201 (bus 2, device 8) path: 2.5
0bda:5411 (bus 2, device 7) path: 1.1
046d:c52b (bus 2, device 6) path: 2.2
05e3:0749 (bus 2, device 5) path: 5.1.1
0bda:5411 (bus 2, device 4) path: 1
0bda:0411 (bus 2, device 3) path: 5.1
1a40:0201 (bus 2, device 2) path: 2
0bda:0411 (bus 2, device 1) path: 5
```

mcuee, 2021-03-03:
I know nothing about Rust but anyway I just installed rust and probe-rs-cli list and nothing wrong here under my Mac Mini M1.

```
mcuee@mcuees-Mac-mini ~ % echo $PATH
/Users/mcuee/.cargo/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Library/Apple/usr/bin
mcuee@mcuees-Mac-mini ~ % probe-rs-cli list
No devices were found.
```

mcuee, 2021-03-03:
Example run.

```

mcuee@mcuees-Mac-mini rusb % cargo run --release --example libusb_info         
   Compiling pkg-config v0.3.19
   Compiling cc v1.0.67
   Compiling libc v0.2.87
   Compiling memchr v2.3.4
   Compiling once_cell v1.7.2
   Compiling rusb v0.8.0 (/Users/mcuee/build/libusb/examples/rusb)
   Compiling regex-syntax v0.6.22
   Compiling thread_local v1.1.3
   Compiling libusb1-sys v0.5.0
   Compiling aho-corasick v0.7.15
   Compiling regex v1.4.3
    Finished release [optimized] target(s) in 6.59s
     Running `target/release/examples/libusb_info`
libusb v1.0.24.11584
has capability? true
has hotplug? true
has HID access? false
supports detach kernel driver? false
```

Yatekii, 2021-03-03:
Yes, but that is an M1 :)

@noppej and @svenstucki both have it segfault on Catalina and BigSur (Intel HW).

It is fine on Linux. Can't tell for Windows but I haven't heard complaints.

mcuee, 2021-03-03:
I have used intel emulation as well without any issues. In any case, please post the libusb debug info using the examples like listdevs.

noppej, 2021-03-03:
@mcuee ... I'd be happy to send the logs, but can you please provide me with instructions (just the command line) on how to compile and run listdevs.c with homebrew?

svenstucki, 2021-03-03:
@noppej clone the repo, then do the following:

```
brew install autoconf
brew install autotools
./autogen.sh
make
```

You also need the xcode command line tools.

---

I ran the examples on my machine (MBP 2020, 11.1) and they don't segfault for me (built like described above - rev bda2344f5e7d460decd3968492faa7a5fb2797cd). However I do have a segfault when running `probe-rs-cli`. Will comment in the `rusb` repo as well.

```
$ ./examples/listdevs
0d28:0204 (bus 20, device 31) path: 3.1
0424:2422 (bus 20, device 30) path: 3
$ ./examples/dpfp_threaded
Could not find/open device
```

a1ien, 2021-03-03:
@svenstucki can you clone libusb-sys1 and update sub module
```
git clone --recurse-submodules git@github.com:a1ien/libusb1-sys.git
cd libusb1-sys/libusb
git checkout bda2344
cd ../
cargo run --example list_devices --features=vendored
```

svenstucki, 2021-03-03:
Hi @a1ien - just did that and works fine for me, it lists my devices.

noppej, 2021-03-03:
@mcuee I rebuilt with homebrew as you suggested, and now those tests run with the same results as @svenstucki 

```
libusb % ./examples/listdevs
0483:374e (bus 20, device 14) path: 3
05ac:8511 (bus 20, device 3) path: 5
libusb % ./examples/dpfp_threaded
Could not find/open device
```

noppej, 2021-03-03:
Is it fair to assume that the XCode failure was a red-herring, and that I can close this particular Issue for this repo?

Yatekii, 2021-03-03:
Yes. The issue is two clashing versions of libusb1-sys.

closed 2021-03-03 by noppej