libusb/libusb #883: application crash on init

ehrenmann1977, 2021-03-07:
I am trying to connect to a usb device but on the init the app crash, i have win 10 with and i disabled the updates to keep fast.

```
   libusb_context *ctx = NULL; //a libusb context for library intialization
    int r=libusb_init(&ctx);
    libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);   //initializing the library

```

mcuee, 2021-03-08:
If you want to use usbdk, you need to install usbdk. If not, please use WinUSB instead.

mcuee, 2021-03-10:
Please also post the debug log by running your application with LIBUSB_DEBUG=4 set.

mcuee, 2021-03-12:
Close for now. Please reopen if you can reproduce the issue and post the debug log.

closed 2021-03-12 by mcuee

ehrenmann1977, 2021-03-13:
the issue was compiling the library again with my compiler mingw

mcuee, 2021-03-14:
Thanks for the updates.