libusb/libusb #884: USB Composite Device [ Entity not found ]

thread-liu, 2021-03-08:
I have a high-speed USB device, which is a Composite Device (Winusb and CDC). I use libusb(python) to read/write winusb. When I
use the function `libusb.bulk_transfer`, I will receive an error code `Entity not found`. Please give me a help, Thanks!!

The USB description:
```powershell

        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01
bAlternateSetting        : 0x00
bNumEndpoints            : 0x02 (2 Endpoints)
bInterfaceClass          : 0x0A (CDC-Data)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)
Data (HexDump)           : 09 04 01 00 02 0A 00 00 00   

        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x02
bAlternateSetting        : 0x00
bNumEndpoints            : 0x02 (2 Endpoints)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)
Data (HexDump)           : 09 04 02 00 02 FF 00 00 00           

  Configuration:
    wTotalLength:         106
    bNumInterfaces:       3
    bConfigurationValue:  1
    iConfiguration:       0
    bmAttributes:         c0
    MaxPower:             50
    Interface:
      bInterfaceNumber:   0
      bAlternateSetting:  0
      bNumEndpoints:      1
      bInterfaceClass:    2
      bInterfaceSubClass: 2
      bInterfaceProtocol: 1
      iInterface:         0
      Endpoint:
        bEndpointAddress: 82
        bmAttributes:     03
        wMaxPacketSize:   8
        bInterval:        255
        bRefresh:         0
        bSynchAddress:    0
    Interface:
      bInterfaceNumber:   1
      bAlternateSetting:  0
      bNumEndpoints:      2
      bInterfaceClass:    10
      bInterfaceSubClass: 0
      bInterfaceProtocol: 0
      iInterface:         0
      Endpoint:
        bEndpointAddress: 01
        bmAttributes:     02
        wMaxPacketSize:   64
        bInterval:        0
        bRefresh:         0
        bSynchAddress:    0
      Endpoint:
        bEndpointAddress: 81
        bmAttributes:     02
        wMaxPacketSize:   64
        bInterval:        0
        bRefresh:         0
        bSynchAddress:    0
    Interface:
      bInterfaceNumber:   2
      bAlternateSetting:  0
      bNumEndpoints:      2
      bInterfaceClass:    255
      bInterfaceSubClass: 0
      bInterfaceProtocol: 0
      iInterface:         0
      Endpoint:
        bEndpointAddress: 03
        bmAttributes:     02
        wMaxPacketSize:   512
        bInterval:        0
        bRefresh:         0
        bSynchAddress:    0
      Endpoint:
        bEndpointAddress: 83
        bmAttributes:     02
        wMaxPacketSize:   512
        bInterval:        0
        bRefresh:         0
        bSynchAddress:    0
```

```python
import libusb
import ctypes

class usb_cmd_message(ctypes.Structure):
    _fields_ = [
    ("message_type",    ctypes.c_uint32, 32),
    ("length",          ctypes.c_uint32, 32),
]

msg = usb_cmd_message()
msg.message_type = 0x00
msg.length = 10

r = libusb.init(None)
dev = libusb.open_device_with_vid_pid(None, 0x1f3a, 0x218a)

r = libusb.claim_interface(dev, 0x02)
if r != 0:
    print("usb control transfer: {0}".format(libusb.strerror(libusb.error(r))))

buffer = ctypes.create_string_buffer(128)
usb_trans_size = ctypes.c_int()
r = libusb.control_transfer(dev, 0x21, 0xa, 0, 0, ctypes.cast(ctypes.pointer(msg),
                                                            ctypes.POINTER(ctypes.c_ubyte)), ctypes.sizeof(msg),
                            1000)
if r < 0:
    print("usb control transfer: {0}".format(libusb.strerror(libusb.error(r))))

r = libusb.bulk_transfer(dev, 0x03,
                         ctypes.cast(ctypes.pointer(buffer), ctypes.POINTER(ctypes.c_ubyte)),
                         128,
                         ctypes.byref(usb_trans_size), 1000)
if r < 0:
    print("libusb endpoint out: {0}".format(libusb.strerror(libusb.error(r)).decode()))

while True:
    r = libusb.bulk_transfer(dev, 0x83, ctypes.cast(ctypes.pointer(buffer),
                                         ctypes.POINTER(ctypes.c_ubyte)), 128,
                                         ctypes.byref(usb_trans_size), 1000)
    if r < 0:
        print("libusb endpoint in: {0}".format(libusb.strerror(libusb.error(r)).decode()))
        break
    else:
        print(buffer.value)

libusb.release_interface(dev, 2)
```

mcuee, 2021-03-08:
You need to use WinUSB as the driver for the full device.

Usually if you need both WinUSB and CDC, you need to use IAD.

mcuee, 2021-03-08:
Please refer to the following issue.
https://github.com/libusb/libusb/issues/85

thread-liu, 2021-03-08:
> use WinUSB as the driver for the full device

Thanks for your help, I can't understand, what the mean about `use WinUSB as the driver for the full device`?

mcuee, 2021-03-10:
Use WinUSB to replace the device driver of the USB composite parent.

Read the Zadig wiki here.
https://github.com/pbatard/libwdi/wiki/Zadig

mcuee, 2021-03-12:
I will treat this as a duplicate of #85 and close this issue for now. Please reopen if you think the issue is different.

closed 2021-03-12 by mcuee