libusb/libusb #886: Crash in usbi_signal_transfer_completion

bclay1297, 2021-03-10:
I am attempting to get a USB to DMX controller working with the following code.  Every time I get to the libusb_claim_interface call the program crashes in usbi_signal_transfer_completion which as you can see is not in the code below.  I assume this is invoked by a callback somewhere.  I pulled the basic code example from PiOpenLighting.  I stepped though the code in Visual Studio 2017 and everything seems ok until it crashes with an error "Exception thrown: read access violation.
dev_handle->dev was 0x8BAEA8B4800."

	r = libusb_init(NULL);

		libusb_device **device_list;
		size_t device_count = libusb_get_device_list(NULL, &device_list);

		int index = 4;  // index of the USB device

		libusb_device *usb_device = device_list[index];
		libusb_open(usb_device, &mUsbDevHandle);
		struct libusb_device_descriptor device_descriptor;
		libusb_get_device_descriptor(usb_device, &device_descriptor);

		unsigned int  debug_level = 0;  

		libusb_set_debug(NULL, debug_level);
//		libusb_set_configuration(mUsbDevHandle, 1); // set device to be I/O

		r = libusb_claim_interface(mUsbDevHandle, 0);

Any thoughts would be greatly appreciated.
Bruce

mcuee, 2021-03-10:
You need to mention more about the device (USB descriptors) and driver used (winusb or others). Then please generate the debug log by setting LIBUSB_DEBUG=4 and then run your program.

bclay1297, 2021-03-10:
I was using libusbK  but finally got winusb to work.  Not that I got my
code to work just that it stopped crashing.  I am pretty much shooting in
the dark here.  I self singed the uDMX drive that came with the USB to DMX
adapter (USB\VID_16C0&PID_05DC).  Actually it seems I am still using the
libusbK.dll.  I thought I had removed it but windows apparently reinstalled
it because it is showing up under driver details along with winusb.sys,
WdfCoInstaller01009.dll and WinUSBCoInstaller2.dll

So at this point it looks like I can initialize but the write always
fails.  I am trying to send just a few bytes to the DMX controller.  Is
bulk transfer the only option for writing to the device?

The device in question is  vid=16C0 pid=05DC

Source code and debug dump follow per you request

The initialize code contains

r = libusb_init(NULL);

unsigned int  debug_level = 4;  //LIBUSB_DEFAULT_DEBUG_LEVEL;

libusb_set_debug(NULL, debug_level);

mUsbDevHandle = libusb_open_device_with_vid_pid(NULL, vid, pid);

std::string manufacturer;
std::string product;
std::string serial;

if (mUsbDevHandle != NULL)
{
libusb_device **device_list;
size_t device_count = libusb_get_device_list(NULL, &device_list);

int index = 4;
libusb_device *usb_device = device_list[index];

// libusb_open(usb_device, &mUsbDevHandle);

struct libusb_device_descriptor device_descriptor;
libusb_get_device_descriptor(usb_device, &device_descriptor);

libusb_set_configuration(mUsbDevHandle, 1); // set device to be I/O

r = libusb_claim_interface(mUsbDevHandle, 0);
// if (r < 0)
{
// fprintf(stderr, "usb_claim_interface error %d\n", r);
// return 2;
}
// printf("Interface claimed\n");

endPoint = active_config(usb_device, mUsbDevHandle);

status = 0;
}

return(status);

and the write code contains

int UsbWrapper::UsbWrite(unsigned int numBytes, unsigned char *dataBuf)
{
int n, ret;

memcpy(mTransferBuf, dataBuf, numBytes);

//write transfer
//probably unsafe to use n twice...
[... 413 lines cut ...]
        Number of Interfaces: 1
        Length: 9
        Desc_Type: 2
        Config_index: 0
        Total length: 18
        Configuration Value: 1
        Configuration Attributes: 128
        MaxPower(mA): 250

EndPoint Descriptors:
        Size of EndPoint Descriptor: 255
        Type of Descriptor: 255
        Endpoint Address: 0x0ff
        Maximum Packet Size: ffff
        Attributes applied to Endpoint: 255
        Interval for Polling for data Tranfer: 255
[128.945264] [00005ae8] libusb: debug [libusb_alloc_transfer] transfer
000001DFE4206AF8
[128.945698] [00005ae8] libusb: debug [libusb_submit_transfer] transfer
000001DFE4206AF8
[128.947362] [00005ae8] libusb: error [winusbx_submit_bulk_transfer] unable
to match endpoint to an open interface - cancelling transfer
[128.949065] [00005ae8] libusb: debug [libusb_free_transfer] transfer
000001DFE4206AF8
ERROR in bulk write: -5

mcuee, 2021-03-10:
[128.947362] [00005ae8] libusb: error [winusbx_submit_bulk_transfer] unable
to match endpoint to an open interface - cancelling transfer

You need to understand the communication protocol in order to use libusb to communicate to your device. Apparently you are not writing to the correct endpoint in this case.

bclay1297, 2021-03-11:
How can we determine the correct endpoint?  The debug log showed and
endpoint address of 0xff.  I tried that even though it does not look valid
but it did not help

mcuee, 2021-03-11:
You have to know which USB ENDPOINT in order to use libusb to talk to the device. If you do not know, then nobody can help you.

If you do not know, then you need to spend the efforts to carry out reverse engineering.

First thing is to get to know the USB descriptors. You can use USBView.
https://ftdichip.com/utilities/

mcuee, 2021-03-11:
I will close this issue. This github is more for problem reporting and not for technical support. You can ask in libusb mailing list for further support.

closed 2021-03-11 by mcuee