libusb/libusb #887: manpages?

jlmgtech, 2021-03-10:
I understand this is likely going to be an annoying request, so the answer "no" is expected and fine, but I feel I should ask just in case.  libusb 0.x used to have manpages, and I noticed that doxygen is used as the documentation engine for the current incarnation.  A lot of C programmers on posix environments still use terminal environments with editors such as vi that have tight integration with man.  Is it possible to generate manpages for libusb?  How difficult would it be to include that as an option with the release?

Thank you in advance.

hjelmn, 2021-03-10:
I don't see any reason we could not generate man pages from the doxygen. Not a huge ask but may take some work if the generated files need any editing. I will mark this as a feature request and we will have to see if anyone picks it up. I may do the work if I find time but it would be after the next release at the earliest.

jlmgtech, 2021-03-11:
Awesome!  Thank you :-)