libusb/libusb #888: cataline 10.15 not support

developerss, 2021-03-12:
there is no way to install this plugin in cataline 10.15 
it works in 10.13 last /
is there a way i can install cataline 10.15 libusb plugin .,
i tried all ruby and bin bash all failed please show me a new way if possible to install it on 10.15

mcuee, 2021-03-12:
Which plugin? You can use homebrew to install libusb under macOS.

mcuee, 2021-03-12:
Homebrew lbusb
https://formulae.brew.sh/formula/libusb#default

mcuee, 2021-03-12:
If you are talking about Ruby libusb bindings, that belongs to different project.

closed 2021-03-12 by mcuee

developerss, 2021-03-12:
thank you for your suggestion  the home brew not run at cataline so without home brew i can install libusb ?>

mcuee, 2021-03-13:
I do not understand what you say about Homebrew not run under macOS Catalina 10.15. It works fine. 

If you do not want to use Homebrew, you can try the XCode project.

Anyway, github is for issue report, not for technical support. Please use libusb mailing list for further support.

developerss, 2021-03-13:
home brew not working in Caroline