libusb/libusb #890: remove "num_iso_packets" from structure "struct usbi_transfer"

lerdmann, 2021-03-15:
"num_iso_packets" is defined in both, "struct usbi_transfer" and also in "struct libusb_transfer".

This can lead to a clash as function "libusb_alloc_transfer" will set the value in "struct usbi_transfer" whereas function "libusb_fill_iso_transfer" will set the value in "struct libusb_transfer" which can lead to a mismatch.

I suggest to change "libusb_alloc_transfer" to :

`DEFAULT_VISIBILITY
struct libusb_transfer * LIBUSB_CALL libusb_alloc_transfer(
	int iso_packets)
{
	size_t priv_size;
	size_t alloc_size;
	unsigned char *ptr;
	struct usbi_transfer *itransfer;
	struct libusb_transfer *transfer;

	assert(iso_packets >= 0);
	if (iso_packets < 0)
		return NULL;

	priv_size = PTR_ALIGN(usbi_backend.transfer_priv_size);
	alloc_size = priv_size
		+ sizeof(struct usbi_transfer)
		+ sizeof(struct libusb_transfer)
		+ (sizeof(struct libusb_iso_packet_descriptor) * (size_t)iso_packets);
	ptr = calloc(1, alloc_size);
	if (!ptr)
		return NULL;

	itransfer = (struct usbi_transfer *)(ptr + priv_size);
	itransfer->priv = ptr;
	usbi_mutex_init(&itransfer->lock);
	transfer = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);
	usbi_dbg("transfer %p", transfer);
        if (transfer) transfer->num_iso_packets = iso_packets;
	return transfer;
}`

and to either remove parameter "num_iso_packets" from structure "struct usbi_transfer" and also from function "libusb_fill_iso_transfer" or ignore it (leave as deprecated). That will ensure that there is only one value for "num_iso_packets" and that this value will always be an integral part of the transfer and not some internal control variable.

mcuee, 2022-04-01:
@lerdmann Just wondering if you can create a pull request for this. Thanks.

seanm, 2026-01-06:
While working on [-fbounds-safety](https://clang.llvm.org/docs/BoundsSafety.html) support (https://github.com/libusb/libusb/pull/1659) I've come across, I think, basically this same issue, or at least close enough that I'll comment here instead of creating a new Issue.

I think `libusb_fill_iso_transfer()` has a design flaw.

Its current signature and implementation is:

```c
static inline void libusb_fill_iso_transfer(struct libusb_transfer *transfer,
	libusb_device_handle *dev_handle, unsigned char endpoint,
	unsigned char * buffer, int length, int num_iso_packets,
	libusb_transfer_cb_fn callback, void *user_data, unsigned int timeout)
{
	transfer->dev_handle = dev_handle;
	transfer->endpoint = endpoint;
	transfer->type = LIBUSB_TRANSFER_TYPE_ISOCHRONOUS;
	transfer->timeout = timeout;
	transfer->buffer = buffer;
	transfer->length = length;
	transfer->num_iso_packets = num_iso_packets; // 🔥
	transfer->user_data = user_data;
	transfer->callback = callback;
}
```

`struct libusb_transfer` ends with:

```c
struct libusb_transfer {
...   
	/** Number of isochronous packets. Only used for I/O with isochronous
	 * endpoints. Must be non-negative. */
	int num_iso_packets;

	/** Isochronous packet descriptors, for isochronous transfers only. */
	struct libusb_iso_packet_descriptor iso_packet_desc[LIBUSB_FLEXIBLE_ARRAY];
};
```

The last field is a variably-sized [C99 flexible array](https://en.wikipedia.org/wiki/Flexible_array_member) and the number of elements in the array is given by the preceding `num_iso_packets` field.  When this stuff is allocated in `libusb_alloc_transfer()` there is a multiplication `sizeof(struct libusb_iso_packet_descriptor) * (size_t)num_iso_packets;` and thus the correct size is allocated.

So far, so good.

But look at `libusb_fill_iso_transfer()`...  If you pass a larger `num_iso_packets`, the `num_iso_packets` field *is* updated, but the buffer in the `iso_packet_desc` field *is not* expanded (there is no `realloc()`!

Does this analysis seem correct?

seanm, 2026-01-07:
I think I understand what this is trying to do... I'll work on a PR...