libusb/libusb #891: Bug with some mass storage usb key bulk transfer timeout

supercairos, 2021-03-15:
Hi,

I have a bug with the USB device. A no-name USB disk (Mass Storage) that is not working (and probably many others, I haven't tested lots of devices).
The issue is that some bulk transfer timeout after 20s and then get canceled.

I'm on Windows and manly using USBDK (1.0.0-22) as a backend. 
It works with version 1.0.23 but doesn't on 1.0.24.

I've managed to pinpoint the issue to this particular commit:
https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6

It seems that after the transfer is submitted, the completion callback is never called.

- Submit here : https://github.com/libusb/libusb/blob/9c28ad219b654011783a42ec888ca87dbda704a6/libusb/os/windows_common.c#L743
- This is never called: 
https://github.com/libusb/libusb/blob/9c28ad219b654011783a42ec888ca87dbda704a6/libusb/os/windows_common.c#L806

It works well with previous commit (when doing the same within the USBDK backend). After the transfer, the GetOverlappedResult is called. 
If I disable USBDK and install the special WinUSB driver on the USB disk (aka: I use WinUSB backend) It works well. But the burden of having to install a driver makes it impossible to use for my use case. 
There is of course no errors in the console whatsoever and I've also compiled my code on Linux and it works like a charm. 

I'm not a pro on IOCompletionPort so if you could help me look into this issue, I would be very glad :)

Thanks 💯

mcuee, 2021-04-07:
Thanks for the finding.

Just wondering why you want to use libusb with a USB Mass Storage device. Thanks.

supercairos, 2021-04-15:
I'm building a USBoverIP tool to communicate remotely with a distant QEMU instance. We use USBREDIR to communicate. It works well for most device but since 1.0.24, It doesnt work with all the USB Mass Storage devices (some works, other don't)

mcuee, 2023-02-17:
@supercairos 

Just wondering if you can try latest libusb git HEAD to see if the issues are still there or not. Thanks.

supercairos, 2023-02-17:
Hi,

Thanks for the heads-up. I'm not working on this project anymore but I've forwarded to the team behind it internally. 
We'll keep you updated once we've tested it :)

pcorbineau, 2023-04-27:
Hi,

I do the test with 
 * libusb 1.0.26
 * usbredir 0.13.0
 * usbdk 1.0.22
 
 Like @supercairos said, We still have issues with mass storage. Webcam, or other devices are well forwarded to our qemu instance.
 
 I also tried to reinstall driver using zadig utility tool to force WinUSB driver to this mass storage. And This works but we can't ship this to our customers.
 
 Do you have any updates about this ?
 
 Thanks 👍

mcuee, 2023-04-28:
Thanks for the update. 
Please try the latest version of libusb git version to see if that helps.

mcuee, 2023-04-28:
Yet another thing to try is to use libusb-win32 filter driver instead of usbdk.

mcuee, 2024-04-30:
In the end, I guess both usbkd and libusb-win32 filter are not recommended.

mcuee, 2024-04-30:
usbdk has not been updated for quite a while and there seems to be multiple issues.
https://github.com/daynix/UsbDk
https://github.com/daynix/UsbDk/issues