libusb/libusb #895: Should accept DeviceInterfaceGUID  REG_SZ with Unicode null-terminator 0x0000 i.e. 2-bytes

CraigHutchinson, 2021-03-23:
https://github.com/libusb/libusb/blob/bda2344f5e7d460decd3968492faa7a5fb2797cd/libusb/os/windows_winusb.c#L1589

A device implementing a DeviceInterfaceGUID  field which reports 40-bytes (19+null-terminated unicode characters) is reported as having "unexpected type/size of DeviceInterfaceGUID for '%s'". 

The current code only allows the worst case where only a single-byte null is provided. As per the documentation the user should cater for incorrect termination however the current logic is in error when handling the correct/expected null termination exists!

For example a well formed unicode string:
`using t = decltype(u"DeviceInterfaceGUID");`

where t becomes 40-bytes as each character including the null consumes 2-bytes:
`char16_t[20]`

CraigHutchinson, 2021-03-25:
@mcuee Could you or somebody with permission link this to the pull-request #896 Cheers

mcuee, 2021-07-11:
#927 has been merged to fix this.

closed 2021-07-11 by mcuee