libusb/libusb #897: windows_winusb not re-enumerating devices correctly

CraigHutchinson, 2021-03-23:
The Dfu-Util project causes a reset of a device from DFU-Runtime to DFU-Mode. On windows the WinUSB device `libusb_get_device_list()` backend  re-uses the existing device reporting "found existing device for session [62]" descriptors which causes a bug in that changed/updated device descriptors are not accessible via the LibUsb api: [https://sourceforge.net/p/dfu-util/tickets/110/libusb_get_device_list](https://sourceforge.net/p/dfu-util/tickets/110/libusb_get_device_list)

# Current WinUSB implements only the following checks
1.  `DevInst`  is used to match session_id
https://github.com/libusb/libusb/blob/bda2344f5e7d460decd3968492faa7a5fb2797cd/libusb/os/windows_winusb.c#L1672
2. `priv->dev_id` path is used to match the device
https://github.com/libusb/libusb/blob/bda2344f5e7d460decd3968492faa7a5fb2797cd/libusb/os/windows_winusb.c#L1691

# The issue
When the device is re-enumerated it is the `ClassGuid` that actually changes and should also be checked

I am happy to try and make a patch with some pointers i.e:
-  I am not sure if the `ClassGuid` is cached in priv for the device at present to perform the extra comparison in the same manner as at L1691 already?

tormodvolden, 2021-03-24:
dfu-util bug link should be: https://sourceforge.net/p/dfu-util/tickets/110/

CraigHutchinson, 2021-03-24:
This appear to possible be a previously reported issue that was unresolved: https://github.com/libusb/libusb/issues/295

CraigHutchinson, 2021-03-24:
In the case of the issue reported here calling `libusb_exit(ctx)` followed by `libusb_init(ctx)` to instantiate a new context is a workaround

mcuee, 2021-03-25:
IMHO the ultimate fix will be #86 Windows hotplug support. Before that, it may be difficult to fix.

tormodvolden, 2021-03-25:
Craig suggested comparing the ClassGuid. Here is a compile-tested patch, adding ClassGuid to the cached priv and comparing it. Maybe Craig can test it and let us know.

[0001-winusb-Compare-ClassGuid-in-check-for-existing-devic.patch.txt](https://github.com/libusb/libusb/files/6208291/0001-winusb-Compare-ClassGuid-in-check-for-existing-devic.patch.txt)

CraigHutchinson, 2021-03-26:
I will review the patch soon and 

> Craig suggested comparing the ClassGuid. Here is a compile-tested patch, adding ClassGuid to the cached priv and comparing it. Maybe Craig can test it and let us know. [...]

I will check this asap :)

CraigHutchinson, 2021-03-26:
I can confirm this change DID fix the issue :D

 I am not sure this is a 'generic' fix for all occurrences of issues such as this as I m not 100% what the ClassGUID represents here i.e. Does it work because just because in this case the devices is switching between a Composite during RT mode and a different type in DFU mode.

CraigHutchinson, 2021-03-26:
I have adjusted the title of the issue to be more specific. That is that re-enumeration isn't happening when the ClassGuid of my device was changed.

It is possible there are other, currently unknown issues for re-enumeration on windows that may be addressed in separate Issue topics as they are discovered.

The proposed patch @tormodvolden works and I made a pull-request with the respective code adjustments which directly address this issue.

tormodvolden, 2021-03-26:
Thanks for the testing. Did you do any changes to my (git-formatted) patch other than splitting it up and adding a comment? Otherwise I see a slight attribution issue here.

For the GUID, I /think/ it will change if the triplet vid/pid/bcdDevice changes and that would already make the difference in your case. Maybe it is also changes with interface changes.

CraigHutchinson, 2021-03-26:
So true, comment was only addition I made.

please if you make a pull request with your changes it will be correctly attributed and I will close mine. Cheers.

Emteq Ltd, Sussex Innovation Centre, Brighton BN1 9SB
Company Registration: 09708880

This e-mail (including any attachments) is considered confidential and is intended only for the recipient(s) listed above. Any review, use, disclosure, distribution or copying of this e-mail is prohibited except by or on behalf of the intended recipient. If you have received this email in error, please notify me immediately by reply e-mail, delete this e-mail, and do not disclose its contents to anyone. Thank you.

tormodvolden, 2021-03-26:
From looking into where the ClassGuid comes from, my theory is as follows: The ClassGuid is assigned based on a matching driver or one of the predefined classes. The matching might involve looking at the interface descriptors. Windows will treat a device as new and re-evaluate the ClassGuid assignment if vid/pid/bcdDevice has changed. It would be possible that the ClassGuid stays the same even if some descriptors have changed, but checking for ClassGuid is a good improvement.

tormodvolden, 2021-05-22:
@CraigHutchinson Maybe you can close your pull request to avoid confusion, thanks.

mcuee, 2021-05-24:
I have closed #899 and refer to #900.

CraigHutchinson, 2021-05-24:
@mcuee Thanks for that, got it sorted before I started back at work :)

CraigHutchinson, 2021-05-24:
The associated change #900 rectifies the reported issue.

mcuee, 2021-06-08:
I will leave this open until #900 gets merged.

mcuee, 2021-07-09:
#900 has been merged.

closed 2021-07-09 by mcuee